MarlinFirmware/Marlin #15888: USB Serial and internal serial conflict in SKR1.3

masgar, 2019-11-13:
I'm using a SKR1.3 connected through internal serial (TFT pins) to control a SKR mini that act as "home made" MMU2.
With marlin 2.0.x released in September all works fine. With latest release of marlin 2.0.x all works fine until I send a command through USB serial. It crash. The octopi log says that "device reports readiness to read but returned no data (device disconnected or multiple access on port?"

boelle, 2019-11-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-11-13 by boelle

masgar, 2019-11-13:
IMVHO it is a bug due to changes applied in the lasts commits, anyway I'll try to fix elsewhere. Thanks.

boelle, 2019-11-13:
i only see a question even if i try to read slowly

and the issue template was not followed so hard to guess what it is

masgar, 2019-11-13:
Ok, my fault. Sorry.

Max

masgar, 2019-11-13:
Just for your info, changing mmu2.h it works again:

/*
#if SERIAL_USB
  #define MMU_RX_SIZE 256
  #define MMU_TX_SIZE 256
#else
  #define MMU_RX_SIZE  16
  #define MMU_TX_SIZE  16
#endif
*/

#define MMU_RX_SIZE  1024
#define MMU_TX_SIZE  1024

Thanks,
Max

thisiskeithb, 2019-11-14:
MMU2 users on OctoPrint 1.3.12 [were experiencing similar problems](https://community.octoprint.org/t/prusa-mmu2-discussion-tips-hints/6797) when the MMU printer profile wasn’t configured correctly.

masgar, 2019-11-14:
It is not an octoprint issue. I connect both boards to my laptop on 2 different serial (cu.usbserial14201 and cu.usbserial14101). I can see dialog from "mmu" and marlin but as soon as I try to send a command to marlin board the serial are disconnected. Changing mmu2.h (or mmu2.cpp) as above, the issue is fixed. In older commits of marlin 2.0.x there wasn't this definition and the TX/RX size was in mmu2.cpp at 2048

mmu2.cpp from september commits
char MMU2::rx_buffer[2048], MMU2::tx_buffer[2048];

mmu2.cpp from current commit:
char MMU2::rx_buffer[MMU_RX_SIZE], MMU2::tx_buffer[MMU_TX_SIZE];

so I changed mmu2.h

Max