MarlinFirmware/Marlin #15891: [BUG] (short description)

joirma, 2019-11-13:
When I compile for skr 1.3 i get this error:

1º I edit platformio.ini and set default_envs to LPC1768

2º Then Run build:

> Executing task: C:\Users\jiran\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; framework: arduino; board: nxp_lpc1768)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing framework-arduino-lpc176x @ >=0.2.2
Error: Detected unknown package 'framework-arduino-lpc176x'
The terminal process terminated with exit code: 1

As you can look i take a compilation error.

PackageManager: Installing framework-arduino-lpc176x @ >=0.2.2
Error: Detected unknown package 'framework-arduino-lpc176x'
The terminal process terminated with exit code: 1

This happens to me for both vscode + platformio core 4.1.0 on windows, and platformio core from linux command line.

joirma, 2019-11-13:
the problem is the Spanish dns that have been banned by court order the domain: https://raw.githubusercontent.com/

closed 2019-11-13 by joirma

erberru70, 2019-11-29:
> Cuando compilo para skr 1.3 me sale este error:
> 
> 1º Branch: bugfix-2.0.x
> 2º Edito platformio.ini y configuro default_envs en LPC1768
> 
> 3º Ejecutar build:
> 
> ## Resultado:
> Tarea de ejecución: C: \ Users \ jiran.platformio \ penv \ Scripts \ platformio.exe run
> 
> Procesamiento LPC1768 (plataforma: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip ; framework: arduino; placa: nxp_lpc1768)
> 
> ## PackageManager: Instalación de framework-arduino-lpc176x @> = 0.2.2
> Error: Paquete desconocido detectado 'framework-arduino-lpc176x'
> El proceso del terminal finalizó con el código de salida: 1
> Como puedes ver, tomo un error de compilación.
> 
> PackageManager: Instalación de framework-arduino-lpc176x @> = 0.2.2
> Error: Paquete desconocido detectado 'framework-arduino-lpc176x'
> El proceso del terminal finalizó con el código de salida: 1
> 
> Esto me sucede tanto para vscode + platformio core 4.1.0 en Windows como para Platformio Core desde la línea de comandos de Linux.

hello, excuse me, I have the same problem,
What is the solution to compile marlin?