MarlinFirmware/Marlin #15892: [BUG] AUTO_BED_LEVELING_3POINT movement to far

schmerzlich, 2019-11-13:
Marlin-Dev2.1.x
Anet A8 Plus

AUTO_BED_LEVELING_3POINT ignores PROBE_OFFSET and point override (PROBE_PT_x_X) settings.

X - Axis runns in hard stop at X-max position.

The only way to set up usable probing positions is by setting the BED_SIZE smaller (as it is) or to set the MIN_PROBE_EDGE larger. But thats not the way of proper usability.

boelle, 2019-11-13:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

shitcreek, 2019-11-13:
Is your machine a delta/scara?

sjasonsmith, 2019-11-14:
> Marlin-Dev2.1.x

This branch isn't currently up to date. You should ignore it and keep using bugfix-2.0.x for now.
Can you try it with the latest state of bugfix-2.0.x and report back if you still have the issue? If yes, be sure to post configs.

You will have to patch up your configs a bit. Definition of probe offsets and points has all changed versus what you are using now.

Be sure to use the latest state of bf2. A change was committed 15 hours ago for one issue impacting non-mesh bed leveling and probe offsets.

schmerzlich, 2019-11-14:
Hello sjasonsmith,
thank you, I'll try this evening and give you a report. 

Hey shitcreek, no it's not a delta.

boelle, 2019-11-16:
what report do you have for us?

schmerzlich, 2019-11-18:
I am sorry but everybody of my family is actually ill so I havent got any
time to analyse the problem so far.

boelle, 2019-11-18:
no problemo, have been ill myself with stomac flu

sjasonsmith, 2019-11-24:
@schmerzlich, I'm sorry to hear your family was sick. I suggest we have @boelle (or you) close this issue, and it can be re-opened or a new one can be created if it doesn't work with the bugfix-2.0.x branch.

I have very recently tested AUTO_BED_LEVELING_3POINT on bugfix-2.0.x with and without custom probe points, and I believe it will work for you from that branch.

boelle, 2019-11-24:
as @sjasonsmith says we can reopen this one if it does not work

closed 2019-11-24 by boelle