MarlinFirmware/Marlin #15893: [BUG] 2.0 SKR PRO case light keeps dark (compile warn: ...variable unused...)

docgalaxyblock, 2019-11-13:
### Description

I tried to activate the case light function at my skr pro. the case light keeps dark and after watching the compiling log I found a warning
`Marlin\src\feature\caselight.cpp:61:64: warning: unused variable 'n10ct' [-Wunused-variable]
   const uint8_t i = case_light_on ? case_light_brightness : 0, n10ct = INVERT_CASE_LIGHT ? 255 - i : i;`
Could that be the problem why my case light keeps dark?

### Steps to Reproduce
[Marlin-SKR-PRO-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3842571/Marlin-SKR-PRO-bugfix-2.0.x.zip)

1. setup a skr pro at marlin
2. Enable Case light
3. Compile it -> there should be the warning

#### Additional Information

I tried different commits since september and the function did not work at all commits i have tested.
I use VS Code
Tested at other board (skr 1.3 and ramps) and there is the case light working fine. (Used the same project only changed the board and the board specific pins) Seems to be a skr pro or stm32 specific bug

Vertabreak, 2019-11-13:
i use this feature on the gt2560 and it works correctly are you sure you configured it correctly for the board.

docgalaxyblock, 2019-11-13:
like I said
At the SKR 1.3 and arduino mega/Ramps it works perfect

LinoBarreca, 2019-11-13:
which pin are you using as "case light"?

gilliands, 2019-11-13:
I have an SKR 1.3 and I used to use the servo pin and two unused GPIO for PWM to drive an RGB LED strip via a buffer.
      #define RGB_LED_R_PIN     P2_00
      #define RGB_LED_G_PIN     P3_25
      #define RGB_LED_B_PIN     P3_26
      #define RGB_LED_W_PIN     -1
When I boot, only the red light works and the others are zero.  I have to actively enter the LED menu and change to a colour or even just turn it off and back on again for normal operation.  Looks like only one pin is being set up for PWM after power up.
Used to work until about a week ago round the time when the platform IO updates made change to the ADC pins.  Coincidence?

LinoBarreca, 2019-11-13:
@gilliands completely unrelated to the OP

Evg33, 2019-11-15:
I use White LED Strip 24V on Heater2 as case light. Working ok.

Evg33, 2019-11-15:
I have an SKR Pro and I use the Heater2 for PWM to drive an White LED 24V strip directly from board.
The case light working.

LinoBarreca, 2019-11-15:
@Evg33 @docgalaxyblock isn't replying since 3 days. I guess he realized he made a mistake in his config and disappeared

docgalaxyblock, 2019-11-15:
@Evg33 I have it at switchmode (no pwm) and I am using HE1
Could you send me your complee config please

Evg33, 2019-11-15:
@docgalaxyblock 

pins_BIGTREE_SKR_PRO_V1.1.h 
`#define HEATER_2_PIN       -1   // PB0  // Heater1
`
Configuration_adv.h
```
/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  //#define CASE_LIGHT_PIN 4                  // Override the default pin if needed
  #define CASE_LIGHT_PIN PB0
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 127   // Set default power-up brightness (0-255, requires PWM pin)
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  #define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif

```

LinoBarreca, 2019-11-15:
@evg33 you shouldn't modify pins file.
The logic behind this is PB0 (CPU pin) - HEATER_2_PIN (BOARD pin) - CASE_LIGHT_PIN (DEVICE pin)
If you restore pins file to default and #define CASE_LIGHT_PIN HEATER_2_PIN
you are actually saying "I connected case light wires to heater 2 port" and should work the same. Making it easier to read and maintain.

Evg33, 2019-11-15:
ok. thx.

Evg33, 2019-11-15:
@docgalaxyblock

Configuration_adv.h

```
/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  //#define CASE_LIGHT_PIN 4                  // Override the default pin if needed
  #define CASE_LIGHT_PIN HEATER_2_PIN
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 127   // Set default power-up brightness (0-255, requires PWM pin)
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  #define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif
```

docgalaxyblock, 2019-11-15:
@LinoBarreca Thank you that you metioned that
Such a smoother and nicer way

But it did not work for me
@Evg33 Would it be possible that you send me your hole project as a zip? PN me

Evg33, 2019-11-15:
@docgalaxyblock https://github.com/Evg33/Marlin/tree/SKR_Pro

davidcgu, 2019-11-16:
Don't use a heat mosfet pin, use a fan pin.

boelle, 2019-11-16:
> At the SKR 1.3 and arduino mega/Ramps it works perfect

Sure you are not having a hardware issue or using wrong type of pin?

LinoBarreca, 2019-11-16:
@davidcgu for some lights (I.e. led bars) the fan pin is not recommended since the fan transistor is too small. Tou will likely burn it. That's why the default in marlin is a heater (they are meant to suck serious power so they have adequate mosfets)

Evg33, 2019-11-16:
> Don't use a heat mosfet pin, use a fan pin.

Why?
Mosfets for heaters are designed for more current than for fans.

boelle, 2019-11-16:
> Don't use a heat mosfet pin, use a fan pin.

Dude, you are a bit wrong here, a fan pin is not able to supply enough current and you just fry the mosfet,

to everyone else: bad advise was given here

LinoBarreca, 2019-11-16:
Well...it might work in some circumstances, provided that you know what you are doing...but if you don't just avoid taking a risk.

LinoBarreca, 2019-11-16:
Just leaving my two cents here about the "it doesn't work" problem. Since you didn't give us more informations about the lights you are using I assume you are powering a led strip. In this case: are you sure you connected the cable with the right polarity? Did you try flipping it? 😂😂😂😂

docgalaxyblock, 2019-11-16:
@LinoBarreca I am using four 10W COB Led strips. Atm they are directly connected to my powersupply.
If I configure the controller Fan to use the heater 2 the it works...so no fried mosfet no hardware error.
I will try to compile @Evg33 s firmware now and do stp by step adjusting to my needs until it does not work

LinoBarreca, 2019-11-16:
>
> If I configure the controller Fan to use the heater 2 the it works...so no fried mosfet no hardware error.

I didn't understand. Are you saying that you told marlin to use fan 0 with connector heater 2 and tested that the connector works by changing the fan speed?
Also... 4*10W how many volts and connected how?

boelle, 2019-11-16:
> Atm they are directly connected to my powersupply.

boelle, 2019-11-16:
> controller Fan

that is  fan to cool of the electronics, make sure to not confuse us

boelle, 2019-11-16:
> I will try to compile @Evg33 s firmware

use a tool like winmerge to make a note of all changes he have done so you can compare and tell us what changes make it work for main marlin, forks/copies are not supported

LinoBarreca, 2019-11-16:
@boelle im pretty sure @evg33 uses a standard source with a proper configuration. I still believe that is an user error not related to marlin. By the way, does sanitycheck controls if the same board port is used for two different things?
For example "hey, you defined case light to use heater 2 but you said you have 3 heaters so there's a conflict"

boelle, 2019-11-16:
we just need to be 100% sure, 

but yeah given that so very few have a problem with case light i also think this is an config issue and not a bug

docgalaxyblock, 2019-11-16:
So my setup:
Hotend: bed mosfet
Bed: external mosfet controlled by a Ext pin
Fan 0:  layer fan
Fan 1: E0 auto fan
Fan 2: controller fan
Heater 1&2 unesed
Heater 0 should be the case light

boelle, 2019-11-16:
that is more complicated than it has to be

docgalaxyblock, 2019-11-16:
why? its so clear and useful ^^

LinoBarreca, 2019-11-16:
Not to look rude but what the fu*k.
That isn't a 3d printer it's a rubik's cube 😂😂😂 the more you change the more risk of mistakes you have.
By the way: how do you warm the nozzle? With a lighter?

boelle, 2019-11-16:
this is more simple:

Hotend: heater 0 (hotend connector)
Bed: heater 1 (bed connector)
Fan 0: layer fan
Fan 1: E0 auto fan
Fan 2: controller fan
case light could be heater 2

you dont need an external mofet

boelle, 2019-11-16:
i agree with @LinoBarreca you are making it far more complicated for yourself and the risk of mistakes are higher

LinoBarreca, 2019-11-16:
@boelle the skr pro has 3 heaters & 1 bed (oversized) so I would simplify that too.

docgalaxyblock, 2019-11-16:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/picture/BIGTREETECH-SKR-PRO-V1.1.jpg
[image]
I've got a 600W 400² heated bed connected to an external mosfet
My 50W catrige uses the original bed mosfet (picture: BED)

LinoBarreca, 2019-11-16:
That explains pretty much.

docgalaxyblock, 2019-11-16:
that is also the reason why I am using such beafy cob leds

boelle, 2019-11-16:
yes, but your setup is far more complicated than it has to be 

Hotend: heater 0
Bed via external mosfet connected to bed connector
Fan 0: layer fan
Fan 1: E0 auto fan
Fan 2: controller fan
case light could be any of the remaining heaters

docgalaxyblock, 2019-11-16:
Ok I found the differences between me and @Evg33 
I am using the official marlin repo, evg the BTT ones there it seems to be fixed

boelle, 2019-11-16:
you sure?

[image]

docgalaxyblock, 2019-11-16:
Not completly because I only downloaded the zip and did not take a look at the link
The conf.h looks like the one from the BTT repo not like the one from the official one

Only @Evg33 knows the answer
I need to do some work now. hopefully @Evg33 can answer this questin until I am back

boelle, 2019-11-16:
the default config prob has more extruders than you use

set extruder to 1 and then you can use any of the unused heaters for the case light

boelle, 2019-11-16:
> @Evg33 can answer this questin until I am back

look at the pic, it has the answer

davidcgu, 2019-11-16:
I'm also using cabin light and with some leds the power consumption is minimal, in worst case you can use a mosfet, give it a try and will see, will work on a fan mosfet output but not on the heaters mosfets.... but keep playing as much as u wanted....

LinoBarreca, 2019-11-16:
@davidcgu if you know that the amps used by the lights are less than the specs for the part yes. If you don't know just don't try. Or you'll fry everything.

boelle, 2019-11-16:
ultimately we need to know if this is a config/user error or not

the issue list if for bugs and feature requests only, not long debates on how to do things :-D

docgalaxyblock, 2019-11-16:
I found the "problem"
If I activate `  #define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.` I am getting the warning from the topic. If I leave it deactivated I do not get it
The Printer is in use atm, will test it asap to see if I could use the case light now
If true there is a little bug at Marlin

boelle, 2019-11-16:
there is a difference between warning and fail when you compile

i also get warnings but if the compile is a success i wont worry about it

davidcgu, 2019-11-16:
I tell u.... will never work fine with the heat pins....

sjasonsmith, 2019-11-16:
@LinoBarreca could the heater pins have changed to disallow HW PWM in the changes you made? I thought you tried to keep HW PWM on both fan and heater pins.

docgalaxyblock, 2019-11-16:
@davidcgu 
Do you recommend me to use a fan pin because the Heater No.2 has strange behavior (it lights up shortly only when the case light brightness is been updated)?

I have defined HE0 as my case light pin, seems to been working fine

But there is a bug somewhere at the `#define CASE_LIGHT_NO_BRIGHTNESS` function. If you enable it you can not turn on the case light.

davidcgu, 2019-11-16:
> @davidcgu
> Do you recommend me to use a fan pin because the Heater No.2 has strange behavior (it lights up shortly only when the case light brightness is been updated)?
>
> I have defined HE0 as my case [...]

I'm using a fan pin with a led strip, the cosumtion should be really low, if required you coul install a small mosfet (around 50 cents)  I can control the brightness and works perfect, I can not provide you deep info but het pins work on a different way than a normal PWM what makes it does not work properly, I told that many posts ago and seems nobody really believe me... test it yourself...

I have offical marlin from 9 Nov.

Regards

Evg33, 2019-11-16:
I use official marlin with changed configs.
I don't use BTT versions.

boelle, 2019-11-26:
@docgalaxyblock btw what version of marlin are you using?

LinoBarreca, 2019-11-26:
@boelle you can close this. it's definitely working on bugfix 2.0.x

closed 2019-11-26 by boelle

docgalaxyblock, 2019-11-27:
@boelle At the moment I opend the Issue the newest bugfix
@LinoBarreca Did you enable NO_BRIGHTNESS?!
I have checked it again (Marlin v. 13:30 timezone Berlin)
And it did not work either my skr pro, skr 1.3 or ethen a ramps
If you enable the no brightness function it do not work anymore!!!
So Issue still exists