MarlinFirmware/Marlin #15895: [BUG] SKR-MINI-E3-V1.2 TouchMi -SDCARD

kramer04, 2019-11-13:
if I set the Z-offset for the TouchMi and save it, as soon as I start printing the printer freezes

### Steps to Reproduce
Go to configuration --> TouchMi --> Init TouchMi --> Probe Z Offset --> Save
 launch a print and the printer freezes
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3843031/Configuration.zip)

Expected to print after probing Z Offset

**Actual behavior:** [What actually happens]
The printer freezes and i've to to download again the firmware

damvcoool, 2019-11-13:
It maybe a Problem with the Virtual EEPROM, try modifying

Marlin/src/pins/stm32/pins_BTT_SKR_MINI_E3_V1_2.h

line 38
#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE)
to 
#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 2048 - 2 * EEPROM_PAGE_SIZE)

kramer04, 2019-11-14:
Now it seems that's ok.
No freeze

boelle, 2019-11-16:
will close as fixed

> Now it seems that's ok.

closed 2019-11-16 by boelle