MarlinFirmware/Marlin #15896: [BUG] Ramps-1.5 + Marlin-1.1.9 + Discount Smart Controller: Rotary Encoder not working

rocketkittens, 2019-11-13:
### Description
 Ramps-1.5 + Marlin-1.1.9 + Discount Smart Controller:  
  **Rotary Encoder click opens menu, but rotating encoder doesn't move cursor**

### Steps to Reproduce
Configuration.h https://pastebin.com/raw/3Wifr32y  
Configuration_adv.h https://pastebin.com/raw/i3ejKMBs
Powered by: USB or 12v + 5v to Arduino

**Expected behavior:** [What you expect to happen]
Rotating the rotary encoder on RepRap Discount Smart Controller should move cursor in the menu.

**Actual behavior:** [What actually happens]
Cursor does not move when rotating the rotary encoder. However pressing/clicking the rotary encoder opens the LCD menu.

If this is not a software issue, how can i debug the rotary encoder component - check continuity while rotating, if im able to find the pin allocated to rotation?  It is not getting enough power?

shitcreek, 2019-11-14:
1.x is no longer being worked on. You should try 2.x.

rocketkittens, 2019-11-14:
I'm unable to build 2.0-bugfix:
**sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp:38:19: fatal error: timer.h: No such file or directory**

ellensp, 2019-11-14:
Try the GLCD tester https://github.com/ellensp/rrd-glcd-tester  designed for a ramps 1.4 so check 1.5 hasnt changed the pins.

rocketkittens, 2019-11-14:
> 
> 
> Try the GLCD tester https://github.com/ellensp/rrd-glcd-tester designed for a ramps 1.4 so check 1.5 hasnt changed the pins.

I tried it, it just fills the entire screen... I dont see any program.

rocketkittens, 2019-11-14:
**I found the issue,** i am working inside an enclosure and with poor lighting, i used a flashlight and noticed half of the Ramps board is not fully inserted into the Arduino, i made sure all pins are seated and viola!

closed 2019-11-14 by rocketkittens