MarlinFirmware/Marlin #15897: [BUG] Printer Ignores Set Acceleration Values When Doing Extrusions

theofficialgman, 2019-11-14:
The problem I observe can be seen when doing any movement (x,y, or z) that simultaneously involves using the extruder. Whenever a G1 command is executed that includes E values, the values for acceleration set by M204 P are ignored. To me, it seems that the printer attempts to accelerate as quickly as possible. Setting the eeprom and firmware values for acceleration low (as in 500mm/s^2) does not affect this.

Observed on an 8 bit Mega 2560 based arduino board. Marlin-Bugfix-2.0.x 227951a (the most recent as of posting)

Going back to an older (last year) 1.1.9 firmware shows no signs of problems.
The last firmware which I was one also had this issue (bugfix-2.0.x built on october 30th)

*Steps to Reproduce*
1. Start any print and observe intense rattling and rapid movement of the printer which causes missed steps, etc.
2. For a safer demonstration, startup a print and set the extruder temperature to 0 which will force marlin to interpret any move which includes an extrusion as only a travel move. Immediately when marlin activates the hotend too cold feature and prevents extrusion, the printer follows the acceleration values set in the firmware/eeprom/config.
3.Manually heating up the extruder mid-print reactivates the extruder and moves that involve an extrude have the issue again.

**Expected behavior:** 
Printer slows speed change according to set accelerations

**Actual behavior:** 
Printer ignores acceleration values for movements that include an extrusion

#### Additional Information

Purely travel moves function correctly according to their defined acceleration values.
Setting the jerk higher or lower does not stop the problem.
I am currently using the classic jerk.
I have manual_mesh_bed_leveling enabled
I have a cartesian style printer

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3844068/configs.zip)

boelle, 2019-11-24:
@theofficialgman have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

theofficialgman, 2019-11-24:
@boelle I was actually leaving this bug report open as reference for #15908 which detailed the same problem as this bug report but was posted after mine and gained more traction for some reason. Problem was solved yesterday and merged into the 2.0.x bugfix branch with commit 0fcd1f4.

closed 2019-11-24 by theofficialgman