MarlinFirmware/Marlin #15900: [BUG] Mesh Bed Leveling Broken

theofficialgman, 2019-11-14:
If mesh bed leveling is enabled, the execution of any move that includes movements in X or Y will cause a reset. Movements in Z are not affected and can be executed. This exists up to the most recent release as of posting, Marlin-Bugfix-2.0.x d1d1437

The bug has existed, from my testing, since the 30th of september (the oldest firmware that I tested since I had to change eeprom saved mesh values due to the V70 and V71 eeprom versions on different firmwares)

### Steps to Reproduce

1. Enable MESH_BED_LEVELING is cofiguration.h
2. Configure a mesh any way you like.
3. Begin a print and, at any point during or before the print, enable the mesh and the next move involving X or Y will cause the printer to reset.

**Expected behavior:** 
Printer uses the mesh to adjust Z height during the first few layers of a print

**Actual behavior:** 
Printer resets.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3848971/Configs.zip)

closed 2019-11-14 by theofficialgman

sbailey128, 2020-05-01:
Hi,

I too am having the same problem. Also have a i3 style printer. Do you figure out the problem or a work-around?

theofficialgman, 2020-05-01:
@sbailey128 I am trying very hard to remember exactly what the problem was back then. I feel like it was just with me missing the change of how points are defined in the mesh as I think I manually imputing the numbers in over gcode. Maybe it was something like the X,Y values used to go from (1 to n) and now they go from (0 to n-1). Are you manually defining the points or are you doing it through the printer lcd?

sbailey128, 2020-05-01:
I am using the printer lcd to run the mesh bed leveling routine. That part works perfect. Doing an M420 v1 command afterwards (from my computer) shows the correct 9 values for the mesh. Like you though, once I try to do a print...my Arduino does a reset as soon as it starts to move. I have looked at the configuration.h file adnauseam (and a few changes) to no effect.

theofficialgman, 2020-05-01:
@sbailey128 I'm assuming that doing G29 S0 prints out the mesh as well just fine?

sbailey128, 2020-05-01:
I will try those commands and get back to you. Thanks

sbailey128, 2020-05-01:
G29 S0 - prints out the mesh...it looks correct.

M504 - prints out "OK"

sbailey128, 2020-05-01:
Correction...M504 - prints out "EEPROM OK"

sbailey128, 2020-05-01:
Hi again...so do you currently use Mesh Bed Leveling on your I3 and it works ok now?

theofficialgman, 2020-05-01:
I do and yes it does work. I am running a firmware version released very soon after I originally opened this issue (November 18, 2019 firmware).

sbailey128, 2020-05-01:
I am using the latest version (2.0.5.3). My printer came with version 0.9?? (I think) and did not have mesh leveling then. Thought I give the new version a try because it had mesh leveling. I am convinced then it is something in my configuration.h file that is causing this. Any ideas?

theofficialgman, 2020-05-01:
Could be, are you editing your original 0.9 config file or starting from scratch with the premade 2.0.5.3 config files? I've attached mine incase you want to look (note they were built back in November so some stuff might have changed). I also started off at 0.9 a few years ago, Marlin has come a long way since then.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4564892/Configs.zip)

sbailey128, 2020-05-01:
I am using the configuration.h file that came with version 2.0.5.3 and editing it. There is ALOT of new stuff in there. Thanks for sending your file...I'll take a look.

sbailey128, 2020-05-01:
From your configuration.h file, you were using Marlin version 2.0.0.0. I will download that version of the software from Github and try your settings (slightly modified to match my hardware). I'll let you know what I find.

theofficialgman, 2020-05-01:
Let me give you the exact version I was using, I'm not sure exact which comment is was but here are the files. You will just have to swap out the config files.
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4565553/Marlin-bugfix-2.0.x.zip)

sbailey128, 2020-05-01:
Ok...thanks.

sbailey128, 2020-05-02:
Hi again. Well...your config files worked. Mesh bed leveling all good. Still not sure what your mods did to make it work, but I am not going to worry about that right now. Thanks again.