MarlinFirmware/Marlin #15901: [BUG] speaker tone make the controller fan to stop momentary

WireBe, 2019-11-15:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/3849409/Configurations.zip)

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

 with #define speaker , each tone output make the E0_AUTO_FAN to spin down while every TONE is played, and resume after the tone ended. But NOT with //#define speaker (buzzer output)

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

  

1. compile with #define SPEAKER.
2. heat the hot end to make is't fan to start (40 deg in included config files).
3. output a tone by pressing on rotary switch or by M300 to hear the fan slowing down.

**Expected behavior:** [What you expect to happen]
Fan continue to spin at full speed 255

**Actual behavior:** [What actually happens]
Fan slow down as long as the tone is playing
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2019-11-15:
This sounds to me like there is a conflict between the timers for the two outputs. I'm not familiar enough with that board to know a proper fix (if there is one), but you can probably work around the issue by using `FAN_SOFT_PWM`.

thinkyhead, 2019-11-15:
Maybe someone with more LPC176x savvy can enlighten us whether P1_22 and P1_31 share any timers or vectors. In the meantime, do a quick check to make sure there is no short between P1_22 and P1_31 on the board.

WireBe, 2019-11-15:
@thinkyhead   
I'll check for a short later today, 
but when i comment out `#define SPEAKER` , OR use a previous commit this bug disappear.
 
@sjasonsmith i'll try FAN_SOFT_PWM too and let you know,

p3p, 2019-11-15:
The fallback software PWM (P1_22) runs at the same ISR priority as the Tone generator but there isn't a conflict as such, also a 255 PWM signal is just gpio high so it shouldn't be effected..

WireBe, 2019-11-15:
@thinkyhead 
No short pin confirmed between P1_22 and P1_31 ( physical pin 36 and 20 on chip )
I've tried an old .bin that report august31 on the printer info that work fine. 

@sjasonsmith i've tried `FAN_SOFT_PWM` had no effect.

boelle, 2019-11-25:
@p3p do you think this is a config error in some way?

boelle, 2019-11-26:
@WireBe btw what version of marlin are you using?

WireBe, 2019-11-26:
@boelle 2.0.x

WireBe, 2019-12-03:
I made a new try with release 2.0.x, same issue. 
I attach new configs files, made from scratch.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3917282/Configuration.zip)

boelle, 2019-12-07:
@WireBe since 2.0 was just released a few days ago has this changed this issue at all?

WireBe, 2019-12-08:
@boelle no change with release 2.0, 
i can confirm it is NOT hardware related since i flash with an old backup and my rig is ok.

thanks for you time boelle. it's appreciated.

boelle, 2019-12-24:
and still no change with all the updates in the past 16 days?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

WireBe, 2019-12-27:
December 26,  
2.0.1, 2019-12-24, fresh config. Same Results. same BUG.

i repeat for the **third** time: I have a .bin file from an old marlin, who can prove my hardware workin.  

Thanks.
@boelle  I found someone for testing, and he should report here, as soon as he has results.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4004059/config.zip)

boelle, 2020-01-08:
@WireBe when will those results come in?

WireBe, 2020-01-09:
@boelle, I sent email to ask. we were in Christmas holiday, don't you?

boelle, 2020-01-14:
xmas in Denmark ends 27/12 and new year is only one day 1/1-2020

WireBe, 2020-01-14:
@boelle  So, i wish you Merry Christmas an Happy New Year 2020!

This ''**_unconfirmed bug_**'' has been **_confirmed_** by the engineer/maker of this TH3d Board himself.
Direct communications are made between him and Scott from Marlin.
I'll close this bug as soon as they merge the TH3D branch's fix of it into the main one.

Thanks.

boelle, 2020-01-30:
a lot of updates and fixing has happend in the last week, is the problem still there?

WireBe, 2020-02-02:
I saw no indication that this bug as been addressed or that the TH3D fix as been merged. I'll ask for any news about it.

thinkyhead, 2020-02-02:
A couple of timers have been tweaked in the recent releases, but not LPC176x. The issue could have accidentally been fixed by some change along the way, but if you've tested today's "bugfix" branch and the issue is still there let us know.

I have a TH3D EZ-Board here on a CR-10S. I don't use tones a lot so I haven't noticed this. But I will give it a look. The board is a little quirky in the way it beeps when starting up.

If TH3D has a fix for this, great.

To get some insight, you can try enabling `PINS_DEBUGGING` and then `M43` will give a report of all the pin assignments.

thinkyhead, 2020-02-02:
Unrelated, probably, but I see they made this change in U2.R1:
```cpp
#define CONTROLLERFAN_SPEED 255
```

WireBe, 2020-02-02:
@thinkyhead 
Hoo, interesting! i'll give it a try asap and let you know,

WireBe, 2020-02-02:
@thinkyhead 
I tested last bugfix and found the exact same behavior as before.
I pasted M43 result. 
Don't know if it's normal, but the speaker made a endless repeated  bip tone at ~1hz as soon as issued the M43 command. Had to reset for it to stop.

Enough lost time for me, i just turned off //#define speaker  

@boelle You can close this one, if living with don't bother Marlin's team. I can't help anymore.

[CONFIGS.zip](https://github.com/MarlinFirmware/Marlin/files/4144588/CONFIGS.zip)

Send: M43
Recv: PIN: 0.00        <unused/unknown>                       Input  = 1
Recv: PIN: 0.01        <unused/unknown>                       Input  = 1
Recv: PIN: 0.02 (A 2)  <unused/unknown>   Analog in =  1000   Input  = 1
Recv: PIN: 0.03 (A 3)  <unused/unknown>   Analog in =   836   Input  = 0
Recv: PIN: 0.04        X_SERIAL_TX_PIN                        Output = 1
Recv: PIN: 0.05        X_SERIAL_RX_PIN                        Input  = 1
Recv: PIN: 0.06        SDSS                                   protected
Recv: .                SS_PIN                                 protected
Recv: PIN: 0.07        DOGLCD_SCK                             protected
Recv: .                SCK_PIN                                protected
Recv: PIN: 0.08        MISO_PIN                               protected
Recv: PIN: 0.09        DOGLCD_MOSI                            protected
Recv: .                MOSI_PIN                               protected
Recv: PIN: 0.10        Y_SERIAL_TX_PIN                        Output = 1
Recv: PIN: 0.11        Y_SERIAL_RX_PIN                        Input  = 1
Recv: PIN: 0.15        LCD_PINS_D4                            Output = 0
Recv: PIN: 0.16        LCD_PINS_RS                            Output = 0
Recv: PIN: 0.17        <unused/unknown>                       Input  = 1
Recv: PIN: 0.18        LCD_PINS_ENABLE                        Output = 0
Recv: PIN: 0.19        Z_SERIAL_TX_PIN                        Output = 1
Recv: PIN: 0.20        Z_SERIAL_RX_PIN                        Input  = 1
Recv: PIN: 0.21        E0_SERIAL_RX_PIN                       Input  = 1
Recv: PIN: 0.22        E0_SERIAL_TX_PIN                       Output = 1
Recv: PIN: 0.23 (A23)  <unused/unknown>   Analog in =   956   Input  = 0
Recv: PIN: 0.24 (A24)  <unused/unknown>   Analog in =   984   Input  = 0
Recv: PIN: 0.25 (A25)  <unused/unknown>   Analog in =   835   Input  = 0
Recv: PIN: 0.26 (A26)  <unused/unknown>   Analog in =   803   Input  = 0
Recv: PIN: 0.27        <unused/unknown>                       Input  = 0
Recv: PIN: 0.28        <unused/unknown>                       Input  = 0
Recv: PIN: 0.29        <unused/unknown>                       Input  = 1
Recv: PIN: 0.30        <unused/unknown>                       Input  = 0
Recv: PIN: 1.00        <unused/unknown>                       Input  = 1
Recv: PIN: 1.01        <unused/unknown>                       Input  = 1
Recv: PIN: 1.04        E0_DIR_PIN                             protected
Recv: PIN: 1.08        E0_ENABLE_PIN                          protected
Recv: PIN: 1.09        Y_ENABLE_PIN                           protected
Recv: PIN: 1.10        Y_DIR_PIN                              protected
Recv: PIN: 1.14        Z_ENABLE_PIN                           protected
Recv: PIN: 1.15        Z_DIR_PIN                              protected
Recv: PIN: 1.16        X_DIR_PIN                              protected
Recv: PIN: 1.17        X_ENABLE_PIN                           protected
Recv: PIN: 1.18        <unused/unknown>                       Output = 1
Recv: PIN: 1.19        <unused/unknown>                       Output = 1
Recv: PIN: 1.20        <unused/unknown>                       Output = 1
Recv: PIN: 1.21        <unused/unknown>                       Output = 1
Recv: PIN: 1.22        E0_AUTO_FAN_PIN                        protected
Recv: .                FAN1_PIN                               protected
Recv: .                ORIG_E0_AUTO_FAN_PIN                   protected
Recv: .                ORIG_E1_AUTO_FAN_PIN                   protected
Recv: .                ORIG_E2_AUTO_FAN_PIN                   protected
Recv: PIN: 1.23        <unused/unknown>                       Input  = 1
Recv: PIN: 1.24        X_MIN_PIN                              protected
Recv: .                X_STOP_PIN                             protected
Recv: PIN: 1.25        Y_MIN_PIN                              protected
Recv: .                Y_STOP_PIN                             protected
Recv: PIN: 1.26        Z_MIN_PIN                              protected
Recv: .                Z_STOP_PIN                             protected
Recv: PIN: 1.27        FIL_RUNOUT_PIN                         Input  = 0
Recv: PIN: 1.28        <unused/unknown>                       Input  = 1
Recv: PIN: 1.29        <unused/unknown>                       Input  = 1
Recv: PIN: 1.30 (A62)  BTN_ENC                                Analog in =   918
Recv: PIN: 1.31 (A63)  BEEPER_PIN                             Output = 0
Recv: PIN: 2.00        X_STEP_PIN                             protected
Recv: PIN: 2.01        Y_STEP_PIN                             protected
Recv: PIN: 2.02        Z_STEP_PIN                             protected
Recv: PIN: 2.03        E0_STEP_PIN                            protected
Recv: PIN: 2.04        SERVO0_PIN                             Input  = 0
Recv: PIN: 2.05        HEATER_BED_PIN                         protected
[... cut ...]
Recv: PIN: 2.11        KILL_PIN                               Input  = 1
Recv: PIN: 2.12        <unused/unknown>                       Input  = 1
Recv: PIN: 2.13        E1_DIR_PIN                             Input  = 1
Recv: PIN: 3.25        BTN_EN2                                Analog in =     0
Recv: PIN: 3.26        BTN_EN1                                Input  = 1
Recv: PIN: 4.28        <unused/unknown>                       Input  = 1
Recv: PIN: 4.29        E1_ENABLE_PIN                          Input  = 1
Recv: ok

thinkyhead, 2020-02-03:
Thanks for the extra info. I'll play around with the beeper and fan speed on my EZBoard next chance I get. The Tone library on the LPC1768 is probably not unlike others, using some kind of interrupt that steals a lot of cycles. But it's weird that it could steal enough to affect FAN PWM.

But note that when a fan is at 100% power there is no software involved in keeping the fan going. So if it still slows down, that implies some current is being stolen someplace else. Its interesting, anyway. Maybe the 5V regulator is sketchy…?

GriffinPaquette, 2020-02-03:
@thinkyhead The PWM choice of 255 for the controller fan was to maximize the cooling to the board itself. I run mine on the U2 branch at 70% of that without issue. Definitely not a 5V regulator issue there.

I have been able to replicate this though the customer's bin and my own compilation, but have yet to see where TH3D had made any notable changes to fix it. With that being said, the U2 branch has been working fine with the EZBoard. I have completely ruled out a hardware issue as others have already done themselves.

thinkyhead, 2020-03-06:
> I have completely ruled out a hardware issue as others have already done themselves.

Are you also testing with `E0_AUTO_FAN_PIN` set to 1.22? I expect that `BEEPER_PIN` and the `E0_AUTO_FAN_PIN` share a timer, and possibly the modes are set wrongly for the associated pins.

closed 2020-07-29 by github-actions[bot]