MarlinFirmware/Marlin #15902: [BUG] Immediate temperature runaway halt on E1 with SINGLENOZZLE

eightheads, 2019-11-15:
### Description
I am getting a temperature runaway halt on E1 immediately on boot with today's latest pull. The same configuration with a pull from 16 days ago did not exhibit this issue.

### Steps to Reproduce

1. Define 2 extruders
2. Define SINGLENOZZLE
3. Define hotend thermistor on TEMP_SENSOR_0

**Expected behavior:** 
Expect extruders to share temp sensor 0

**Actual behavior:**
On boot Marlin gives an immediate temperature runway halt on E1. Due to this cannot connect to control board.

#### Additional Information
SKR V1.3 control board. No issues from pull 16 days ago and no issues when moving back to that firmware.

p3p, 2019-11-15:
Make sure you are not reusing any files from the old Marlin version, including the boards pins file.

eightheads, 2019-11-15:
> Make sure you are not reusing any files from the old Marlin version, including the boards pins file.

I just ran a complete fresh pull and same issue, no files were reused from previous pull.

p3p, 2019-11-15:
There has been some issues since I updated the LPC framework with the adc filters taking time to initialise but this shouldn't be an issue anymore so I'm unsure what to ask you to diagnose the problem.

Is this only a problem for you with `SINGLENOZZLE` configured?

If it is a initialisation problem setting `#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000` would work around it or you can try reverting some of the 12bit adc change by updating `src/HAL/HAL_LPC1768/HAL.h` to see if its a problem with that: 
```
#define HAL_ADC_RESOLUTION     10   // 15 bit maximum, raw temperature is stored as int16_t
//#define HAL_ADC_FILTERED            // Disable oversampling done in Marlin as ADC values already filtered in HAL
```

eightheads, 2019-11-15:
> There have been some issues since I updated the LPC framework with the adc filters taking time to initialise but this shouldn't be an issue anymore so I'm unsure what to ask you to diagnose the problem.
> 
> Is this only a problem for you with `SINGLENOZZLE` configured?
> 
> If it is a initialisation problem setting `#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000` would work around it or you can try reverting some of the 12bit adc change by updating `src/HAL/HAL_LPC1768/HAL.h` to see if its a problem with that:
> 
> ```
> #define HAL_ADC_RESOLUTION     10   // 15 bit maximum, raw temperature is stored as int16_t
> //#define HAL_ADC_FILTERED            // Disable oversampling done in Marlin as ADC values already filtered in HAL
> ```

I will hopefully have time to test some of these suggestions tomorrow. I'll try defining a single extruder first then go from there.

eightheads, 2019-11-17:
> There have been some issues since I updated the LPC framework with the adc filters taking time to initialise but this shouldn't be an issue anymore so I'm unsure what to ask you to diagnose the [...]

Just tried with the latest from today and same problem. I also tried defining a single extruder and same issue. I will now try the suggestions you placed above.

eightheads, 2019-11-17:
@p3p 

> If it is a initialization problem setting `#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000` would work around it or you can try reverting some of the 12bit adc change by updating `src/HAL/HAL_LPC1768/HAL.h` to see if its a problem with that:
> 
#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000 will allow it to boot but it shortly hits thermal runaway after the grace period is over. On boot it shows Extruder as 300c and bed at 120c immediately. This is with the same previous and verified thermistor definitions of 13 and 1.

> ```
> #define HAL_ADC_RESOLUTION     10   // 15 bit maximum, raw temperature is stored as int16_t
> //#define HAL_ADC_FILTERED            // Disable oversampling done in Marlin as ADC values already filtered in HAL
> ```

This did not seem to help or change anything. And these are with a single extruder defined and sharednozzle undefined.

eightheads, 2019-11-17:
Pull from 10/31 (#15739) runs without issue with the same configuration

p3p, 2019-11-17:
Can you attach the exact source you are compiling in a zip file.

eightheads, 2019-11-18:
> Can you attach the exact source you are compiling in a zip file.

@p3p can I email you directly? I have some specific questions.

p3p, 2019-11-18:
Marlins discord would be the best option https://discord.gg/n5NJ59y

eightheads, 2019-11-18:
It appears I had old ADC definitions for the thermistor pins for my board. I will test after work today but I believe this issue has been my own and I am going to close this ticket. If I continue to have issues I will re-open or create a new ticket with better information. Thanks all!

closed 2019-11-18 by eightheads

matrimcauthon11, 2020-01-02:
Hi @eightheads, Sorry to bug you but I am in the exact spot after trying to setup an skr mini e3 using the 2.0.0 marlin tag. How did you go about updating the pins if i might inquire? Thanks for the help!

eightheads, 2020-01-02:
> Hi @eightheads, Sorry to bug you but I am in the exact spot after trying to setup an skr mini e3 using the 2.0.0 marlin tag. How did you go about updating the pins if i might inquire? Thanks for the help!

You should be able to update to the latest Marlin 2.0 pins file and work again.