MarlinFirmware/Marlin #15903: [BUG] SKR Pro - TMC2209 UART - only X driver gets serial address

PGrave, 2019-11-15:
### SKR Pro - TMC2209 UART - only X driver gets serial address

### Steps to Reproduce
[GIT_REPORT.zip](https://github.com/MarlinFirmware/Marlin/files/3850221/GIT_REPORT.zip)

1. [Compile latest Marlin 2.0]
2. [Flash firmware]
3. [TMC CONNECTION ERROR on display]
4. [M122 on Octoprint shows X driver with address and communication enabled, no addresses and no communication with Y, Z, E, E1]

**Expected behavior:** [All drivers with UART communication ]

**Actual behavior:** [Only X axis driver communicates]

#### Additional Information
Previous compiled and working firmware using STM32@5.6.0; BTT Serial TMC

sjasonsmith, 2019-11-15:
How are your jumpers configured beneath the drivers?

sjasonsmith, 2019-11-15:
If you take a picture at an angle I think you can see all the pins near and under the drivers without removing them. I would just like to confirm that all the jumpers look correct to match your config. I'm going to try talking to multiple 2209s on my board, I only used a 2209 on X when I initially tested.

sjasonsmith, 2019-11-15:
OK, it is working for me. I only have four 2209 drivers, so I only have X through E.
I just rebased this config, so it is my SKR Pro configs on top of 27cb4db80eae73eb842f1a944260517bb8783eb3 .

```
M122
                X       Y       Z       E
Address         0       0       0       0
Enabled         false   false   false   false
Set current     800     800     800     800
RMS current     795     795     795     795
MAX current     1121    1121    1121    1121
Run current     25/31   25/31   25/31   25/31
Hold current    12/31   12/31   12/31   12/31
CS actual       12/31   12/31   12/31   12/31
PWM scale       14      14      14      14
vsense          1=.18   1=.18   1=.18   1=.18
stealthChop     true    true    true    true
msteps          16      16      16      16
tstep           max     max     max     max
pwm
threshold
[mm/s]
OT prewarn      false   false   false   false
off time        4       4       4       4
blank time      24      24      24      24
hysteresis
-end            2       2       2       2
-start          1       1       1       1
Stallguard thrs 0       0       0       0
DRVSTATUS       X       Y       Z       E
stst            *       *       *       *
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
                X       0xC0:0C:00:00
                Y       0xC0:0C:00:00
                Z       0xC0:0C:00:00
                E       0xC0:0C:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
ok
```

PGrave, 2019-11-15:
Hello @sjasonsmith and thank you for your reply. First of all, let me tell you that this configuration has been printing ok with Marlin 2.0, compiled on 26-10-2019. Just downloaded current version, changed configuration, configuration_adv and pins files to match my hardware. Compared versions old with new, just updated my specifics on new files. I updated and upgraded platform (PlatformIo on VSCode) and libraries. So, no hardware changes or modifications since last compile, a few days ago, just software.

PGrave, 2019-11-15:
Jumpers are all in the same position, UART, and slave addresses all zero on configuration_adv. Haven't changed that part of the configuration. While using BTT serial implementation (previous compile), drivers get addresses. I'll try to send a screen from M122 report, I've downgraded to previous firmware and all is working.

sjasonsmith, 2019-11-15:
Could you take a picture of the jumpers? I understand that you haven't changed anything, but I can't figure out why it works fine on my board and not yours, if we are doing the same thing.

PGrave, 2019-11-15:
I've also sent a zip file with my configs and platformIo compile log.

PGrave, 2019-11-15:
[image]
Picture of jumpers above,
and this is a current M122 result:
Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	550	550	550	450	450
Recv: RMS current	520	520	520	428	428
Recv: MAX current	733	733	733	603	603
Recv: Run current	16/31	16/31	16/31	13/31	13/31
Recv: Hold current	8/31	8/31	8/31	6/31	6/31
Recv: CS actual	8/31	8/31	8/31	6/31	6/31
Recv: PWM scale	10	10	10	7	7
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold	98	98	658	64	64
Recv: [mm/s]		100	100	3	30	30
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0xC0:08:00:00
Recv: 		Z	0xC0:08:00:00
Recv: 		E	0xC0:06:00:00
Recv: 		E1	0xC0:06:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok

sjasonsmith, 2019-11-15:
What is wrong with that M122 result? This is with your old firmware or the new firmware?

PGrave, 2019-11-15:
This is my old firmware, to which I've returned. So, to be clear: from this working setup to the not working one, just Marlin version, new STM32 5.7.0, default Marlin TMC library and my changes/adapts on the config files.

sjasonsmith, 2019-11-15:
Can you post your configs for the old/working firmware, like you did for the new firmware?

PGrave, 2019-11-15:
On the new configuration, which is not loaded on the machine now, only X driver gets an address. I'll reflash new firmware (it has to be later, sorry) and paste here the result of M122.
Please be aware that the files in the zip on the first post are the NEW configs. Maybe those can be useful, might be a stupid mistake from me, too many awaken hours :) Thanks in advance for any possible help!

PGrave, 2019-11-15:
Thank you @thinkyhead for your help. I've init memory by the display command, which was acknowledged with a confirmation response. Then 
a save configuration command. But I can and will issue those by serial console, after reflashing with new firmware, just to ensure.

LinoBarreca, 2019-11-15:
It looks like you are using tmcstepper 0.6.0

LinoBarreca, 2019-11-15:
@sjasonsmith do you?

sjasonsmith, 2019-11-15:
I was using 0.5.2.

LinoBarreca, 2019-11-15:
@sjasonsmith Does spi use softserial?
Why should it? I rather guess that tmcstepper uses 5.7.0 spi library

sjasonsmith, 2019-11-15:
SPI is separate. SoftwareSerial is only used for 2208/2209 drivers, which use a UART. I don't know what TMCStepper uses for SPI, I don't have any SPI drivers.

LinoBarreca, 2019-11-15:
Oh. I'm sorry. I read it wrong. He's on uart. Sorry. I was boarding the plane. Be back in 3h.

thisiskeithb, 2019-11-15:
Marlin likely needs to be patched: https://github.com/teemuatlut/TMCStepper/issues/93

LinoBarreca, 2019-11-15:
Likely tmcstepper needs to be patched...since, if it's as @sjasonsmith said, the next stm framework (5.8.0) will contain the same softserial which is in marlin now.

LinoBarreca, 2019-11-15:
Maybe @teemuatlut should be involved here as well.

TB1405, 2019-11-17:
i also use a SKR PRO and TMC2209 on X and Y. I downloaded the neweset bugfixes today but also got the error message after m122 
`Send: M122
Recv: 		X	Y
Recv: Address		0	0
Recv: Enabled		false	false
Recv: Set current	950	950
Recv: RMS current	939	939
Recv: MAX current	1324	1324
Recv: Run current	16/31	16/31
Recv: Hold current	8/31	8/31
Recv: CS actual	8/31	0/31
Recv: PWM scale	10	0
Recv: vsense		0=.325	0=.325
Recv: stealthChop	true	false
Recv: msteps		16	256
Recv: tstep		max	0
Recv: pwm
Recv: threshold	152	2432
Recv: [mm/s]		65	65
Recv: OT prewarn	false	false
Recv: off time	4	0
Recv: blank time	24	16
Recv: hysteresis
Recv: -end		2	-3
Recv: -start		1	1
Recv: Stallguard thrs	130	130
Recv: DRVSTATUS	X	Y
Recv: stst		*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... Error: All LOW`

Will it work if use the TMCStepper 0.5.2. 
i changed in my platformIO.ini theTMCStepper >=0.5.2,<0.6.0 to TMCStepper@0.5.2 but i doesnt load the 0.5.2 libary

PGrave, 2019-11-17:
It should work with TMC0.5.2. To force it, use the change made on platformIo.ini and you should use platformIo "Clean" option, to clean the pioenvs folder and force the download of new libraries. I think like that it will load the 0.5.2 version.
I haven't tested it yet, but should go fine like that.

PGrave, 2019-11-18:
Tested with TMC0.5.2, M122 result:
Send: M122
Recv: 		        X	Y	Z	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	550	550	550	550	550
Recv: RMS current	520	520	520	520	520
Recv: MAX current	733	733	733	733	733
Recv: Run current	16/31	16/31	16/31	16/31	16/31
Recv: Hold current	8/31	8/31	8/31	8/31	8/31
Recv: CS actual	8/31	8/31	8/31	8/31	8/31
Recv: PWM scale	10	10	10	10	10
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold	98	98	658	64	64
Recv: [mm/s]		100	100	3	30	30
Recv: OT prewarn	false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst		        *	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0xC0:08:00:00
Recv: 		Z	0xC0:08:00:00
Recv: 		E	0xC0:08:00:00
Recv: 		E1	0xC0:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok

PGrave, 2019-11-18:
Tested with TMC0.6.1, M122 result:
Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false
Recv: Set current	550	550	550	550	550
Recv: RMS current	520	520	520	520	520
Recv: MAX current	733	733	733	733	733
Recv: Run current	16/31	16/31	16/31	16/31	16/31
Recv: Hold current	8/31	8/31	8/31	8/31	8/31
Recv: CS actual	8/31	8/31	8/31	8/31	8/31
Recv: PWM scale	10	10	10	10	10
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold	98	98	658	64	64
Recv: [mm/s]		100	100	3	30	30
Recv: OT prewarn	false	false	false	false	false
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0xC0:08:00:00
Recv: 		Z	0xC0:08:00:00
Recv: 		E	0xC0:08:00:00
Recv: 		E1	0xC0:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok

PGrave, 2019-11-18:
From my side, this is a closed issue. Thanks to all that helped! Go Marlin Team! :)

closed 2019-11-18 by PGrave