MarlinFirmware/Marlin #15906: [BUG] strange build error, missing timer.h ?

AndKe, 2019-11-15:
### Description

Current 2.0 bugfix fail to build, I also see reference to STM32Core when building for Atmega2560

### Steps to Reproduce
downloded master  2.0-dev, building using Arduino 1.8.10

**Actual behavior:** [What actually happens]

```

/home/andre/Arduino/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/andre/Arduino/arduino-1.8.10/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.10/tools-builder -tools /home/andre/Arduino/arduino-1.8.10/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.10/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10810 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_177968 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/andre/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/andre/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/andre/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/andre/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/home/andre/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/andre/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/andre/Arduino/projects/Marlin-bugfix-2.0.x-new/Marlin/Marlin.ino
......................
I/home/andre/Arduino/projects/libraries/U8glib/src /tmp/arduino/sketch/src/HAL/HAL_STM32/SoftwareSerial.cpp -o /dev/null
Alternatives for timer.h: []
ResolveLibrary(timer.h)
  -> candidates: []
/tmp/arduino/sketch/src/HAL/HAL_STM32/SoftwareSerial.cpp:38:10: fatal error: timer.h: No such file or directory
Multiple libraries were found for "U8glib.h"
 #include <timer.h>
 Used: /home/andre/Arduino/projects/libraries/U8glib
          ^~~~~~~~~
compilation terminated.
Using library U8glib at version 1.19.1 in folder: /home/andre/Arduino/projects/libraries/U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

```

I did delete /tmp/arduino*  and restrted Arduino IDE after switching to atmegea2560 , so I am at a loss as why HAL_STM32 is mentioned ?

shitcreek, 2019-11-15:
Are you using bugfix 2.0 or Dev 2.1? Also, please complete the template and provide config files.

AndKe, 2019-11-15:
Hi, its very late now so I'll be quick: It is bugfix2.0 - and config files cannot affect this problem.  the only relevant part would be this:    #define MOTHERBOARD BOARD_RAMPS_14_EFB

shitcreek, 2019-11-16:
Iirc, this is an issue with the new Arduino. Try an older version of Arduino or use PlatformIO - it's way better anyway.

sjasonsmith, 2019-11-16:
This was already fixed in bd321d4b7c1cb9a64956c953b64fed8586ac61fc. Make sure you are using the latest commits from bugfix-2.0.x.

boelle, 2019-11-16:
as this has to do with arduino ide and has been fixed with https://github.com/MarlinFirmware/Marlin/commit/bd321d4b7c1cb9a64956c953b64fed8586ac61fc

i will close this one, wr can always reopen if there is an issue and config files are provided

closed 2019-11-16 by boelle