MarlinFirmware/Marlin #15908: [BUG] heavy vibrations when printing on delta, SKR1.3 with TMC 2208 UART

arminth, 2019-11-16:
Somewhere in between September 15 2019 an now, something must have been changed in the codebase (or libraries) that causes this strange issue.
With a fresh copy of Marlin 2.0.x downloaded on Sep 15 2019, I managed to create a perfectly working configuration for my Anycubic Kossel Linear plus that I had pimped with a BTT SKR1.3 board and 2208 Steppers in UART mode.
In November, I transfered the configuration to another fresh copy of Marlin2.0.x to be up to date with the latest features and configuration. I started from the clean Delta Configuration in the samples.cfg and changed evrything according to my setup going thru the configuration files line by line transferring the relevant dasta into the new configs.
Formware compiled and uploaded normally. 
I started a delta calibration print (basically a big circle with a dia of about160mm). The print startetd and immediately when print-moves started (not happening on non extruding moives afaik), there was a heavy vibration noticeable. It looked almost like stuttering at a frequency of about 30-40Hz. 
I thought it was related to https://github.com/MarlinFirmware/Marlin/issues/15692 but it turned out, that it is more of a delta-problem.
Strange thing is, that I cant compile my Septmber Version (which I still have) anymore. It throws a compile error that looks like an issue with updated library of TMC or so. But I am not an expert in that respect. I still have the bin-file so I can make my delta a working machine anytime, but it's my tinkerhorse anyways, so not too bad. Esxcept for it needs a full calibration on each firmwarechange, which is quite time consuming.
@epicurean70 experienced similar issues.

I attach my configuration files for refernce. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3854112/Marlin.zip)
 
Hope this can be fixed!
Cheers
Armin
P.S.: If needed I can make a video of the stuttering.

theofficialgman, 2019-11-16:
I believe this is the same issue that I posted 3 days ago, [BUG] Printer Ignores Set Acceleration Values When Doing Extrusions
#15897

DenisSemenov1970, 2019-11-18:
I have the same problem on my Delta. But if the turn off LIN_ADVANCE this problem disappears. But I need LIN_ADVANCE! 
Version 14.09.2019 works correctly 

Help please.

jharmn, 2019-11-19:
I had the same issue recently, and reverted to a Aug 22nd build (as I had it working on that before). In my case, linear advance on/off didn't change the vibration problem. I experimented with jerk etc extensively, to no positive effect (lower jerk did appear to soften the effect somewhat).

karabas2011, 2019-11-20:
The same was on MKS SBASE, but was gone after low down seg/sec to 80 and something else, I do not remember now

DenisSemenov1970, 2019-11-20:
I found out after what change the vibration begins. This happened 09.10.2019 after [Enable junction deviation by default (#15481)]. I will try to find out what happened after this change.

ManuelMcLure, 2019-11-20:
With a newer version you can try enabling ``CLASSIC_JERK`` - if that causes the vibration to go away it's definitely Junction Deviation related.

thisiskeithb, 2019-11-20:
`JUNCTION_DEVIATION_MM` can be tuned, so you shouldn't have to revert back to `CLASSIC_JERK` to fix the issue.

ManuelMcLure, 2019-11-20:
I was thinking of it as a debugging test, not a solution/workaround :)

ModestAndroid, 2019-11-21:
I'm having this issue too. Just got an Anycubic Kossel Linear Plus, installed an SKR 1.3 and uploaded the newest Marlin. Tried printing once, and it was just a vibrating nightmare. I never had junction deviation enabled though... Classic jerk was set from the start.

ModestAndroid, 2019-11-21:
So I took a look at #15897 and commented out linear advance completely. No more jittering/stuttering/vibrating like crazy. Running Marlin downloaded today. Not able to use TMCStepper library 0.6.1 though... Won't compile it with the rest (defaults to 0.5.2).

jharmn, 2019-11-21:
Reporting back, I reflashed today with latest...I disabled linear advance, and it did indeed resolve the issue. I guess I hadn't tried that before and remembered wrong.

DenisSemenov1970, 2019-11-21:
I found the problem. In change #15481 not correct replacement instead of !BOTH( JUNCTION_DEVIATION, LIN_ADVANCE) to HAS_CLASSIC_E_JERK.
The correct replacement instead of !BOTH (JUNCTION_DEVIATION, LIN_ADVANCE) to !HAS_LINEAR_E_JERK.

Need to fix in the following files:
Marlin\src\gcode\config\M200-M205.cpp line 162 instead of #if HAS_CLASSIC_E_JERK to #if !HAS_LINEAR_E_JERK,
Marlin\src\gcode\config\M92.cpp line 79 instead of #if HAS_CLASSIC_JERK to #if HAS_CLASSIC_JERK && !HAS_LINEAR_E_JERK,
Marlin\Marlin\src\lcd\menu\menu_advanced.cpp line 475 instead of #if HAS_CLASSIC_E_JERK to #if !HAS_LINEAR_E_JERK,
Marlin\Marlin\src\module\configuration_store.cpp lines 2261,2776 and 2794, instead of #if HAS_CLASSIC_E_JERK to #if !HAS_LINEAR_E_JERK.

This is true for today's latest version. 
Problems with vibration of the DELTAS disappear.
I will try to pull the request to make these changes.

arminth, 2019-11-21:
@DenisSemenov1970 that is great news!
Let's hope, PR makes it into the code quickly! 
Please let us know when you do the PR!

arminth, 2019-11-21:
@DenisSemenov1970 
I think something went wrong with your PR. It says you want to merge 5139 commits from bugfix-2.0.x into 1.1.x.
Maybe @thinkyhead could help.

p3p, 2019-11-23:
LPC176x builds will now the latest TMCStepper lib by default, some people have mentioned that is enough to fix the vibrations issues for them.

theofficialgman, 2019-11-24:
@arminth It looks like the pull request was merged yesterday into the bugfix 2.0.x branch. Consider closing this bug report if the problem has been solved.

closed 2019-11-25 by arminth