MarlinFirmware/Marlin #15910: MKS Robin Lite v1.1. Freeze at "Update 100%"

inetd128, 2019-11-16:
I have the same issue like at:
https://github.com/MarlinFirmware/Marlin/issues/15143
1. Build MKSLite.bin
2. Put SD Card in Printer]
3. Turn on and wait
4. Freeze at "Updating 100%"

This problem appeared around 2-4 days ago. Because I compiled Marlin 5 days ago and firmware started after update. I just updated my Marlin sources from git yesterdays and get this problem.

boelle, 2019-11-17:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

inetd128, 2019-11-17:
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/3855529/conf.zip)

inetd128, 2019-11-21:
ping

boelle, 2019-11-24:
no need to post ping as text, the people with knowledge are very few and they work for free so it will just take time

boelle, 2019-11-26:
btw what version are you using?

inetd128, 2019-11-29:
Bugfix-2.0.x from git

boelle, 2019-12-07:
@inetd128 since 2.0 was just released a few days ago has this changed this issue at all?

inetd128, 2019-12-08:
Has been tested few minutes ago. Nothing changed, Still freeze at "Update 100%".
BTW, there is absolutely the same bug https://github.com/MarlinFirmware/Marlin/issues/15998

boelle, 2019-12-11:
oki... since the isuues are the same and you came first i will close the other one so we dont have to hunt down dublicates

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

inetd128, 2019-12-25:
Still the same

boelle, 2020-01-08:
@inetd128  still no good news with the latest bugfix 2.0.x?

inetd128, 2020-01-08:
I don't know. Switched to klipper 1 week ago.

boelle, 2020-01-14:
will close this one

you where the only one with the problem and since you now run klipper it does not make much sense to keep it open, 

if others see this after its closed and they have the same problem we can reopen it

closed 2020-01-14 by boelle