MarlinFirmware/Marlin #15911: [BUG] Unable to compile bugfix-2.0.x for melzi boards - oversize

CRCinAU, 2019-11-17:
### Description

After the commit 27cb4db80eae73eb842f1a944260517bb8783eb3 - it is no longer possible to build  bugfix-2.0.x for the Creality range of boards based on the sanguino_atmega1284p board with:
* SD Card enabled
* BLTouch enabled.

```*** [checkprogsize] Explicit exit, status 1
DATA:    [===       ]  28.0% (used 4594 bytes from 16384 bytes)
PROGRAM: [==========]  100.2% (used 130316 bytes from 130048 bytes)
```

If I do a git revert 27cb4db80eae73eb842f1a944260517bb8783eb3, then all builds well:

```DATA:    [===       ]  35.0% (used 5730 bytes from 16384 bytes)
PROGRAM: [==========]  99.7% (used 129718 bytes from 130048 bytes)
```

RudyBenner, 2019-11-17:
where are your configuration files?

RudyBenner, 2019-11-17:
Built sketch from default files, including your settings from above.
I added the LCD display.

Linking .pio\build\melzi\firmware.elf
Building .pio\build\melzi\firmware.hex
Checking size .pio\build\melzi\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [===       ]  26.3% (used 4302 bytes from 16384 bytes)
PROGRAM: [========  ]  78.7% (used 102356 bytes from 130048 bytes)

boelle, 2019-11-17:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

CRCinAU, 2019-11-17:
Added Configuration.h & Configuration_adv.h

[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/3855574/Configuration_files.zip)

I also copy across the _Bootscreen.h and _Statusscreen.h from `config/examples/Creality/Ender-3/` if that matters?

boelle, 2019-11-17:
nope should not matter, but it can take a while before the code people have time to look at it
i'm just the Custodian making sure that they have what they need to start look at things

drzejkopf, 2019-11-17:
@CRCinAU Try disabling AUTOTEMP

CRCinAU, 2019-11-17:
> @CRCinAU Try disabling AUTOTEMP

Yep - that was enough to squeeze everything back in:
```
DATA:    [===       ]  34.9% (used 5712 bytes from 16384 bytes)
PROGRAM: [==========]  99.3% (used 129110 bytes from 130048 bytes)
```

Is that a new feature or something just enabled by default now?

I know this is frowned upon, but if we start running low on PROGRAM again, would pulling out the Marlin boot logo be possible? It seems like thats not really wanted - seeing as there isn't a simple define that I can find to disable it, but sadly these 128K devices are being stretched harder and harder :\

boelle, 2019-11-17:
yes they are, i held on for a long time but gave up and got a re-arm instead of the 2560mega

and the old stuff will get more expensive as time goes on making it a no brainer to switch, but a small warning, if you want to go 32 bit think twice about getting stuff from china, there is an old saying that you get what you pay for

CRCinAU, 2019-11-17:
I've personally got a SKR Mini e3 on the way to tinker with... No doubt that'll lead to more frustrations - but there's also a lot of folks that are still buying Ender 3's etc that will still get the 128Kb board - so I'd like to keep them updated for others as long as I can...

closed 2019-11-17 by CRCinAU

drzejkopf, 2019-11-17:
@CRCinAU No, it's here for a long time, it was in 1.0.x I believe, but that feature is barely used, so I assume it's safe to disable it in most cases. And, commenting
`#define SHOW_BOOTSCREEN`
should be enough to disable Marlin bootlogo

CRCinAU, 2019-11-17:
> @CRCinAU No, it's here for a long time, it was in 1.0.x I believe, but that feature is barely used, so I assume it's safe to disable it in most cases. And, commenting
> `#define SHOW_BOOTSCREEN`
> should be enough to disable Marlin bootlogo

Shoot - you're right! I just checked mine again and it doesn't display the Marlin bootscreen... Although I would have sworn it did. Disregard that as an old man getting confused :D