MarlinFirmware/Marlin #15912: [FR] (feature request title) Microsoft Standard Driver for 3D Printers

Glod76, 2019-11-17:
Can this be integrated into marlin to be detected by windows 10?

https://docs.microsoft.com/en-us/windows-hardware/drivers/3dprint/microsoft-standard-driver-for-3d-printers-

BeatSlayer, 2019-11-17:
I don't believe that's possible for most of the boards we use because of the fdti/chg serial conversion chips they have.

ellensp, 2019-11-17:
I'm against this.
Marlin/Reprap is for all platforms not just one.  
Would have to write new drivers for all platforms.

sl1pkn07, 2019-11-17:
NO

InsanityAutomation, 2019-11-17:
It could be standard per platform (eg lpc17xx, avr, stm32) but too many differences between platforms to be global. Just not feasible.

ejtagle, 2019-12-07:
At some time, i did implement it for Arduino Due, and it worked. But, suddenly, it stopped to work. The microsoft 3d printer driver uses standard Gcode, so, it does not require any change to the gcode processing.. just the addition of some USB descriptors..

LinoBarreca, 2020-01-08:
Can be done for sure on STM, I would not do it though: Too much hassle and too little ROI. I would suggest Microsoft to add a second layer driver on com ports to recognize most common 3d firmware reset messages and be immediately compatible with 99.5% of printers worldwide...

extesy, 2020-01-08:
@LinoBarreca Why do you think adding usb descriptors is too much hassle? If you were to quantify the effort, how long would it take to implement?

LinoBarreca, 2020-01-08:
@extesy it isn't "just a descriptor" to override.
You got to implement the whole CDC (for every family of processors).
If you are wondering "how difficult can it be" download STM32Cube and look at the CDC application (which is a basic "serial port")
And I see no real benefit in doing it since the "driver" for windows doesn't provide any added value now besides letting microsoft know you got a 3d printer and its bed size.

ellensp, 2020-01-08:
Design a interface box.  A real reprap type 3d printer on one side, the crap MS wants on the other side...   Then any printer can be used.  If it provides bed size, then add a gcode to return bed size so the interface can query it.  But I'm not going to do this, waste of time and resources.

LinoBarreca, 2020-01-08:
@ellensp or a second layer driver without any extra hardware.....but I'm not going to do this because it's just useless.

closed 2020-06-30 by github-actions[bot]