MarlinFirmware/Marlin #15915: [BUG] LPC1768 errors in functions `u8g_i2c_***'

Shoorup4eg, 2019-11-17:
### Description

When compiling for a board with a LPC1768 you get errors in functions `u8g_i2c_***'

### Steps to Reproduce

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3855373/Configuration.zip)

#### Additional Information

```jsx

[ Deviot 2.2.1 ] Starting...
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; framework: arduino; board: nxp_lpc1768)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html

PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768

HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash

DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)

PACKAGES: framework-arduino-lpc176x 0.2.2, toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)

Converting Marlin.ino

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF Modes: Finder ~ off, Compatibility ~ strict

Found 6 compatible libraries

Scanning dependencies...

Dependency Graph

|-- <Servo> 1.0.0

|-- <LiquidCrystal> 1.0.0

|-- <U8glib-HAL> 0.4.1

|-- <TMCStepper> 0.5.2

|-- <Adafruit NeoPixel> 1.3.0

|-- <SailfishLCD>

Building in release mode

...

Linking .pio\build\LPC1768\firmware.elf
c:/users/shoorik/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.cpp.o: in function `u8g_i2c_start':

C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.cpp:62: multiple definition of `u8g_i2c_start'; .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.c.o:C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.c:59: first defined here

c:/users/shoorik/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.cpp.o: in function `u8g_i2c_init':

C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.cpp:117: multiple definition of `u8g_i2c_init'; .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.c.o:C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.c:114: first defined here

c:/users/shoorik/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.cpp.o: in function `u8g_i2c_send_byte':

C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.cpp:157: multiple definition of `u8g_i2c_send_byte'; .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.c.o:C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.c:154: first defined here

c:/users/shoorik/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.cpp.o: in function `u8g_i2c_stop':

C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.cpp:169: multiple definition of `u8g_i2c_stop'; .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.c.o:C:\fw\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/LCD_I2C_routines.c:166: first defined here

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1768\firmware.elf] Error 1
```

ellensp, 2019-11-17:
Cannot replicate with your provided files and current marlin bugfix 2.0 
But: the i2c code for the LPC1768 was updated a few days ago...  
Update your copy of marlin to the latest and try again.

Shoorup4eg, 2019-11-17:
sorry my mistake. Unpacked a fresh archive of source codes into an old working folder. As a result, I still have the old Marlin \ src \ HAL \ HAL_LPC1768 \ u8g \ LCD_I2C_routines.c file which prevented compilation

closed 2019-11-17 by Shoorup4eg