MarlinFirmware/Marlin #15916: [BUG ?] Can't pull latest commits from Github

Patag, 2019-11-17:
Since a couple of days, I'm not able to pull the latest commits from Marlin Repo, branch bugfix-2.0.x., as I used to do before. Am I wrong somewhere ?
I'm using VScode/Platformio and when trying to "Pull" from "Source control" menu, I get this error message in the terminal :
```
>git pull --tags origin bugfix-2.0.x
From https://github.com/MarlinFirmware/Marlin
 * branch                bugfix-2.0.x -> FETCH_HEAD
 ! [rejected]            1.1.9      -> 1.1.9  (would clobber existing tag)
```
Any idea ? Please note I've never ever used 1.1.9 branch. 
Thanks in advance and sorry if my issue is not Marlin related.

boelle, 2019-11-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Patag, 2019-11-17:
Is it a joke ? @boelle are able yourself to pull latest commits ? Who maintain the Marlin repo ? I'm suspecting a mismatch in commits labeling or so.
Please, stop playing with the "Close" button !

sjasonsmith, 2019-11-17:
It probably has to do with this: #15881

Patag, 2019-11-17:
Hi Jason, could you explain this to @boelle ? I'd like also @thinkyhead opinion.
Thanks

boelle, 2019-11-17:
> Who maintain the Marlin repo ?

@thinkyhead 

> I'm suspecting a mismatch in commits labeling or so.

Did more look like you asked questions on how to use github, but i was wrong

> Please, stop playing with the "Close" button !

Only if issues are very clear bug or feature requests
the issue list is not for questions and support work

boelle, 2019-11-17:
> are able yourself to pull latest commits ?

yes, 

i have forked marlin and then created a new branch on my fork with settings for my printer
i update the main branch which pulls in the latest commits, and i then rebase my machine branch

in that way getting my machine updated is much faster

sjasonsmith, 2019-11-17:
@Patag, I don't really use any of the tags in Marlin, so I haven't had the same issue you had. Since the tag was changed upstream, you'll probably have to delete your local tag and pull it again. (Just a guess, not a tag expert here).

rekkitatu, 2019-11-17:
You should be able to delete the local tag in terminal `git tag --delete 1.1.9` and after that the pull should work.

Patag, 2019-11-17:
Thanks @rekkitatu , it works. But where does this "1.1.9" come from ? I've never used this branch ?
Did I miss something ? Furthermore, I've 2 clones of the Marlin repo (1 for SKR Pro and 1 for SKR V1.3) and they all behave like this.
@boelle sorry if my initial description wasn't clear enough.
So, Ok, it works now, I close.
Thank you all

closed 2019-11-17 by Patag

boelle, 2019-11-17:
> But where does this "1.1.9" come from ?

maybe @thinkyhead made a small mistake, we are only humans, and i know he does put in far more hours than is healthy.