MarlinFirmware/Marlin #15919: [BUG] (error compile at menu.h)

Ccleorina87, 2019-11-17:
Error compile at menu.h on LPC1768 MKS SGen-L

no matching function for call to 'TMenuEditItem<MenuEditItemInfo_float3>::action(const char* const&, int, float*, int, int)'

call of overloaded 'draw(bool, int8_t&, const char* const&, int, float*,

error started on 15/11/2019 update - Reduced string storage using tokens (#15593) 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3855640/Configuration.txt)

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3855641/Configuration_adv.txt)

atoomnetmarc, 2019-11-18:
I got this exact same error message also. A few hours of compiling with different options got met to this offender:

#define PID_EXTRUSION_SCALING

Try disabling it. Its in Configuration_adv.txt.

Vertabreak, 2019-11-18:
the cause is having 
 #define PID_EDIT_MENU
and 
#define PID_EXTRUSION_SCALING
the edit menu is the root cause encounter this days ago.

Ccleorina87, 2019-11-18:
> I got this exact same error message also. A few hours of compiling with different options got met to this offender:
> 
> #define PID_EXTRUSION_SCALING
> 
> Try disabling it. Its in Configuration_adv.

it fix... thanks

closed 2019-11-24 by boelle