MarlinFirmware/Marlin #15922: [BUG] (Error compiling MMU2)

TB1405, 2019-11-17:
### Description

Compiling with MMU2 isnt possible anymore. 

Download the newest marlin and activate mmu2 support and activate #define MMU2_MENUS  . Succesfully Compiling isnt possible anymore.
Following error message will come up
`In file included from Marlin\src\lcd\menu\menu_mmu2.cpp:29:0:
Marlin\src\lcd\menu\menu_mmu2.cpp: In function 'void menu_mmu2_load_filament()':
Marlin\src\lcd\menu\menu_mmu2.cpp:66:15: error: 'MSG_MMU2_FILAMENT0' is not a member of 'Language_en'
   ACTION_ITEM(MSG_MMU2_FILAMENT0, []{ _mmu2_load_filament(0); });
               ^
Marlin\src\lcd\menu\menu.h:367:24: note: in definition of macro '_MENU_INNER_P'
   PGM_P const plabel = PLABEL;                                \
                        ^~~~~~
Marlin\src\lcd\menu\menu.h:424:54: note: in expansion of macro '_MENU_ITEM_P'
 #define MENU_ITEM_P(TYPE, PLABEL, V...)              _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
                                                      ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:432:55: note: in expansion of macro 'MENU_ITEM_P'
 #define ACTION_ITEM_P(PLABEL, ACTION)                 MENU_ITEM_P(function, PLABEL, ACTION)
                                                       ^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:53: note: in expansion of macro 'ACTION_ITEM_P'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:67: note: in expansion of macro 'GET_TEXT'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                                   ^~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:66:3: note: in expansion of macro 'ACTION_ITEM'
   ACTION_ITEM(MSG_MMU2_FILAMENT0, []{ _mmu2_load_filament(0); });
   ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:66:15: note: suggested alternative: 'MSG_MMU2_FILAMENT_N'
   ACTION_ITEM(MSG_MMU2_FILAMENT0, []{ _mmu2_load_filament(0); });
               ^
Marlin\src\lcd\menu\menu.h:367:24: note: in definition of macro '_MENU_INNER_P'
   PGM_P const plabel = PLABEL;                                \
                        ^~~~~~
Marlin\src\lcd\menu\menu.h:424:54: note: in expansion of macro '_MENU_ITEM_P'
 #define MENU_ITEM_P(TYPE, PLABEL, V...)              _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
                                                      ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:432:55: note: in expansion of macro 'MENU_ITEM_P'
 #define ACTION_ITEM_P(PLABEL, ACTION)                 MENU_ITEM_P(function, PLABEL, ACTION)
                                                       ^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:53: note: in expansion of macro 'ACTION_ITEM_P'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:67: note: in expansion of macro 'GET_TEXT'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                                   ^~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:66:3: note: in expansion of macro 'ACTION_ITEM'
   ACTION_ITEM(MSG_MMU2_FILAMENT0, []{ _mmu2_load_filament(0); });
   ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:67:15: error: 'MSG_MMU2_FILAMENT1' is not a member of 'Language_en'
   ACTION_ITEM(MSG_MMU2_FILAMENT1, []{ _mmu2_load_filament(1); });
               ^
Marlin\src\lcd\menu\menu.h:367:24: note: in definition of macro '_MENU_INNER_P'
   PGM_P const plabel = PLABEL;                                \
                        ^~~~~~
Marlin\src\lcd\menu\menu.h:424:54: note: in expansion of macro '_MENU_ITEM_P'
 #define MENU_ITEM_P(TYPE, PLABEL, V...)              _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
                                                      ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:432:55: note: in expansion of macro 'MENU_ITEM_P'
 #define ACTION_ITEM_P(PLABEL, ACTION)                 MENU_ITEM_P(function, PLABEL, ACTION)
                                                       ^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:53: note: in expansion of macro 'ACTION_ITEM_P'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:67: note: in expansion of macro 'GET_TEXT'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                                   ^~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:67:3: note: in expansion of macro 'ACTION_ITEM'
   ACTION_ITEM(MSG_MMU2_FILAMENT1, []{ _mmu2_load_filament(1); });
[... 141 lines cut ...]
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:433:67: note: in expansion of macro 'GET_TEXT'
 #define ACTION_ITEM(LABEL, ACTION)                  ACTION_ITEM_P(GET_TEXT(LABEL),  ACTION)
                                                                   ^~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:70:3: note: in expansion of macro 'ACTION_ITEM'
   ACTION_ITEM(MSG_MMU2_FILAMENT4, []{ _mmu2_load_filament(4); });
   ^~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\src\src\lcd\menu\menu_mmu2.cpp.o] Error 1`

if you disable #define MMU2_MENUS 
following error message will come up 

`Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M701()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:104:10: error: 'class MMU2' has no member named 'load_filament_to_nozzle'; did you mean 'set_filament_type'?
     mmu2.load_filament_to_nozzle(target_extruder);
          ^~~~~~~~~~~~~~~~~~~~~~~
          set_filament_type
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M702()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:208:10: error: 'class MMU2' has no member named 'unload'
     mmu2.unload();
          ^~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\gcode\feature\pause\M701_M702.cpp.o] Error 1
[ConfigurationH.zip](https://github.com/MarlinFirmware/Marlin/files/3855837/ConfigurationH.zip)
`

GMagician, 2019-11-17:
Please check PR linked above. Not sure it fix and if correct filament numbering. I don't have MMU2 to test

TB1405, 2019-11-17:
made the changes  
this error comes with enabled  #define MMU2_MENUS
`In file included from Marlin\src\lcd\menu\menu_mmu2.cpp:29:0:
Marlin\src\lcd\menu\menu_mmu2.cpp: In lambda function:
Marlin\src\lcd\menu\menu_mmu2.cpp:77:45: error: '_mmu2_eject_filament' was not declared in this scope
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
                                             ^
Marlin\src\lcd\menu\menu.h:370:39: note: in definition of macro '_MENU_INNER_P'
     MenuItem_##TYPE::action(plabel, ##V);                     \
                                       ^
Marlin\src\lcd\menu\menu.h:427:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:435:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define ACTION_ITEM_N_P(N, PLABEL, ACTION)          MENU_ITEM_N_P(function, N, PLABEL, ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:436:51: note: in expansion of macro 'ACTION_ITEM_N_P'
 #define ACTION_ITEM_N(N, LABEL, ACTION)           ACTION_ITEM_N_P(N, GET_TEXT(LABEL), ACTION)
                                                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:77:3: note: in expansion of macro 'ACTION_ITEM_N'
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
   ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:77:45: note: suggested alternative: '_mmu2_load_filament'
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
                                             ^
Marlin\src\lcd\menu\menu.h:370:39: note: in definition of macro '_MENU_INNER_P'
     MenuItem_##TYPE::action(plabel, ##V);                     \
                                       ^
Marlin\src\lcd\menu\menu.h:427:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:435:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define ACTION_ITEM_N_P(N, PLABEL, ACTION)          MENU_ITEM_N_P(function, N, PLABEL, ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:436:51: note: in expansion of macro 'ACTION_ITEM_N_P'
 #define ACTION_ITEM_N(N, LABEL, ACTION)           ACTION_ITEM_N_P(N, GET_TEXT(LABEL), ACTION)
                                                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:77:3: note: in expansion of macro 'ACTION_ITEM_N'
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
   ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp: In lambda function:
Marlin\src\lcd\menu\menu_mmu2.cpp:77:45: error: '_mmu2_eject_filament' was not declared in this scope
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
                                             ^
Marlin\src\lcd\menu\menu.h:375:58: note: in definition of macro '_MENU_INNER_P'
       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);  \
                                                          ^
Marlin\src\lcd\menu\menu.h:427:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:435:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define ACTION_ITEM_N_P(N, PLABEL, ACTION)          MENU_ITEM_N_P(function, N, PLABEL, ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:436:51: note: in expansion of macro 'ACTION_ITEM_N_P'
 #define ACTION_ITEM_N(N, LABEL, ACTION)           ACTION_ITEM_N_P(N, GET_TEXT(LABEL), ACTION)
                                                   ^~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:77:3: note: in expansion of macro 'ACTION_ITEM_N'
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
   ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_mmu2.cpp:77:45: note: suggested alternative: '_mmu2_load_filament'
   ACTION_ITEM_N(1, MSG_MMU2_FILAMENT_N, []{ _mmu2_eject_filament(0); });
                                             ^
Marlin\src\lcd\menu\menu.h:375:58: note: in definition of macro '_MENU_INNER_P'
       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);  \
                                                          ^
Marlin\src\lcd\menu\menu.h:427:52: note: in expansion of macro '_MENU_ITEM_N_P'
 #define MENU_ITEM_N_P(TYPE, N, PLABEL, V...)       _MENU_ITEM_N_P(TYPE, N, false, PLABEL, ##V)
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:435:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define ACTION_ITEM_N_P(N, PLABEL, ACTION)          MENU_ITEM_N_P(function, N, PLABEL, ACTION)
[... 108 lines cut ...]
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:435:53: note: in expansion of macro 'MENU_ITEM_N_P'
 #define ACTION_ITEM_N_P(N, PLABEL, ACTION)          MENU_ITEM_N_P(function, N, PLABEL, ACTION)
                                                     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:436:51: note: in expansion of macro 'ACTION_ITEM_N_P'
[... cut ...]
Marlin\src\gcode\feature\pause\M701_M702.cpp:104:10: error: 'class MMU2' has no member named 'load_filament_to_nozzle'; did you mean 'set_filament_type'?
     mmu2.load_filament_to_nozzle(target_extruder);
          ^~~~~~~~~~~~~~~~~~~~~~~
          set_filament_type
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M702()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:208:10: error: 'class MMU2' has no member named 'unload'
     mmu2.unload();
          ^~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\gcode\feature\pause\M701_M702.cpp.o] Error 1`

GMagician, 2019-11-17:
Can you post your config files? I compiled it defining MMU2 and its menus and compiled fine

TB1405, 2019-11-17:
The problem was my config. I had enabled FILAMENT_LOAD_UNLOAD_GCODE  in the conf.adv without this it compiles succesfully also without any changes. 
Maybe this function is not compatibel with the mmu2 feature

GMagician, 2019-11-17:
MSG_MMU2_FILAMENT0 shouldn't exists....sure you made all modifies I made in PR?
Please post your config files

GMagician, 2019-11-17:
I checked my PR. In line 65 there is `MSG_MMU2_FILAMENT_N,` not `MSG_MMU2_FILAMENT0`.
Please be sure to download poste file.

TB1405, 2019-11-17:
alright i checked the changes from you again and now it compiles succesfully with enabled  #define MMU2_MENUS 
I will try it tomorrow with my mmu2
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3856094/Configuration_adv.zip)

InsanityAutomation, 2019-11-17:
that has an issue that breaks without a spi display as well. Its been on my list to adjust that function a bit to rectify some of that.

TB1405, 2019-11-17:
Couldnt try it. I think there is a problem with TMC  Drivers libary the Y Axis doesnt work.

TB1405, 2019-11-18:
I have now loaded the changed version to my printer.
The problem is the LCD menu now shows filament 2-6 instead of 0-5. 
But it sems it works.

GMagician, 2019-11-18:
> The problem is the LCD menu now shows filament 2-6 instead of 0-5

Now it should be fixed, numbering from 1 to 5

TB1405, 2019-11-18:
Is it possible to download the complete file that you have changed?

GMagician, 2019-11-18:
I think its possible but I don't really what's better to do.

I think you may go to my repository and download as zip and then overwrite your sources but preserving your configuration files or else "edit" changed filed and copy & paste on your source (open in an editor).

GMagician, 2019-11-20:
@TB1405 PR has been merged. This may be closed now

TB1405, 2019-11-20:
Alright i will check it in the next Time

TB1405, 2019-11-20:
It is right now! Thank you

PR #15923 (2019-11-20): Fix MMU2 compile error
Should fix #15922
closed 2019-11-20 by TB1405