MarlinFirmware/Marlin #15926: Extruder won't move after updating Marlin 2.0 (SKR 1.3 w/ 2208 v3.0 steppers)

NickTom88, 2019-11-18:
I installed my SKR 1.3 w/ 2208 v3.0 setup in my Ender 3 Pro back when 2.0 bugfix was having vibration issues with Classic Jerk and Junction Deviation. I was able to finally upload a version that worked successfully but it still didn't seem to print 100% correct. After about 3 weeks of what I would call "mediocre" prints I decided to check my settings and also check the most recent version of 2.0 bugfix. I settings for the stepper drivers/motors didn't change between the 2 version, but now my extruder motor won't move. If I disable Stealthchop I can feel it energize when I try extruding via Octoprint, but it work extrude or retract. I've connected the motor/cable to a different axis and everything worked fine. I've also swapped the 2208's around and it didn't change anything. 

To make things worse, it appears that now no matter what version I try and flash to, they all have the same problem... Today I received a new SKR 1.3 board too and also downloaded the freshest 2.0 thinking that might be the issue - nope. Same thing. I hope I'm not doing something stupid, but I am at a loss and can't seem to get any answers anywhere else...

### Steps to Reproduce
1. Use SKR 1.3 board with TMC2208 v3.0 stepper drivers
2. Configure settings to recommended - I used Teaching Tech's video. This worked when I initially installed my upgraded board/drivers. Everything was done in Atom.
      - NOTE: I know that the "Prevent Cold Extrusion" is disabled.  I did this to test a separate motor 
                   for movement.
3. Test extruder motor through your favorite tool. I tried used Pronterface, Octoprint, and even the printer controls. M122 in Pronterface shows everything looking good.

**Expected behavior:** PLA squirts out and I'm a happy person

**Actual behavior:** Nothing. The extruder motor energizes but doesn't move. 

[2.0_bugfix_NT.zip](https://github.com/MarlinFirmware/Marlin/files/3856327/2.0_bugfix_NT.zip)

Thanks in advance.

psavva, 2019-11-18:
This is most probably caused by Lin advance and the TMC drivers.

Set MINIMUM_STEPPER_PULSE to 1 or disable LIN Advance.

If you upgrade to the latest version of Making 2.x, your issue will resolved using the same config settings...  There was a pr to set minimum stepper Pulse as a conditional.

Setting SQUARE_WAVE_STEPPING is the recommended solution and will also fix the problem, assuming you are facing the same

NickTom88, 2019-11-18:
> This is most probably caused by Lin advance and the TMC drivers.
> Set MINIMUM_STEPPER_PULSE to 1 or disable LIN Advance.
> If you upgrade to the latest version of Making 2.x, your issue will resolved [...]

@psavva So this setup can't use linear advance? Or could linear advance in the F/W just be set to 0 and then changed in the slicer? This is odd because it seemed to work immediately after upgrading. 

Thanks for your advice either way. I'll try these options after work.

psavva, 2019-11-18:
You CAN use Linear Advance.
You just need to either set SQUARE_WAVE_STEPPING or MINIMUM_STEPPER_PULSE to 1 for it to work.

In the latest version, a conditional was added.

NickTom88, 2019-11-19:
UPDATE: I changed the settings to what @psavva has recommended and it appeared to work at first, but after the first 5 layers of an 20mm XYZ cube the extruder stopped again. I ran out of time to tinker with it so I will try to work it a little more and add the updated Config files this evening.

NickTom88, 2019-11-20:
Alright... So with Square Wave Stepping disabled and Min Stepper Pulse set to 1, the extruder will work ONLY when i manually extrude. Using octoprint I can extrude filament just fine, but when I begin to print a job it won't extrude anything... 

@psavva are these config updates only to the generic config & config_adv files or are all examples updated as well? This may seem like a dumb question, but none of these issues are making any sense to me...
[2.0 Bugfix v2.zip](https://github.com/MarlinFirmware/Marlin/files/3867018/2.0.Bugfix.v2.zip)

psavva, 2019-11-20:
Try to disable Linear Advance.  Do you face the same issues, or does everything work as expected when you comment out the Lin advance define.

It's not enough to set k=0.  You need y actually fully disable and recompile.

NickTom88, 2019-11-20:
@psavva it looks like Linear Advance is the issue. I've disabled the Min Step Pulse and Square Wave Step as well for this compilation. So far it's printing the xyz cube.

Guess I need to rename the title of this bug?

psavva, 2019-11-20:
LIN Advance should be working without issue given that you've either enabled  SQUARE_WAVE_STEPPING or set MINIMUM_STEPPER_PULSE to 1 or higher.

@teemuatlut   I think we may have a new bug?

@NickTom88 Can you also try Enabling LIN ADVANCE and set MINIMUM_STEPPER_PULSE to 10 and test it this config will work as a workaround if you want to use LIN ADVANCE.  

A title update seems to be warranted.

@thinkyhead can you add the label for a potential bug.

NickTom88, 2019-11-20:
@psavva I'll do this either in the morning or after work. Will keep you posted.

NickTom88, 2019-11-21:
UPDATE:
First off, I think I may have a new bug... I ended up using Stop Print due to how I'm printing this test piece; the nozzle lifted, homed itself, and then resumed printing where it left off. I had to turn off the printer in order to stop it.

Onto the other results:
- Using LIN ADVANCE with SQUARE_WAVE_STEPPING the print made it to approx. 16 layers and then stopped. It appeared to print fine until then.

- Using LIN ADVANCE with MINIMUM_STEPPER_PULSE at 10 didn't print at all. 

- No LIN ADVANCE works just fine though. But I'm definitely sacrificing print quality. My first successful cube has some horrible ghosted corners.

randellhodges, 2019-11-21:
Are you using Octoprint? If so, you need to enable action commands, otherwise if you stop the print from the printer, Octoprint won't know and it'll keep sending gcode to the printer.

NickTom88, 2019-11-21:
@randellhodges this was actually while connected directly to the printer after flashing fw, which I don't normally do. I usually just use the Cancel command in octoprint. Maybe Cura 4.4.0 has a glitch.

NickTom88, 2019-11-22:
Today's Update: @psavva @teemuatlut @thinkyhead @Sebastianv650

I decided to keep messing with the settings and find something that would work, because I can't stand the quality without LIN ADVANCE. I'm currently running a configuration that seems to do the job. 
The following are the changes I've made
#define CLASSIC_JERK
#define LIN_ADVANCE
#define MINIMUM_STEPPER_POST_DIR_DELAY 20
#define MINIMUM_STEPPER_PRE_DIR_DELAY 20
#define MINIMUM_STEPPER_PULSE 2
#define MAXIMUM_STEPPER_RATE 400000
DISABLED Settings
#define STEALTHCHOP_E
[2.0 Bugfix v3.zip](https://github.com/MarlinFirmware/Marlin/files/3880753/2.0.Bugfix.v3.zip)

NickTom88, 2019-11-23:
Continued testing current settings... Currently everything is still working, but the extruder motor is operating loud, clunky, and hot. My next test will be to run it with MIN STEPPER PULSE at 0 and STEALTHCHOP disabled for the extruder.

I will confidently say that with the current firmware, the extruder cannot operate with STEALTHCHOP enabled. It will get roughly 15 layers in and stop working. I've tested this multiple times with different prints. While I don't think the prints are effected by this, I think it's putting stress on the motor.

Strahler85, 2019-11-25:
That's are already known fact, that tmc2208 at extruders in stealthchop is not a good choice, because the tmc2208 regonize a overcurrent (happens when you have a lot of retract) and stops working. In normal mode it is better but not perfect. Sometimes my extruders stops in a 24h print. My solution was to change the Extruder driver to a lv8729 and now I don't have issues anymore.

NickTom88, 2019-11-25:
@Strahler85 if it's a known fact can you show me where this is pointed out? Maybe I've overlooked it, but I haven't seen any documentation on this from any forum, instructions, YoutTube video, or any release notes. If this is a "tribal knowledge" situation, then maybe there should be a note someone in the Config file to warn the user. Like I said, this originally wasn't and issue until I updated the firmware to try and correct the vibration issues from Jerk/JD.

Using the LV8729 driver how it the motor noise? Right now my motor sounds extremely clunky and gets pretty warm.

Strahler85, 2019-11-25:
If you print slow without LA and with Cura which avoid a lot of retracts it works most of the time. I only know german forum for 3d printing and there they recommand not using a TMC2208 as extruder driver and if you use a TMC2208, use it in spreadcycle.
LV8729 is not as quiet as the TMC2208 in stealthchop but the motor stays cold.

I would recommand to put the TMC2208 in spreadcycle for the extruder and look if you have any issue with it or use a different extruder driver (LV8729 or DRV8825 instead of a TMC2208)

boelle, 2019-12-07:
@NickTom88 2.0 was just released a few days ago, did that change anything at all?

NickTom88, 2019-12-07:
@boelle Just downloaded and compiled it. I did it after it was released but for S&G's I just did it. I've also changed out my E0 stepper driver to TMC2209 to push a little more juice. Everything seems to be working with the attached configurations.

TLDR Version
- LA Enabled
- Square Wave Stepping Enabled
- MIN STEPPER PULSE disabled.

Everything seems to be working quiet and smooth. Although My prints still seem to have some inconsistent little blobs that were never there before... I'm not going to point the finger at FW yet though.

rado79, 2019-12-09:
I got the same problem with same hardware. 2.0 Bugfix downloaded yesterday.
First i thought the extruder doesn´t move, but it moves so slowly that you would not recognize it at first look. But you can hear moving, and it will move some degrees in long time (a few minutes).

LA enabled
Square Wave disabled
E spreadcycle
minimum stepper pulse 0/1/2

before i used a 3 month old 2.0 bugfix, working good

rado79, 2019-12-09:
i try a lot of settings by enabling and changing parameter at this:

minimum stepper dir delay
maximum stepper rate
min steps per segment
maximum stepper rate

nothing changes.

rado79, 2019-12-09:
the only thing wich works for me:

#define SQUARE_WAVE_STEPPING

square wave stepping was not included in my older 2.0 bugfix, and the new bugfix (downloaded yesterday) does only work enabling this.

does work
#define LIN_ADVANCE

others got back to disabled/default
//#define MINIMUM_STEPPER_POST_DIR_DELAY 20
//#define MINIMUM_STEPPER_PRE_DIR_DELAY 20
//#define MINIMUM_STEPPER_PULSE 2
//#define MAXIMUM_STEPPER_RATE 400000
//#define STEALTHCHOP_E

NickTom88, 2019-12-09:
@rado79 that's what happened to me as well. Ultimately I ended up buying a 2209 driver for my E0 so that I could turn Stealthchop back on and keep the extruder motor cooler and quieter.

rado79, 2019-12-09:
there is no need to enable stealthchop on E for me. 950mA in spreadcycle is more than enough torque, and there is no noise from extruder incuding linear advanced at my setup. stepper motor gets around 40°C in long jobs.

btw. I changed today the X and Y stepper motor (from stock 1.8°) to 0.9° from OMC (17HM19-2004S), they are noisier than the stock 1.8° in stealthchop, and louder than the E in spreadcycle. but i don´t know why. before swap X and Y was very quiet, now you hear whining.
is there anything to do in config ? maybe CHOPPER_TIMING CHOPPER_DEFAULT_24V ?

boelle, 2019-12-11:
so this was solved or is there still a gremlin left?

workinghard, 2019-12-14:
#define SQUARE_WAVE_STEPPING 
Did the trick for me (SKR V1.3 + TMC2208 + linear advance)

NickTom88, 2019-12-16:
@boelle I'm gonna go ahead and say it's closed. Looks like the protocols are in place for whoever is trying to set this up to work.

closed 2019-12-16 by NickTom88

rado79, 2019-12-25:
With new Marlin 2.0.1 extruder works with 
#define SQUARE_WAVE_STEPPING and with 
//#define SQUARE_WAVE_STEPPING, no more problem for me - SKR 1.3 TMC2208/2209

NickTom88, 2019-12-28:
@rado79 did you notice any difference in quality or performance having it on or off?

rado79, 2020-01-06:
> @rado79 did you notice any difference in quality or performance having it on or off?

there is no visible quality difference for me, but stepper moving sounds like lower frequency/more muted with uncommented #define SQUARE_WAVE_STEPPING.

sjasonsmith, 2020-01-06:
> did you notice any difference in quality or performance having it on or off?

There is probably not currently a difference when enabling or disabling SQUARE_WAVE_STEPPING. Eventually SQUARE_WAVE_STEPPING will allow some optimization to reduce time wasted in the interrupt routines, but that is not yet implemented.

rado79, 2020-01-06:
maybe i have mixed something up with ADAPTIVE_STEP_SMOOTHING in my perception?