MarlinFirmware/Marlin #15927: [BUG] [bugfix 2.0-UBL] Button not working for pause on stow/deploy with G29 P1 (same with #13114 #13705)[BUG]

timeburn, 2019-11-18:
Having PAUSE_BEFORE_DEPLOY_STOW enabled along with UBL causes UBL to hang at "Deploy Z-Probe", and not respond to the button.

Deploy/Stow works correctly for Delta Calibration, in the same build.
If i switch the build to use ABL, it works normally.

This is on an SKR 1.3

sjasonsmith, 2019-11-18:
I can confirm this is an issue, at least when using a delta printer. I discovered it when I was preparing PR #15887 and was unable to test this form of probing with my changes.

Clickable links to the other two issues which were closed without fixing: #13114 #13485

sjasonsmith, 2019-11-18:
I found the cause of the issues and posted a PR. Hopefully it will be resolved soon.

sjasonsmith, 2019-11-21:
@timeburn, can you try this with the latest state of bugfix-2.0.x, and close the issue if it is fixed?

timeburn, 2019-11-22:
Tested with the latest off git today, it's working correctly now.

closed 2019-11-22 by timeburn