MarlinFirmware/Marlin #15932: [FR] Need a Small Filament Retract Before Calling resume_print()

joenelis, 2019-11-18:
**Situation:**  I'm printing a Front Panel for an enclosure using an 2 month old Ender 3 which is running the out-of-the-box Creality v1.1.6.  I have Raised Lettering on this Front Panel.   I'm doing a Filament Change to a contrasting Filament Color right before the Layers which print this Raised Lettering.   I run through the entire M600 triggered sequence, including doing one or more Extrudes to insure that the Nozzle is ejecting the new color.  I pull that ejected tail off of the Nozzle and select 'Resume Printing'.    So far, all is fine.

**Problem:**  Upon selecting 'Resume Printing', as expected the Nozzle is returned to the position where it was when interrupted.  BUT, the Nozzle has continued to ooze Filament during the travel time and the first Raised Letter that is printed has too much Filament - the letter is deformed.   Here is what it looks like (as sliced by Cura, the first Raised Letter for this job just happens to be is the 'P' in GPS).    
[image]

Whenever I Manually pulled the Filament back about 5mm at the Extruder BEFORE I pressed  'Resume Printing', I no longer have this problem  By doing that Manual retraction I took the pressure off and the Nozzle did not ooze during the travel back to the position where it was interrupted.   Here is what I got when I did that Manual Filament Retraction Step:
[image]

**Suggested Solution:**  Looking at the M600 processing in Marlin_main.cpp, near the end of that M600 function, it looks to me like what is needed is an instruction added in after _wait_for_filament_reload()_ and before _resume_print()_ to do an automatic 5mm Filament retract.   Might be as easy as easy as a call to M702 (but it was not obvious where I would have first store the 5mm - I did not study it long enough to figure that out).

**Comment:**  There can always be be some corner case where doing that hard coded automatic 5mm retraction could somehow affect the quality of the print job.   However, it seems to me that if it does not adversely impact the Raised Lettering on my Panel, it would take a real edge case for it to show up.

Joe Nelis
Pittsburgh, PA

joenelis, 2020-06-24:
This remains an issue and should not be closed.  Without the suggested step to add in the automated 5mm filament retract, it will trash the print job that is doing text in a new color, as the  example shows.   It probably needs to have a higher priority than I initially gave it.
Joe Nelis
Pittsburgh PA

MoellerDi, 2020-06-24:
Did you test with the latest bugfix-2.0.x ?

Retract to prevent oozing before going back to resume_position is at:
https://github.com/MarlinFirmware/Marlin/blob/c135db1ce7b8274a014cf9108f569b6cb442aaaf/Marlin/src/feature/pause.cpp#L579
it seems it's already implemented. See PAUSE_PARK_RETRACT_LENGTH

What else is missing?

joenelis, 2020-06-24:
Hello MoellerDi,

No, I did not - I missed seeing this.  I will test and close the issues if that fixes it.

Thank You

Joe Nelis
 

Wednesday, June 24, 2020, 4:40:02 PM, you wrote:

Did you test with the latest bugfix-2.0.x ?
Retract to prevent oozing before going back to resume_position is at:
https://github.com/MarlinFirmware/Marlin/blob/c135db1ce7b8274a014cf9108f569b6cb442aaaf/Marlin/src/feature/pause.cpp#L579
it seems it's already implemented. See PAUSE_PARK_RETRACT_LENGTH
What else is missing?
—

This email has been checked for viruses by AVG antivirus software. 
www.avg.com

MoellerDi, 2020-06-25:
IMHO it was implemented with PR #17625

joenelis, 2020-06-30:
The simple one-line fix shown in this development branch appears to be what is needed:
**unscaled_e_move(-(PAUSE_PARK_RETRACT_LENGTH), feedRate_t(PAUSE_PARK_RETRACT_FEEDRATE));** 
REFERENCE: https://github.com/MarlinFirmware/Marlin/blob/c135db1ce7b8274a014cf9108f569b6cb442aaaf/Marlin/src/feature/pause.cpp#L579

However, whenever this solution (#17625) was merged in, it was wrapped in a commented out conditional compile (//#define FWRETRACT in configuration_adv.h).  So ‘out of the box’, this solution will never be executed.

If one uncomments the //#define FWRETRACT in configuration_adv.h, that adds an avalanche of code to the executable, and it will no longer fit into the limited Ender 3 memory space.

All that is really needed is for Line 625 in the production version of pause.cpp to execute unconstrained.  All of the time.  Just as it did in the development branch.  However I found that the retract length and feedrate parameters were also buried in the #define FWRETRACT conditional compile.

I simply modified my pause.cpp, adding in my own custom #defines for the Retract-Length value and the Retract-Feedrate-mm-s value.  Then I added the simple unconstrained unscaled_e_move() call below the stuff on lines 621-626 that was not executing.   That added 48 bytes to the executable.   It fit into the Ender 3, and it got the job done.    

**IMHO** - when dealing with hardware with limited memory space, having conditional compile #defines (like this #define FWRETRACT) that introduce a large amount of swiss army knife code to the executable is not a real good plan.  Yes, it makes nice modular code.  But whenever you run out of memory, that is no longer so nice.  In this case, once the conditional compile for the Change Filament Feature was activated, only this simple one-line fix was needed to get rid of the oozing.  It should not have been further constrained.  My solution was admittedly quick and dirty, just to get the job done. And I will have to maintain it during future releases.  However it would be a simple matter to revisit this and get the retract parameters needed here out of that FWRETRACT conditional compile and do the simple unconstrained unscaled_e_move() call in pause.cpp.  Just as the author of the fix for #17625 envisioned.  Going forward, the development team should take memory usage into consideration.

I will close this issue, however with reservations.

Joe Nelis
Software by Nelis
Pittsburgh PA  USA

closed 2020-06-30 by joenelis