MarlinFirmware/Marlin #15933: UBL - G29 P1 stops at point 57 (home)

JudgeZ, 2019-11-18:
**Description**:
I have updated to the latest version of Marlin. When I run the G29 P1 command on a freshly imaged board, the command stops at point 57/100. This point happens to coincide exactly where the device homes. Afterwards, the board is unresponsive to any G29 commands. The probe simply does not move once it moves above point 57 nor deploys.

**Board**:
SKR Mini E3 v1.2

**Commands**:
M502 
M500
G28
G29 P1

**Troubleshooting**:
Turning on Debug
Decreasing Z_PROBE_LOW_POINT to -10
Not running G28, M500, M502, etc
Changing my offsets
Changing to different pins (zstop/probe)
Turning on and off ENDSTOP_INTERRUPTS_FEATURE
Promising the printer upgrades if it's acts nice
Threatening to replace the printer

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3856968/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3856969/Configuration_adv.h.txt)
[platformio.ini.txt](https://github.com/MarlinFirmware/Marlin/files/3856971/platformio.ini.txt)
[pins_BIGTREE_SKR_MINI_E3.h.txt](https://github.com/MarlinFirmware/Marlin/files/3856973/pins_BIGTREE_SKR_MINI_E3.h.txt)

Roxy-3D, 2019-11-18:
I'll look at this more carefully tomorrow.    But I think we are going to need to add some debugging lines to print out some values.    It will take a couple of iterations, but we should be able to figure out what is broke.  

UBL logic hasn't changed in almost a year.   But a lot of the code clean up is changing things used by UBL.   And we need to figure out what got broke.

tatusah, 2019-11-18:
Did you try lowering the number of total probe points under this number where it stops? Change GRID_MAX_POINTS_X to 7 so you would get total probe points to 49 and see whether the probing can finish then. http://marlinfw.org/docs/features/unified_bed_leveling.html

tatusah, 2019-11-18:
Also by looking at the configuration.h file it seems that the MIN_PROBE_EDGE 10 and MESH_INSET 1 so the probe will actually never go through all the points and the rest of the points (points behind the 10 mm probe edge margin) are supposed to be set manually. Were you aware of this and was this intentionally?I'd recommend additional troubleshooting with lowering max grid points for UBL and deciding how you want the UBL actually behave and setting safe probe edge and mesh inset accordingly making sure that neither the nozzle or probe hits possible bed attachment clips or any other obstacles.

JudgeZ, 2019-11-18:
I tried your suggestion of lowering it to 7 points. It now hangs on point 31/49 which is again in the homing position. I understand that my probe cannot hit all the points and that there will need to be some intervention by me. I had planned on g29 p3 to fill in the gaps as I have no plans at the moment to print out that far. The system is now unresponsive and hung at point 31 (probe home).

tatusah, 2019-11-18:
OK, thanks, at least it confirms that it seems to be repeatable. I had also issues with the UBL on the Ender 3 & SKR Mini E3 v1.2 before changing to the firmware build with 512K memory mod. Based on the configuration it seems that you also had to disable some features to fit the UBL feature to the default memory. Sorry that I can't help more in this matter. Hopefully others can be of a more help. I guess, you can run with AUTO_BED_LEVELING_BILINEAR until this gets solved.

JudgeZ, 2019-11-18:
my current plan is to upgrade to an skr v1.3. there have been too many issues with this board

Roxy-3D, 2019-11-18:
Does it stop and hang?    Or does it just not complete the probe sequence?

Can you do a G29 T ?   If so...  Can you post it here?

JudgeZ, 2019-11-18:
I'll have to recompile with UBL tonight (I needed to replace a part that was held together with super glue asap...). You can do a G29 T while the probing is hung and see that it hasnt completed all the probes. The printer itself doesnt hang or enter a halted state. The probing stops over home and then never deploys or probes. I have to restart the printer to probe again.

Roxy-3D, 2019-11-18:
OK.   That is good to know.   If the firmware hung...  that is a different type of problem.

I would be interested in seeing the final G29 T from the G29 P1.    And depending upon what it looks like...   We can go down different paths to find the problem.

JudgeZ, 2019-11-19:
I have attached a longer output, but here is the 7x7 mesh which stopped at 31 again; however, I was able to run a g29 p3 afterwards which i could not do before. Here is the g29 T. The probe should have been able to reach at least everything outside of row 6 (possibly 5). A 10x10 mesh looks much more sparse but a similar rectangle.

Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,220)                                      (220,220)
Recv:         0       1       2       3       4       5       6
Recv:  6 |   .       .       .       .       .       .       .
Recv:    |
Recv:  5 | -0.130  -0.009  +0.001  +0.002  -0.074    .       .
Recv:    |
Recv:  4 | -0.047  -0.013  +0.004  +0.005  -0.044    .       .
Recv:    |
Recv:  3 | -0.046  +0.001  +0.002 [-0.034] +0.002    .       .
Recv:    |
Recv:  2 | -0.076  -0.026  -0.023  -0.041  -0.023    .       .
Recv:    |
Recv:  1 | -0.104  -0.076  -0.044  -0.046  -0.024    .       .
Recv:    |
Recv:  0 | -0.136  -0.082  -0.069  -0.064  -0.046    .       .
Recv:         0       1       2       3       4       5       6
Recv:     ( 10, 10)                                      (220, 10)
Recv: 
Recv: ok

[terminal g29 p1 t and g29 t.txt](https://github.com/MarlinFirmware/Marlin/files/3861278/terminal.g29.p1.t.and.g29.t.txt)

Roxy-3D, 2019-11-19:
That looks fairly 'normal'.    It doesn't think the probe can get to the top row of mesh points.
And it doesn't believe the probe can get to the right hand two columns of mesh points.

There may be a bug on how insets are getting handled....

ManuelMcLure, 2019-11-19:
With ``MIN_PROBE_EDGE`` set to 10 and ``MESH_INSET`` set to 1 I'd expect none of the outer row/column to be probed. At least that's what I see on my setup (at least last time I probed my bed - it's been a while).

Roxy-3D, 2019-11-19:
Let's turn on UBL_DEVEL_DEBUGGING in ubl.h      And then do a G29 W to see what UBL thinks about the world.

JudgeZ, 2019-11-19:
I think Manuel is right that it shouldnt be able to hit any of the edge points with the Mesh inset smaller than the probe inset.

Send: G29 W
Recv: Unified Bed Leveling System v1.01 inactive
Recv: No Mesh Loaded.
Recv: Fade Height M420 Z0.0000
Recv: # of samples: 100
Recv: Mean Mesh Height: 0.000000
Recv: Standard Deviation: 0.000000
Recv: Probe Offset M851 Z-0.0800000
Recv: MESH_MIN_X  (_MAX((((230) / 2) - ((230) / 2)) + 1, 0))=1
Recv: MESH_MIN_Y  (_MAX((((230) / 2) - ((230) / 2)) + 1, 0))=1
Recv: MESH_MAX_X  (_MIN(((((230) / 2) - ((230) / 2)) + 230) - (1), 230))=229
Recv: MESH_MAX_Y  (_MIN(((((230) / 2) - ((230) / 2)) + 230) - (1), 230))=229
Recv: GRID_MAX_POINTS_X  10
Recv: GRID_MAX_POINTS_Y  10
Recv: MESH_X_DIST  25.33
Recv: MESH_Y_DIST  25.33
Recv: X-Axis Mesh Points at: 1.000  26.333  51.667  77.000  102.333  127.667  153.000  178.333  203.667  229.000
Recv: Y-Axis Mesh Points at: 1.000  26.333  51.667  77.000  102.333  127.667  153.000  178.333  203.667  229.000
Recv: 
Recv: ubl_state_at_invocation :0
Recv: ubl_state_recursion_chk :0
Recv: Meshes go from 0x000002E0 to 0x0000077F
Recv: sizeof(ubl) :  1
Recv: 
Recv: z_value[][] size: 400
Recv: 
Recv: EEPROM free for UBL: 0x0000049F
Recv: EEPROM can hold 2 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.

JudgeZ, 2019-11-19:
Either way I can at least enter manual entries now so I am happy.

JudgeZ, 2019-11-19:
I can fill in the gaps in the mesh now, and UBL isn't freezing anymore. I can only guess that I must have messed something up somewhere. UBL is amazing.

ReubenBTalbott, 2020-01-16:
Had the same problem but disabling ABL_BILINEAR_SUBDIVISION which I had enabled fixed it for me.

Keltere, 2020-02-23:
I'm currently having the same exact issue but no luck on resolving it.

MTrab, 2020-03-09:
I have the excact same but, stopping at probe no 57.
BTT SKR Pro v1.1 with TMC2208 UART drivers.

onek24, 2020-03-15:
Exact same issue too, stops at probe point 57. Ender 3 with SKR Mini E3 and TMC2209.

Gadjet5, 2020-03-26:
Same issue for me SKR 1.4 turbo TMC2209 + 5v bltouch

lakeroe, 2020-04-15:
Same issue for me SKR 1.3 + TMC2209 + inductive proximity sensor

thinkyhead, 2020-04-16:
Whenever there are homing or leveling issues, we ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machines.

lakeroe, 2020-04-16:
Please find a LOG attached.
It's generated exactly as you described, except I didn't use the bugfix firmware but 2.0.5.3 instead.
If you need I post another LOG using the bugfix firmware ...
[LOG.txt](https://github.com/MarlinFirmware/Marlin/files/4486200/LOG.txt)

c2v4, 2020-04-18:
> Had the same problem but disabling ABL_BILINEAR_SUBDIVISION which I had enabled fixed it for me.

You know that ABL_BILINEAR_SUBDIVISION is only for LINEAR and BILINEAR leveling right? This error is affecting UBL

lakeroe, 2020-04-18:
For the sake of completeness:
I just tried the bugfix-2.0.x firmware and it doesn't work either.
It stops after probing 65 of 100 points.
LOG is attached: [LOG.txt](https://github.com/MarlinFirmware/Marlin/files/4496569/LOG.txt)

lakeroe, 2020-04-18:
Just for curiosity I did another test and replaced the Z-stepper driver from TMC2209 to DRV8825.
It doesn't make any difference ...

lakeroe, 2020-04-28:
At least my issue turned out as a wrong configuration (probably probing point out of mesh area).
It works like a charm now ...

rado79, 2020-05-18:
My configuration is set to  UBL 10x10 Grid, `MESH_INSET` 15, the probing stops at 90 of 100 points.
It did not reach the points in the front row nearest the front edge (Ymin is front) of the bed. 
Changing `MESH_MIN_Y` MESH_INSET to (MESH_INSET + 5) does the job - 100 of 100 points.

Why does +5 to MESH_INSET (15+5) allow to reach the points? My first try was (MESH_INSET - 5) wich ended on 80 of 100 points, first and second row missing. In wich direction shifts MESH_INSET when add or subtract 5 in relation to real edge.

Roxy-3D, 2020-05-18:
With a Mesh_Inset, you are limiting how much of the bed is available to print on.   The root of the problem is the nozzle and the Z-Probe are offset from each other.     Do you want to be able to probe the 'entire' bed, or do you want to be able to print on the entire bed?   The choice is yours....

If to much of the bed is available to print on, the Z-Probe can not reach all the mesh locations and you will have to take some additional actions to fill in those mesh values.

rado79, 2020-05-18:
> With a Mesh_Inset, you are limiting how much of the bed is available to print on.

Okay, confusing me, because it is listed under Unified Bed Leveling // Set Mesh bounds as an inset region of the bed.

What i do not understand:
NOZZLE_TO_PROBE_OFFSET { 0, 40, -1.60 } probe is behind the nozzle.
Probing with MESH_INSET 15, the nozzle stays on bed near Ymin. Missing front row points after G29 P1.
Probing with #define MESH_MIN_Y (MESH_INSET + 5), nozzle goes off the bed to Y-20 position, probe is at Y20. All points reached.

Roxy-3D, 2020-05-19:
The original purpose of the Mesh_Inset came from the original Mesh Bed Leveling system.   The intent was to allow people to put clamps on the bed to hold the glass they were printing on.   What ever the bed size is...   The inset subtracts off a border or region from that printable space.

You can set your bed size so the nozzle can reach the entire printable area of the bed.    And if you set a Mesh_Inset of 41 you will be able to probe the entire new printable region.   You can optionally set the mesh inset of each of the four edges if you want to in order to reclaim a bunch of space.

Or, you can accept that there are unprobable points on your printable bed and fill them in at the end of the probing with a G29 P2 command.

rado79, 2020-05-19:
Okay, so (MESH_INSET +5) shifts the Mesh Edge 5mm in direction bed edge? Not in direction bed Center?

rado79, 2020-05-19:
I think i got it. With mesh_inset 15, front row probepoints Y15 could not be reached because of machine limits (probe behind the nozzle). With mesh_inset 20 (15+5) probe could reach Y20.

boelle, 2020-06-23:
@JudgeZ 
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

closed 2020-07-29 by github-actions[bot]

dan9550, 2020-08-08:
@boelle i am having this same issue where the UBL process stops at point 57 / 100 or 31/64 if i adjust the "GRID_MAX_POINTS_X" value from 10 to 8. I have tested this on a build of the bugfix-2.0.x branch that i compiled today.

Happy to help out where i can to investigate this further, let me know if you need any more information.

johanvanj, 2020-08-26:
I had this exact same problem.

This is however a configuration problem - it would be nice to add something int the compiling step to fail if this config is incorrect.

This problem seems to occur when your **_x or y probe offset_** is greater than your **_PROBING_MARGIN_** and your limits follows the size of your xy printing area ( in my case my printing bed is bigger than the area my printing head can move). This means that at some point the mesh generation ( usually point 57 with 100 points) will try to instruct your printing head / bed to exceed one of the software limits to get the probe to where the nozzle would be when printing (like for instance point 0,0 on the bed will result in a move to -24,-10 in my case but my software limit is 0,0). This then fails the mesh building and Marlin seems to go into a weird state that requires a restart.
In my case setting PROBING_MARGIN to 30 - which is greater than both my x and y probe offsets ) solved it for me.

icthieves, 2020-08-30:
@johanvanj is onto something. I had to do a fair bit of tweaking with those parameters, but I eventually got UBL working on my 235x235 bed with:
```
#define PROBING_MARGIN 30
#define MESH_INSET 50  
```

Note that using the same values for both PROBING_MARGIN and MESH_INSET didn't work, but did reduce the issue (57 points to 91 points). **I didn't have any success until PROBING_MARGIN was _smaller_ than MESH_INSET** (the opposite of what worked 
for Johan, but that's likely due to something being inverted somewhere I guess).