MarlinFirmware/Marlin #15936: [FR] compact platformio.ini

bjoernwedi, 2019-11-18:
Currently the platformio.ini grows and grows. With configuration inheritence we would be able to compact it drastically.

Some thought based on STM32F103RC below:

#
# STM32F103RE
#
[env:STM32F103RE]
platform      = ststm32
framework     = arduino
board         = genericSTM32F103RE
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14
  -DDEBUG_LEVEL=0
build_unflags = -std=gnu++11
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel, SPI
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed = 250000

#
# defaults for STM32F103RC environments
#
[STM32F103RC]
platform          = ststm32
framework         = arduino
board             = genericSTM32F103RC
platform_packages = tool-stm32duino
monitor_speed     = 115200
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = Adafruit NeoPixel, SPI
build_unflags     = -std=gnu++11
build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4

#
# STM32F103RC_fysetc
#
[env:STM32F103RC_fysetc]
extends           = STM32F103RC
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_fysetc.py
lib_ldf_mode      = chain
monitor_speed     = 250000
debug_tool        = stlink
upload_protocol   = serial

#
# BigTree SKR Mini V1.1 / SKR mini E3 / SKR E3 DIP (STM32F103RCT6 ARM Cortex-M3)
#
[STM32F103RC_SKR_MINI]
extends           = STM32F103RC
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = ${STM32F103RC.build_flags}
;build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py ${STM32F103RC.build_flags}

#   STM32F103RC_bigtree .............. RCT6 with 256K
#   STM32F103RC_bigtree_NOUSB ........ RCT6 with 256K (no USB)
#   STM32F103RC_bigtree_512K.......... RCT6 with 512K
#   STM32F103RC_bigtree_512K_NOUSB ... RCT6 with 512K (no USB)
#

[env:STM32F103RC_bigtree]
extends           = STM32F103RC_SKR_MINI
build_flags       = ${STM32F103RC_SKR_MINI.build_flags} -DSTM32_FLASH_SIZE=256 -DUSE_USB_COMPOSITE

[env:STM32F103RC_bigtree_NOUSB]
extends           = STM32F103RC_SKR_MINI
build_flags       = ${STM32F103RC_SKR_MINI.build_flags} -DSTM32_FLASH_SIZE=256

[env:STM32F103RC_bigtree_512K]
extends           = STM32F103RC_SKR_MINI
board_upload.maximum_size=524288
build_flags       = ${STM32F103RC_SKR_MINI.build_flags} -DSTM32_FLASH_SIZE=512 -DUSE_USB_COMPOSITE 

[env:STM32F103RC_bigtree_512K_NOUSB]
extends           = STM32F103RC_SKR_MINI
board_upload.maximum_size=524288
build_flags       = ${STM32F103RC_SKR_MINI.build_flags} -DSTM32_FLASH_SIZE=512

bjoernwedi, 2019-11-18:
Same manner the configuration.h motherboard, serial buad rate and serial ports can be defined and inherited. 
Also printers by config/examples would be managable.

reloxx13, 2019-11-18:
its also possible now to load additional config ini files

https://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs

bjoernwedi, 2019-11-25:
Applied some compacting on **platformio.ini** _(-100 lines)_:
https://github.com/bjoernwedi/Marlin/blob/feature/%2315936/platformio.ini

closed 2020-01-15 by bjoernwedi