MarlinFirmware/Marlin #15939: [BUG] Z connection... Error: All LOW

joydi, 2019-11-18:
Hello everybody,

because of noone can help me i post myproblem here maybe it is a bug.

I have the following hardware:
mks gen l Version 2
tmc2208 1.2

When i connect via ponterface or it is as error in my lcd at after boot there comes an tmc connection error.

In ponterface -> Z connection... Error: All LOW

when i do an an 

m122

SENDING:M122
echo:SD init fail
echo:SD init fail
        X    Y    Z    E
Enabled        false    false    false    false
Set current    800    800    800    800
RMS current    795    795    1436    795
MAX current    1121    1121    2025    1121
Run current    25/31    25/31    25/31    25/31
Hold current    12/31    12/31    12/31    12/31
CS actual    12/31    12/31    0/31    12/31
PWM scale    21    22    0    22
vsense        1=.18    1=.18    0=.325    1=.18
stealthChop    true    true    false    true
msteps        16    16    256    16
tstep        max    max    0    max
pwm
threshold
[mm/s]
OT prewarn    false    false    false    false
off time    3    3    0    3
blank time    24    24    16    24
hysteresis
-end        -1    -1    -3    -1
-start        1    1    1    1
Stallguard thrs
DRVSTATUS    X    Y    Z    E
stst        *    *        *
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
        X    0xC0:0C:00:00
        Y    0xC0:0C:00:00
        Z    0x00:00:00:00     Bad response!
        E    0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing E connection... OK
Disconnected.

You can see that the z stepper weird settings.

With an M906
X driver current: 800
Y driver current: 800
Z driver current: 800
E driver current: 800

I exchange the stepper driver with an working from board that did not solve the problem it is the same.

I set the voltage for xyz to 0,760 and for z at 0,900 

Is some on there that can help me ?

Thanks in advance

### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

teemuatlut, 2019-11-18:
Update to v0.6.1.

joydi, 2019-11-18:
Sorry ? Update what ?

teemuatlut, 2019-11-18:
The TMCStepper library. The v0.6.0 release had some issues and had to be pulled back.
You can either edit the platformio.ini file or wait until PIO updates their library database.

PGrave, 2019-11-18:
I had a similar problem, solved using TMC0.6.1

joydi, 2019-11-18:
Sorry is there an command or something to force the update ?

ManuelMcLure, 2019-11-18:
There was a checkin to the platformio.ini a few days that forces <0.6.0 for TMCStepper. I don't know if that will need to be reverted to allow 0.6.1 to be used.

thisiskeithb, 2019-11-18:
> There was a checkin to the platformio.ini a few days ago that forces <0.6.0 for TMCStepper. I don't know if that will need to be reverted to allow 0.6.1 to be used.

https://github.com/MarlinFirmware/Marlin/commit/d33ca3d0589c426b9ed855fc9320721e3b7243e1

PGrave, 2019-11-18:
I used the path to the library TMC0.6.1 in platformIo.ini
Other option is to use the platformio.ini as is now, because it's still forcing use of TMC0.5.2, which works without problems.
-Have you updated firmware/hardware recently?
Because the issue with the TMC library is a fairly recent one and maybe your problem might be something else.

sjasonsmith, 2019-11-18:
@joydi is it always the Z driver that fails, or does the failing driver change randomly?

joydi, 2019-11-18:
No it is always the Z driver :(

And how i can force to use version 0.5.9 or 0.6.1 of the drivers ?

I try platformio lib install "TMCStepper@0.6.0" in the io cli but he want install it ?

joydi, 2019-11-18:
Hey Guys,

thank you so much ->

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

and so quick !

I think i can print again  :)

closed 2019-11-18 by joydi