MarlinFirmware/Marlin #15945: Cannot build Marlin 2.0 for SKR 1.3

LordBBQ, 2019-11-19:
I recently recieved my BIQU SKR 1.3 mainboard. I have started to configure it in VSCode, on Mac, but tried Atom, and both also on a Windows VM. Here is the terminal error: > Executing task: C:\Users\Joshua Moran\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.0.zip; framework: arduino; board: nxp_lpc1768)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.0 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:133:23: warning: assertions are a deprecated extension [-Wdeprecated]
   #define MOTHERBOARD #define BOARD_SKR_1_3
                       ^
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:324:42: note: in expansion of macro 'MOTHERBOARD'
 #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
                                          ^~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:156:29: note: in expansion of macro '_MB_1'
 #define _DO_7(W,C,A,V...)  (_##W##_1(A) C _DO_6(W,C,V))
                             ^
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:162:28: note: in expansion of macro '_DO_7'
 #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
                            ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:163:28: note: in expansion of macro '__DO_N'
 #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
                            ^~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:164:28: note: in expansion of macro '_DO_N'
 #define DO(W,C,V...)       _DO_N(W,C,NUM_ARGS(V),V)
                            ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:325:19: note: in expansion of macro 'DO'
 #define MB(V...)  DO(MB,||,V)
                   ^~
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:40:7: note: in expansion of macro 'MB'
 #if   MB(RAMPS_13_EFB, RAMPS_14_EFB, RAMPS_PLUS_EFB, RAMPS_14_RE_ARM_EFB, RAMPS_SMART_EFB, RAMPS_DUO_EFB, RAMPS4DUE_EFB)
       ^~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:214:23: error: missing binary operator before token "2016"
 #define BOARD_SKR_1_3 2016
                       ^~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:133:31: note: in expansion of macro 'BOARD_SKR_1_3'
   #define MOTHERBOARD #define BOARD_SKR_1_3
                               ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:324:42: note: in expansion of macro 'MOTHERBOARD'
 #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
                                          ^~~~~~~~~~~
[... 858 lines cut ...]
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
====================================================================== 1 failed, 0 succeeded in 00:00:08.176 ====================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

I tried the unconfigured version of Marlin 2.0, but same error.
Folder attached below:
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3862410/Marlin-bugfix-2.0.x.zip)

teemuatlut, 2019-11-19:
L133:
```cpp
#define MOTHERBOARD #define BOARD_BIGTREE_SKR_V1_3
```

ellensp, 2019-11-19:
Not a bug, user error, as listed above rather tersely. 
Longer version for the confused. 
In Configuration.h you have to set #define MOTHERBOARD to one of boards listed in boards.h  
 see https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/boards.h 
The user has set #define MOTHERBOARD #define BOARD_SKR_1_3   this is not a valid board or the correct syntax as it contain a second #define statement.
It should be  #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

LordBBQ, 2019-11-19:
Thanks, works fine now.

closed 2019-11-19 by LordBBQ

LordBBQ, 2019-11-20:
Sorry, just a joke. They told me that you stole something, so I kicked you. Kick only means that you ‘leave the game’. Sorry if you took it the wrong way.