MarlinFirmware/Marlin #15947: STM32 EXTI Conflicts (F4 only?)

AnHardt, 2019-11-19:
### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-11-19:
is this a bug?

AnHardt, 2019-11-19:
Likely this is the **reason** for several bugs, where one endstop is not working when endstop_interrupts is defined or a one serial TMC does not want to communicate.

Developers have to check if such conflicts do exist on other STM32 boards. 
If yes, we cant really fix that in software, but we can give warnings to avoid that pin combinations.

At least we do know now the reason for, why some configurations do not work.

AnHardt, 2019-11-19:
BTT SKR mini V1.1 
Conflicts:
X_MIN [PA**2**] with X_MAX [PC**2**]
Y_MIN [PC**1**] with Y_MAX [PA**1**]

sjasonsmith, 2019-11-19:
The SKR Pro only uses software serial for TMC drivers, which depends only on timer interrupts. Could there still be a conflict with this, or pin change interrupts are not used?

AnHardt, 2019-11-19:
MKS Robin:
Y_MAX_PIN PC**4**
Z_MIN_PIN PA**4**

@sjasonsmith 
If the software serial does not use the EXTI the EXTI conflict does not matter.  
I don't really know the implementation of the **currently** used library. But i mean to remember to have seen attempts to use PCIs and timers instead of polling (maybe not for STMs).

sjasonsmith, 2019-11-19:
> If the software serial does not use the EXTI the EXTI conflict does not matter.
> I don't really know the implementation of the **currently** used library. But i mean to remember to have seen attempts to use PCIs and timers instead of polling (maybe not for STMs).

AVR uses an interrupt to detect the start bit. LPC and STM32 just use a timer interrupt to poll. This would be good for us to remember though, if anyone attempted to re-implement it differently. This was mentioned when we submitted fixes to SoftwareSerial in Arduino_Core_STM32.

boelle, 2019-12-07:
i assume that this has not changed in the recent 2.0 release?

boelle, 2019-12-24:
what could we do about this one?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle