MarlinFirmware/Marlin #15948: [BUG] SKR1.3 + mks mini 12864LCD. Hangs with black screen

karabas2011, 2019-11-19:
Since some recent snapshots ( currently from 07.nov.2019 )
The problem is
suddenly display shows black square, no matter printer is idle or printing.
If it prints it stops, and impossible to connect via printerface, but pressing on encoder I hear sound of clicks.
Power off - on returns printer to normal state.

Sometimes display shows white screen, not stops, pressing to encoder returns normal displaying.

boelle, 2019-11-19:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

thisiskeithb, 2019-11-19:
I have a MKS Mini 12864 V2 hooked up to an SKR 1.3 running the latest commit (and others before) and haven’t had issues with beeping.

I’ll keep an eye out for the display issues on my setup and see if I can reproduce it.

karabas2011, 2019-11-20:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3868468/Marlin.zip)
attached config files.

karabas2011, 2019-11-20:
today 3-4 black screens had happen.
TCM2208 uart on XY suddenly ( in idle) stops working, motors off/on ( or switch to chop/silent )returns them to work. Z is working.

boelle, 2019-11-26:
btw what version of marlin are you using?

karabas2011, 2019-11-28:
I use marlin 2.0 09-nov-2019 
SKR1.3 TMC2208 MKS MINI LCD 12864 v2
Newer versions do not see sdcard inserted into LCD after booting. You need to eject/insert after boot.

boelle, 2019-12-07:
what happens if you boot with no sd card and insert after boot ?

boelle, 2019-12-07:
@karabas2011 since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

karabas2011, 2019-12-26:
It seems problem solved. At least on 2.0.x bugfix from 11-29-2019 no problem

karabas2011, 2019-12-26:
no hanging, no problems with sdcard on last snaphots

closed 2019-12-26 by karabas2011