MarlinFirmware/Marlin #15949: [BUG] BLTOUCH configuration is missing

LoudMurmur, 2019-11-19:
### Description

BLTOUCH configuration is missing from 2.0 "bugfix"

### Steps to Reproduce

open `configuration.h`, ctrl+f, `X_PROBE_OFFSET_FROM_EXTRUDER` cannot be found

There is no `#if ENABLE(BLTOUCH)` block in the file

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h

**Expected behavior:** 

bltouch configuration is found

**Actual behavior:** 

bltouch configuration is missing

Lord-Quake, 2019-11-19:
I sure you are using an older version of `configuration.h`. That's not a good idea.

You are looking for `NOZZLE_TO_PROBE_OFFSET` in the latest version.

sjasonsmith, 2019-11-19:
And all the advanced BLTOUCH options moved to configuration_Adv.h.

LoudMurmur, 2019-11-20:
Oh my bas, the tutorials I was watching were outdated :x

closed 2019-11-20 by LoudMurmur