MarlinFirmware/Marlin #15951: FR Independant operation of steppers in dual configuration

xeddog, 2019-11-19:
### Description
I am running a CR-10S with a MKS SGEN L controller, updated Marlin bugfix-2.0.x, dual z steppers and enabled auto-align.  When I first configured it, I was getting great results with differences getting to .01mm, and occasionally even 0.00.  Lately, I am not getting the results I once was when running the G34 auto-align.  I am now getting results like .06, .04, .05. ,05, and .06 for the five iterations.  Yes, it actually does get worse at the end.  I have not achieved the .02 accuracy defined for quite some time.  In order to assist in diagnosing this, I was trying to find a way to manually move the individual steppers from the control panel, but, of course, that is not available.  I would not think this would be too difficult because isn't that the way auto-align works anyway?  So it would just need to be added as manual task on the control panel.  Easy,  Right???   What could go wrong?  Besides, what else do you have to do?   (Just kidding.  I know you have a LOT to do, and you are doing a great job.)

### Steps to Reproduce

1.  Home machine
2.  auto-align

#### Additional Information
I know I could reach back and manually turn one of them, but I could not get the accuracy and control of firmware generated commands.

sjasonsmith, 2019-11-20:
An error was accidentally introduced when a triple-Z change I made was merged. @Evgeny-SPB has a pull request open to fix that: #15938 

You should give the change from that PR a try to see if it improves your G34 behavior.

xeddog, 2019-11-20:
Thanks.  I looked through the issues, but not the pr's.  I'll look better next time.  But still, what about a control panel manipulation of the individual steppers in a dual/triple configuration?  Would there be any benefit to anyone besides me?

InsanityAutomation, 2019-11-20:
@xeddog I would support from gcode not from the menu. Seems far too dangerous and potentially damaging if a novice leaves it set. Certain commands like a G28 would need to reset that locked state as well.

xeddog, 2019-11-20:
I could live with that.  I use Octoprint so I have a console to enter gcode commands.

xeddog, 2019-11-21:
Just FYI, I ran a pull/build/flash today and the G34 is working again.  But I would still like to have some means of moving each stepper independently.

closed 2020-06-30 by github-actions[bot]