MarlinFirmware/Marlin #15952: [BUG] Unable to compile Marling bugfix 2.0 with TMC5160 drivers on RUMBA32 board

DropbearNinja, 2019-11-20:
### Description

Unable to compile Marling bugfix 2.0 with TMC5160 drivers on RUMBA32 board

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. clean arduino install
2. isntall stm32duino as per here: https://github.com/stm32duino/wiki/wiki/Getting-Started including stm32cubeprogrammer
3. install TMCstepper library
4. open marlin.ino
5. change only: 
`#define MOTHERBOARD BOARD_RUMBA32
#define X_DRIVER_TYPE  TMC5160`
6. ensure ENDSTOP_INTERRUPTS_FEATURE commented out (default)
7. try all combinations of active/commented out of:
'  //#define TMC_USE_SW_SPI
  //#define TMC_SW_MOSI       PA7
  //#define TMC_SW_MISO       PA6
  //#define TMC_SW_SCK        PA5'

**Expected behavior:** [What you expect to happen]

To compile

**Actual behavior:** [What actually happens]

Error:
`Arduino: 1.8.10 (Windows 10), Board: "3D printer boards, RUMBA32, STM32CubeProgrammer (DFU), Enabled (generic 'Serial'), CDC (no generic 'Serial'), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

WARNING: library TMCStepper claims to run on avr architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::read()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial4readEv+0x0): multiple definition of `SoftwareSerial::read()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial4readEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::available()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial9availableEv+0x0): multiple definition of `SoftwareSerial::available()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial9availableEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::flush()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial5flushEv+0x0): multiple definition of `SoftwareSerial::flush()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial5flushEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::peek()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial4peekEv+0x0): multiple definition of `SoftwareSerial::peek()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial4peekEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::setSpeed(unsigned long)':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial8setSpeedEm+0x0): multiple definition of `SoftwareSerial::setSpeed(unsigned long)'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial8setSpeedEm+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::write(unsigned char)':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial5writeEh+0x0): multiple definition of `SoftwareSerial::write(unsigned char)'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial5writeEh+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::stopListening()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial13stopListeningEv+0x0): multiple definition of `SoftwareSerial::stopListening()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial13stopListeningEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::listen()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerial6listenEv+0x0): multiple definition of `SoftwareSerial::listen()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerial6listenEv+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::~SoftwareSerial()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerialD2Ev+0x0): multiple definition of `SoftwareSerial::~SoftwareSerial()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerialD2Ev+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o: in function `SoftwareSerial::~SoftwareSerial()':

SoftwareSerial.cpp:(.text._ZN14SoftwareSerialD2Ev+0x0): multiple definition of `SoftwareSerial::~SoftwareSerial()'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:SoftwareSerial.cpp:(.text._ZN14SoftwareSerialD2Ev+0x0): first defined here
[... 44 lines cut ...]

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial9active_inE+0x0): multiple definition of `SoftwareSerial::active_in'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial9active_inE+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial10active_outE+0x0): multiple definition of `SoftwareSerial::active_out'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial10active_outE+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial15active_listenerE+0x0): multiple definition of `SoftwareSerial::active_listener'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial15active_listenerE+0x0): first defined here

c:/users/jonha/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\SoftwareSerial\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial5timerE+0x0): multiple definition of `SoftwareSerial::timer'; sketch\src\HAL\HAL_STM32\SoftwareSerial.cpp.o:(.bss._ZN14SoftwareSerial5timerE+0x0): first defined here

collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board 3D printer boards.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

#### Additional Information

config and pins file attached

Please help

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3867613/config.zip)

sjasonsmith, 2019-11-20:
The current STM32 Arduino Core has issues in the SoftwareSerial implementation, so we temporarily have our own in Marlin. SoftwareSerial is structured as a separate library within the Arduino Core, so we were able to exclude it using PlatformIO. It did not occur to us that this would cause a problem when building within Marlin.

sjasonsmith, 2019-11-20:
You don't actually need SoftwareSerial since you are using 5160 drivers. It is only needed as a dependency of TMCStepper, but it doesn't actually have to _work_ for your needs. As a temporary workaround you could probably delete `Marlin/src/HAL/HAL_STM32/SoftwareSerial.cpp` file, and let it use the version it finds in the Arduino Core.

That would probably _not_ work for someone using TMC 2208/9 drivers with software serial. It _might_ work if they are using separate RX/TX pins, but won't work with a single pin.

DropbearNinja, 2019-11-20:
Thank you for such a quick and helpful reply! I tried removing the file, and it compiled and works as expected :) thank you so much!

closed 2019-11-20 by DropbearNinja