MarlinFirmware/Marlin #15955: FILAMENT RUNOUT without ADVANCED_PAUSE disables M125

karabas2011, 2019-11-20:
I am trying to pause print (filament sensor)  with nozzle park but without filament change features
But it disabling  ADVANCED_PAUSE_FEATURE disables M125 in many places in code.
Are there any ways to solve the problem?
SKR1.3 +marlin2.0  9-nov-2019

ManuelMcLure, 2019-11-20:
No. ``PARK_HEAD_ON_PAUSE`` (which adds ``M125``) absolutely requires ``ADVANCED_PAUSE_FEATURE``.

karabas2011, 2019-11-21:
There was some patches for avoiding  ADVANCE, 
something like PARK_HEAD_ON_PAUSE

ManuelMcLure, 2019-11-21:
``PARK_HEAD_ON_PAUSE`` in the configuration files is inside a block that is controlled by ``ADVANCED_PAUSE_FEATURE``, so there is no way to enable one without the other.

karabas2011, 2019-11-21:
Is there way to disable temp waiting on pause?

boelle, 2019-11-24:
will temp close this one until we have the configs as required by the template

ping me by adding a @ before my username and we can reopen this one

closed 2019-11-24 by boelle