MarlinFirmware/Marlin #15956: [BUG] LPC1768 and TMC2208 bootloop

LoudMurmur, 2019-11-20:
# Description

`#define X_DRIVER_TYPE  TMC2208`
`#define Y_DRIVER_TYPE  TMC2208`
`#define Z_DRIVER_TYPE  TMC2208`

will cause a bootloop

### Steps to Reproduce

checkout commit 865055e07a5de40fa615414eba0dbf0ca05fcbce

put 

`#define X_DRIVER_TYPE  TMC2208`
`#define Y_DRIVER_TYPE  TMC2208`
`#define Z_DRIVER_TYPE  TMC2208`

put `C:\GitRepos\Marlin\.pio\build\LPC1768\firmware.bin` in the sd card
install stepper driver with jumpers
power on the board

**Expected behavior:** 

boots

**Actual behavior:** [What actually happens]

bootloop

#### Additional Information

it works fine if I use TMC2100 (with TMC2100 drivers installed) however it will bootloop with TMC2208, configuration files in zip.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3871185/Marlin.zip)

s0ftcorn, 2019-11-20:
I pulled today and im using TMC2208s on X Y Z E0 (extruder) and E1 (2nd Z axis).
I attached my config to compare
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3871251/Marlin.zip)

LoudMurmur, 2019-11-20:
I see no important difference.

I pulled, rebased my commits above, still bootlooping

LoudMurmur, 2019-11-21:
I think my mainboard is defective, I can't use 3 TMC2208, it works with 2 TMC2208

So I tested every possible combinaison, recompiling a marlin firmware each time and putting the uart jumper for TMC2208 while removing it for TMC2100 :

first I tried everyone of my 4 stepper controller to see if it was one of them :
X(TMC2100), Y(TMC2100), Z(TMC2208_1) -> working
X(TMC2100), Y(TMC2100), Z(TMC2208_2) -> working
X(TMC2100), Y(TMC2100), Z(TMC2208_3) -> working
X(TMC2100), Y(TMC2100), Z(TMC2208_4) -> working

One TMC2208 test :
X(TMC2100), Y(TMC2100), Z(TMC2208) -> working
X(TMC2100), Y(TMC2208), Z(TMC2100) -> working
X(TMC2208), Y(TMC2100), Z(TMC2100) -> working

Two TMC2208 test :
X(TMC2100), Y(TMC2208), Z(TMC2208) -> working
X(TMC2208), Y(TMC2208), Z(TMC2100) -> working
X(TMC2208), Y(TMC2100), Z(TMC2208) -> working

Three TMC2208 test :
X(TMC2208), Y(TMC2208), Z(TMC2208) -> BOOTLOOP

Hardware issue?

s0ftcorn, 2019-11-21:
Did you double check you made no changes in the pins.h and the jumpers on the SKR 1.3 are correctly set?

LoudMurmur, 2019-11-21:
Yes I never touched pin.h, I'm sure because it does not appear in the modified list of gitextension.

for the 2208 I did this : https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3-TMC2208-UART-Mode.png

I just remove the "uart jumper" when I plug a TMC2100.

I will doublecheck tonight, I was under time constraint this morning so I might have missed something.

LoudMurmur, 2019-11-22:
Well same issue, I can confirm that everything is plugged correctly and that it will bootlop when using 3 TMC2208 but works fine with 2 on any axis

s0ftcorn, 2019-11-22:
could you try 4? cause im using 5 and it works.

boelle, 2019-11-24:
@LoudMurmur did you try what @s0ftcorn suggested?

also @LoudMurmur does this relate to your other issue #15984 ??

LoudMurmur, 2019-11-25:
nope not tried, I will try in 1-2 days, printer is currently offline because I'm building an enclosure.
this is not linked to #15984

LoudMurmur, 2019-11-26:
So it is still bootlooping with 4 TMC2208

But I just noticed that in octoprint : EEPROM version mismatch (EEPROM=V72 Marlin=V73)

```
Send: N0 M110 N0*125
Recv: echo:PowerUp
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2019-11-18 | Author: (Creality CR-10)
Recv: echo:Compiled: Nov 26 2019
Recv: echo: Free Memory: 7831  PlannerBufferBytes: 1344
Recv: echo:SD card ok
Recv: echo:EEPROM version mismatch (EEPROM=V72 Marlin=V73)
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:EEPROM version mismatch (EEPROM=V72 Marlin=V73)
Recv: echo:Hardcoded Default Settings Loaded
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: echo:SD card ok
Recv: echo:EEPROM version mismatch (EEPROM=V72 Marlin=V73)
Recv: echo:Hardcoded Default Settings Loaded
Recv: 
Recv: 
Recv: echo:LPC1768 (100Mhz) Initialized
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv: echo:PowerUp
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2019-11-18 | Author: (Creality CR-10)
Recv: echo:Compiled: Nov 26 2019
Recv: echo: Free Memory: 7831  PlannerBufferBytes: 1344
Recv: echo:SD card ok
Recv: echo:EEPROM version mismatch (EEPROM=V72 Marlin=V73)
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:EEPROM version mismatch (EEPROM=V72 Marlin=V73)
Recv: echo:Hardcoded Default Settings Loaded
```

s0ftcorn, 2019-11-26:
Marlin can act up weird when the eeprom isnt newly initialized after a FW flash.
You can tell marlin in configuration.h to auto-init the eeprom on any errors. That way you dont need a not-bootlooping FW to reset the eeprom. Although you can just use a working firmware and do a M502 followed by M500.
note that M501 will restore settings from eeprom, which is not what you want i think.

LoudMurmur, 2019-11-26:
ok so I setup marlin to reset the EEPROM on errors.

And it works with 3xTMC2208 on XYZ :D :O thanks everybody.

I tried 4 and it worked too, but the extruder motor was screaming high frequencies when turning, so I removed it.

s0ftcorn, 2019-11-26:
recheck the wiring. usually when motors are screaming the wiring is not okay. For example i have quite a lot of cables that dont connect the right windings to the driver outputs.
Spreadcycle can wine too, but since stealthchop is default, i would recheck the wiring.

boelle, 2019-11-26:
and TMC scream/whine more on 12V than 24V

LoudMurmur, 2019-11-26:
it was 12v, how could the wiring be bad if it prints correctly?

boelle, 2019-11-26:
could means it does not have to be, if wiring is ok then the most likely is that you need 24V
or vref is not correct

but i will close this one as the org issue you had is solved

you can still write thou :-)

closed 2019-11-26 by boelle