MarlinFirmware/Marlin #15958: MMU Menu

tedukes, 2019-11-21:
In the MMU Menu there is Eject Filament and Unload Filament. Both require the hotend to be heated. Why?
Not sure what the difference is for these options. Could the be changed to Unload filament from Nozzle and Unload Filament, similar to the Load options?

Also, the 'bug fix' made to the mmu_menu in todays d/l do not work for me. I had to revert back to a 10/27 d/l.

boelle, 2019-11-21:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

tedukes, 2019-11-21:
This was a feature request. Also I asked similar questions on the Marlin Facebook page per instruction and have not received an answer.

Don't be an jerk!

boelle, 2019-11-21:
dont be rude

i did not see it as a feature reqest as the issue template was not followed, if it had the title would have told what it was

but i did see 2 questions and that was the reason why i closed it as they simply just dont belong on the issue list, here we have only clear and precise bugs and feature request

personally i use discord server to post questions etc, sometimes it take a few days for an reply

GMagician, 2019-11-22:
> In the MMU Menu there is Eject Filament and Unload Filament. Both require the hotend to be heated. Why?

Not sure since I don't have a MMU but one should move filament back to drive gear, other should move it till MMU. Both require hot end hot otherwise filament can't me moved...

tedukes, 2019-11-22:
> dont be rude
>
> i did not see it as a feature reqest as the issue template was not followed, if it had the title would have told what it was
>
> but i did see 2 questions and that was the reason why i [...]

I'll try that route as there seems to no support on Facebook. Just people looking for some help like me.

Thanks

tedukes, 2019-11-22:
> 
> 
> > In the MMU Menu there is Eject Filament and Unload Filament. Both require the hotend to be heated. Why?
> 
> Not sure since I don't have a MMU but one should move filament back to drive gear, other should move it till MMU. Both require hot end hot otherwise filament can't me moved...

I just got this thing and still trying figure out things. I calibrated for my bowden tube length and it loads filament perfectly. I have printed in single mode with no problems except can't seem to figure out the unload. 
I saw an issue on Prusa's github about that. I flashed the latest firmware last night but did not have time to test it.

closed 2020-06-30 by github-actions[bot]