MarlinFirmware/Marlin #15959: [BUG] marling 2.0 + stm32 + tmc5160 - sensorless homing not working

DropbearNinja, 2019-11-21:
### Description

With marlin 2.0 bugfix + stm32 + tmc5160 stallguard doesn't seem to be working

### Steps to Reproduce

1. install clean marlin
2. setup config
3. try to home/trigger stallguard

**Expected behavior:** 

Stallguard to trigger

**Actual behavior:** 

Does not trigger. If you trigger an auto home, the Z axis moves it's 10, then X moves a little and stops. If you send a move command that runs it into an end, it keeps going, doesn't stall.

#### Additional Information

The TMC5160 DIAG1 pin is connected to sense pin of XMIN endstop on board

```
#define X_DRIVER_TYPE  TMC5160
#define X_MIN_ENDSTOP_INVERTING true
#define ENDSTOPPULLUP_XMI
#define USE_XMIN_PLUG

#define SENSORLESS_HOMING
```

> SENDING:M122
> 		X	Y	Z	E
> Enabled		false	false	false	false
> Set current	800	800	800	800
> RMS current	792	792	792	792
> MAX current	1117	1117	1117	1117
> Run current	15/31	15/31	15/31	15/31
> Hold current	7/31	7/31	7/31	7/31
> Global scaler	133/256	133/256	133/256	133/256
> CS actual	7/31	8/31	8/31	8/31
> PWM scale	7	8	8	8
> stealthChop	false	false	false	false
> msteps		16	256	256	256
> tstep		max	max	max	max
> pwm
> threshold
> [mm/s]
> OT prewarn	false	false	false	false
> off time	4	0	0	0
> blank time	24	36	36	36
> hysteresis
> -end		2	-1	-1	-1
> -start		1	6	6	6
> Stallguard thrs	8	-50	0	0
> DRVSTATUS	X	Y	Z	E
> stallguard
> sg_result	33	0	0	0
> fsactive
> stst		*	*	*	*
> olb
> ola
> s2gb
> s2ga
> otpw
> ot
> Driver registers:
> 		X	0x80:07:00:21
> 		Y	0x80:08:00:00
> 		Z	0x80:08:00:00
> 		E	0x80:08:00:00
> Testing X connection... OK
> Testing Y connection... OK
> Testing Z connection... OK
> Testing E connection... OK

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3872304/Marlin.zip)
Pins: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32/pins_RUMBA32.h

Please and thank you for help.

boelle, 2019-11-24:
have you tried any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-03 by boelle