MarlinFirmware/Marlin #15963: MKS Robin Mini + 3.2 TFT, touchscreen not working

KosVas91, 2019-11-21:
I have FLSUN QQ-S printer with MKS robin Mini board and 3.2 TFT screen. Previously I was using [this](https://github.com/berghauz/Marlin-bugfix-2.0.x-FLSUN-QQ-S) build, and everything worked fine. 

Few day ago I decided to update FW to current state and faced problem that I can not solve - LCD screen initializes fine and shows Marlin main screen (with touch buttons), but screen does not react to touches at all.

To exclude misconfiguration I am testing screen with default config changing only MOTHERBOARD, FSMC_GRAPHICAL_TFT, TOUCH_BUTTONS settings in Configuration.h and disabling LCD_RESET_PIN in pins_MKS_ROBIN_MINI.h.
Also tried to use TFT screen calibration settings from working build, but no luck.

Any advices from experienced users?

PS: Looks very similar to [[BUG] MKS Robin TFT24 1.0 Touch not working. #15316](https://github.com/MarlinFirmware/Marlin/issues/15316)

ellensp, 2019-11-21:
The TFT  is just a serial device did you define both serial ports. I have to ask because you didn't attach your configuration files as the template advised....   
Ie 
#define SERIAL_PORT 3
#define SERIAL_PORT_2 1
#define NUM_SERIAL 2
and of course the baud rate must match the baud rate in the TFT configuration.

thisiskeithb, 2019-11-21:
It’s a known issue with those MKS Robin Nano/Mini boards that have a `FSMC_GRAPHICAL_TFT`/tethered TFT screen. The LCD worked when when I originally submitted the board pins, but the code around it has changed enough to break something and I don’t have the correct hardware to test with.

[Here’s a video](https://youtu.be/HtlUYJ9S-b4) with some changes made to Marlin get the TFT working on a Nano, but I can’t test the fix.

KosVas91, 2019-11-21:
> The TFT is just a serial device did you define both serial ports. I have to ask because you didn't attach your configuration files as the template advised....
> Ie
> #define SERIAL_PORT 3
> #define [...]

Sorry, forgot to mention. Serial ports are defined correctly.

> It’s a known issue with those MKS Robin Nano/Mini boards that have a `FSMC_GRAPHICAL_TFT`/tethered TFT screen. The LCD worked when when I originally submitted the board pins, but the code around it [...]

I have watched this video while investigating problem. But if I understood correctly, in video, it was an attempt to fix wrong screen resolution (2,8/3,2 TFT is 320x240 while 3.5 is 480x320), but touchscreen was working from very beginning. In my case resolution is correct.

boelle, 2019-11-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

KosVas91, 2019-11-22:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[Robin_Mini_Config.zip](https://github.com/MarlinFirmware/Marlin/files/3878322/Robin_Mini_Config.zip)

This is a "bare minimum" configuration I have tried, just to initialize TFT and test touchscreen.

boelle, 2019-11-26:
btw. what version of marlin are you using?

KosVas91, 2019-11-26:
> 
> 
> btw. what version of marlin are you using?

bugfix-2.0.x branch, if I correctly understood your question

KosVas91, 2019-11-27:
I have found some time to google a bit, and almost immediately found working solution. Do not know how, but missed it previously. Answer is [here](https://www.facebook.com/groups/1049718498464482/permalink/1883215031781487/).
Adding this to pins_MKS_ROBIN_MINI.h helped:
```
#if ENABLED(TOUCH_BUTTONS)
#define TOUCH_CS_PIN PC2
#define TOUCH_SCK_PIN PB13 // pin 52
#define TOUCH_MOSI_PIN PB15 //PB14 // pin 53
#define TOUCH_MISO_PIN PB14 //PB15 // pin 54

#define LCD_USE_DMA_FSMC // Use DMA transfers to send data to the TFT
#define FSMC_DMA_DEV DMA2
#define FSMC_DMA_CHANNEL DMA_CH5
#define BTN_EN1 -1 // Real pin is needed to enable encoder's push button
#define BTN_EN2 -1 // functionality used by touch screen

#define DOGLCD_MOSI -1 // Prevent auto-define by Conditionals_post.h
#define DOGLCD_SCK -1
#endif
```

The only problem now, is that I am total noob in programming and don't really understand if all lines are necessary or may be single line will help. May be somebody more experienced could take a look and add changes to source?

boelle, 2019-11-27:
@thinkyhead might be able to figure it out, but he have a damm lot of work on his plate as he is the maintainer of marlin

patience is the key word

boelle, 2019-12-07:
@KosVas91 since 2.0 was just released a few days ago has this changed this issue at all?

KosVas91, 2019-12-08:
> 
> 
> @KosVas91 since 2.0 was just released a few days ago has this changed this issue at all?
```
#define TOUCH_SCK_PIN PB13
#define TOUCH_MOSI_PIN PB15
#define TOUCH_MISO_PIN PB14
```

As I wrote previously, you need to define those PINs to make touchscreen working. It is also correct for 2.0.0 release.

KosVas91, 2019-12-13:
Fixed by: https://github.com/MarlinFirmware/Marlin/pull/16178

closed 2019-12-13 by KosVas91