MarlinFirmware/Marlin #15964: [BUG] Bed leveling correction not moving smoothly

VertexityStudio, 2019-11-21:
Hi,

I've a little problem with Marlin 2.0 on MKS Sbase 1.3. I've enable "AUTO_BED_LEVELING_BILINEAR", then make a G29 to probe the bed.
When I print, the corrections on the Z axis are not smooth, the motors move in jerks.

An idea please?

Thanks

[Configuration.h](https://github.com/MarlinFirmware/Marlin/files/3875117/Configuration.txt)
[Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/files/3875118/Configuration_adv.txt)

boelle, 2019-11-21:
moving faster than MCU can handle?
if using TMC what volt?

VertexityStudio, 2019-11-21:
> moving faster than MCU can handle?
What do you mean ? Sorry I'm french

> if using TMC what volt?
MKS Sbase is DRV8825

boelle, 2019-11-21:
> What do you mean ? Sorry I'm french

when trying to move the steppers faster than the cpu can handle it will become "jerky"

> MKS Sbase is DRV8825

oki then 12V is enough, TMC's that are popular needs 24V and active cooling

tpruvot, 2019-11-21:
There was a commit related to AUTO_BED_LEVELING_BILINEAR (ABL) yesterday 

commit 4ede13e36a38aa1407a7ca06c1085d36928057a7
Author: Jason Smith <jason.inet@gmail.com>
Date:   Thu Nov 21 01:19:56 2019 -0800

    Fix G29 for updated meshCount type (#15960)

boelle, 2019-11-21:
@VertexityStudio so try with todays copy of 2.0.x and see if its not running better

tpruvot, 2019-11-21:
else The jerk issue you see might be related to linear advance... if enabled in config, try to comment and reset settings...

VertexityStudio, 2019-11-22:
> There was a commit related to AUTO_BED_LEVELING_BILINEAR (ABL) yesterday
> 
> commit [4ede13e](https://github.com/MarlinFirmware/Marlin/commit/4ede13e36a38aa1407a7ca06c1085d36928057a7)
> Author: Jason Smith
> Date: Thu Nov 21 01:19:56 2019 -0800
> 
> Fix G29 for updated meshCount type (#15960)

I'll check today ;)

> @VertexityStudio so try with todays copy of 2.0.x and see if its not running better

I'll try last version today, thanks

> Else the jerk issue you see might be related to linear advance... if enabled in config, try to comment it and reset settings...

I've already try this ;)

VertexityStudio, 2019-11-22:
> There was a commit related to AUTO_BED_LEVELING_BILINEAR (ABL) yesterday
> 
> commit [4ede13e](https://github.com/MarlinFirmware/Marlin/commit/4ede13e36a38aa1407a7ca06c1085d36928057a7)
> Author: Jason Smith
> Date: Thu Nov 21 01:19:56 2019 -0800
> 
> Fix G29 for updated meshCount type (#15960)

The new version does not change anything, always these corrections in rather brutal stages.
When I had my old 3d printer under Marlin 1, the corrections were really progressive and not incremental and therefore much softer.
It seems that the firmware waits a certain distance of movement before correcting the height and corrects it suddenly by a very fast movement of the motor Z.

boelle, 2019-11-24:
there is also these fine places you can ask for help

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

VertexityStudio, 2019-12-12:
Problem solved with released version, thanks !!!!!!!!!

closed 2019-12-12 by VertexityStudio