MarlinFirmware/Marlin #15965: [BUG] (short description)

Arduino3128, 2019-11-21:
Hi, 
Recently I upgraded my Tevo Tarantula Firmware to Marlin 2.0 version, I enabled the power loss Recovery option. 
The First part works fine, i.e when the power Comes back it shows Power Outage Menu.
But when I select the "Resume Print" option it home X and Y axis and starts the printing from beginning instead of continuing it from where it left.

shitcreek, 2019-11-21:
How recent is your Marlin build? There's been some code change to the power loss recovery in this commit https://github.com/MarlinFirmware/Marlin/pull/15943/ from 19 hours ago.  It was also reworked extensively in this commit https://github.com/MarlinFirmware/Marlin/pull/15135 from 21 days ago.
Try out the latest build if you aren't already. Did it ever work before? Are you certain this is a bug and not hardware or configuration error as we do not provide tech support here.

For best results getting help with configuration and troubleshooting, please use the following resources:
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-11-21:
yep, closing this one as its more a question than a bug

also the issue template was not read and followed, we can reopen if there is a bug and the missing info is provided (read template to find what is missing)

closed 2019-11-21 by boelle