MarlinFirmware/Marlin #15966: I need to know the pwm frequency [FR] (feature request title)

Milu1980, 2019-11-21:
### Description

### Steps to Reproduce

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Milu1980, 2019-11-21:
Hi, i need to know the pwm frequency, because i build a high speed laser engraver, and i need higher frequency... i think is now just 1kHz,  but i dont find  any info... thx.

shitcreek, 2019-11-21:
The laser and spindle feature is still a work in progress and the documentation is behind. We've got a PR for improving this feature https://github.com/MarlinFirmware/Marlin/pull/15335 - it has a specific define for setting the laser's pwm frequency.

Milu1980, 2019-11-21:
this settings are not yet avaible in the marlin... know you how much is the
frequency now ? I would like to have 4 KHz. Thx.

boelle, 2019-11-24:
Closing this one as questions do not belong to the list, debate etc can continue on discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

@shitcreek goes under the username luulac

closed 2019-11-24 by boelle