MarlinFirmware/Marlin #15973: Probe false deployment

NotTooOldDeveloper, 2019-11-22:
Good day all!
Recently I am experiencing a problem with false probe deployment. It occurs during printing and result in probe being lowered down and destroy the print. This happens from time to time and is not depend on time interval or something.
This is a clear that the specific command has been sent to printer because sometimes it stay lowered and another time it come back up.
The printer is not connected to the PC and print is done from the SD card.

boelle, 2019-11-22:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

NotTooOldDeveloper, 2019-11-22:
Hi!
Sorry for omitting config files, my mistake.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3879285/Configuration.zip)

ellensp, 2019-11-22:
This is often a hardware issue. In particular the 5v line dips in voltage enough to allow the probe to deploy. The voltage comes back and the probe comes back.

GMagician, 2019-11-23:
or also due to EMI (noise)

boelle, 2019-11-25:
@NotTooOldDeveloper have you checked if the voltage drops? and tried to keep the wires for the probe seperate from everything else?

NotTooOldDeveloper, 2019-11-26:
Thank you everyone for your guidance! It looks like the problem is rather in hardware than in marlin. As a temporary solution i have commented this line: “#define DEACTIVATE_SERVOS_AFTER_MOVE” and for the last few prints this problem did not appear again. 
Thank you. Case is closed.

closed 2019-11-26 by NotTooOldDeveloper