MarlinFirmware/Marlin #15980: CR10S SKR 1.3 TMC

helicoptergeek, 2019-11-23:
CR10S SKR1.3 TMC2208 BLTouch / Raspberry Pi 3B+ Octoprint

When I run the ABL script on the LCD it will display x/9 probes but once it has completed 6/9 the printer stops probing.

boelle, 2019-11-23:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

helicoptergeek, 2019-11-23:
[Configh and Configadv.zip](https://github.com/MarlinFirmware/Marlin/files/3882560/Configh.and.Configadv.zip)

sjasonsmith, 2019-11-23:
Are you you using the latest commit of bug fix-2.0.x? A small ABL fix went in a few days ago to fix similar problems on ABL.

helicoptergeek, 2019-11-23:
I have been fighting with this for over a week.  I will try with what is up right now and respond back.

helicoptergeek, 2019-11-23:
Thanks sjasonsmith.  That fixed it.  I should have tried that before asking.

closed 2019-11-23 by helicoptergeek