MarlinFirmware/Marlin #15983: Extruder stop working in first print

Kurairaito, 2019-11-23:
Hi, 

I hve a strange bug, let me explain first. When I power on the printer, and launch a print from octoprint, the gcode is executed and after my custom extruder start G-code, the extruder (and only it) stop working.

I tried to execute each line by hand and here is the log I got : 

> Send: T0
> Recv: echo:Active Extruder: 0
> Recv: ok
> Send: M82
> Recv: ok
> Send: M190 S60
> Recv: echo:busy: processing
> Send: M104 S160
> Recv: ok
> Send: G21
> Recv: ok
> Send: G28
> Recv: echo:busy: processing
> Recv: X:165.00 Y:127.00 Z:12.66 E:-125.00 Count X:13200 Y:10160 Z:20256
> Recv: ok
> Send: G29 L1
> Recv: Done.
> Recv: ok
> Send: G29 J
> Recv: Tilting mesh (1/3)
> Recv: echo:busy: processing
> Recv: Tilting mesh (2/3)
> Recv: echo:busy: processing
> Recv: echo:busy: processing
> Recv: Tilting mesh (3/3)
> Recv: echo:busy: processing
> Recv: X:116.50 Y:116.50 Z:12.66 E:0.00 Count X:9320 Y:9320 Z:20256
> Recv: ok
> Send: G27 P0
> Recv: echo:busy: processing
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> Send: M109 S200
> Recv: echo:busy: processing
> Recv: ok
> Send: G12 P0
> Recv: ok
> Send: G92 E0
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> Send: G27 P0
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> Send: G90
> Recv: ok
> Send: G92 E0
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> Send: G1 E120 F1500
> Recv: ok
> Send: G92 E0
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> 
> ;;;;FROM HERE EXTRUDER STOPPED WORKING;;;;
> Send: G1 E50 F200
> Recv: ok
> Send: G92 E0
> Recv: X:244.00 Y:117.00 Z:20.00 E:0.00 Count X:19520 Y:9360 Z:32000
> Recv: ok
> Send: G1 E-5 F1500
> Recv: ok
> Send: M122
> Recv: 		X	Y	Z	E	E1
> Recv: Enabled		true	true	true	true	true
> Recv: Set current	636	636	636	954	954
> Recv: RMS current	611	611	611	939	939
> Recv: MAX current	862	862	862	1324	1324
> Recv: Run current	19/31	19/31	19/31	16/31	16/31
> Recv: Hold current	9/31	9/31	9/31	8/31	8/31
> Recv: CS actual	9/31	9/31	9/31	8/31	8/31
> Recv: PWM scale	34	33	32	10	10
> Recv: vsense		1=.18	1=.18	1=.18	0=.325	0=.325
> Recv: stealthChop	true	true	true	true	true
> Recv: msteps		16	16	16	16	16
> Recv: tstep		max	max	max	max	max
> Recv: pwm
> Recv: threshold
> Recv: [mm/s]
> Recv: OT prewarn	false	false	false	false	false
> Recv: OT prewarn has
> Recv: been triggered	false	false	false	false	false
> Recv: off time	4	4	4	4	4
> Recv: blank time	24	24	24	24	24
> Recv: hysteresis
> Recv: -end		2	2	2	2	2
> Recv: -start		1	1	1	1	1
> Recv: Stallguard thrs
> Recv: DRVSTATUS	X	Y	Z	E	E1
> Recv: stst		*	*	*	*	*
> Recv: olb
> Recv: ola
> Recv: s2gb
> Recv: s2ga
> Recv: otpw
> Recv: ot
> Recv: 157C
> Recv: 150C
> Recv: 143C
> Recv: 120C
> Recv: s2vsa
> Recv: s2vsb
> Recv: Driver registers:
> Recv: 		X	0xC0:09:00:00
> Recv: 		Y	0xC0:09:00:00
> Recv: 		Z	0xC0:09:00:00
> Recv: 		E	0xC0:08:00:20
> Recv: 		E1	0xC0:08:00:00
> Recv: 
> Recv: 
> Recv: Testing X connection... OK
> Recv: Testing Y connection... OK
> Recv: Testing Z connection... OK
> Recv: Testing E connection... OK
> Recv: Testing E1 connection... OK
> Recv: ok

Hardware :
TMC2208 with UART on a SKR V1.3
Marlin Bugfix-2.0 up to date (commit 8f09d2e7904f8f743be149b6b213575e7f36ee29 )
I have a bltouch, with UBL activated, and I have a burge bucket on the right of my ender 3.
I first saw the issue about 1 month ago, when I added the burge bucket and modified the custom g-code. I'm sure it's related, but can't understand where is the issue. For the printer everything is fine, it continue to print, just the extruder that don't extrude anymore.

the G-code I try to execute, it's an example for you to see my custom code, maybe there is something wrong?

[CFFFP_m8_washer_v2.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/3882759/CFFFP_m8_washer_v2.gcode.txt)

What I don't understand is why it fail only on the first time I try to print ?

Can you tell me what I should to to debug this?

Thanks.

Kurairaito, 2019-11-24:
Hi,

Good news !
After researchs, the bug can be fixed by disabling S_CURVE_ACCELERATION

So it seems that this function doesn't works as expected.

regards.

boelle, 2019-11-24:
works fine for me with S_CURVE_ACCELERATION enabled

EvilGremlin, 2019-11-29:
Switch extruder driver to spreadcycle and see what happens. 2208's are finicky.

boelle, 2019-11-29:
@Kurairaito tried what @EvilGremlin suggested?

Kurairaito, 2019-11-30:
No I didn't tried it. Disabling S_CURVE_ACCELERATION worked fine and I
don't want to use spreadcycle because I took tmc2208 for the silent.

As some people don't have issue with this option, there might be a conflit
with another one.

It worked fine for months. Then I upgraded from the mks gen L to the SKR
1.3, I took the head of the bugfix branch + I activate this option at the
same time + a lot of others, because I switched from a low memory card to
this one that could handle everything.

There are so much change in one time that it is impossible for me to find
the couple which create an issue, but something is sure : with the
configuration I gave earlier the bug is there, and the same configuration
without S_CURVE_ACCELERATION doesn't have the issue.

Regards.

boelle, 2019-12-07:
@Kurairaito since 2.0 was just released a few days ago has this changed this issue at all?

Kurairaito, 2019-12-15:
Hi, 

I gave it a try today with commit 6a514a1e8e96bcd26985615377523d1e45ca7b10 and I couldn't reproduce the error. I also increased MINIMUM_STEPPER_PULSE to 2.

closed 2019-12-15 by Kurairaito