MarlinFirmware/Marlin #15984: [BUG] LPC1768 bootlooping after changing GRID_MAX_POINTS_X

LoudMurmur, 2019-11-23:
### Description

LPC1768 bootlooping after changing GRID_MAX_POINTS_X to 6

### Steps to Reproduce

change GRID_MAX_POINTS_X to 6 (or 10, also do it)

**Expected behavior:** [What you expect to happen]

boots

#### Additional Information

this is the only change I made, reverting to 3 works

p3p, 2019-11-23:
Without enabling some form of levelling I can't see why that would change anything at all, there is a reason the config file zip is marked as **required** in the issue template.

thisiskeithb, 2019-11-23:
I'm running the latest commit with an SKR 1.3/`LPC1768` using a 7x7 grid and it probes all 49 points with no issue ([link to config](https://github.com/thisiskeithb/Marlin/tree/b4228058f73b3a876d2a8d77c2042d6da90011bf)).

sjasonsmith, 2019-11-23:
@LoudMurmur, is this issue distinct from #15956? It seems odd to have two open issues about your board rebooting.

ellensp, 2019-11-24:
I suspect this is either a hardware issue or a system error on the users machine.  Make sure you not using the crappy supplied SD card, it is crap and corrupts firmware. Also provide the used config files and even the firmware.bin for others to try and also compare to their generated bin files.

boelle, 2019-11-24:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

LoudMurmur, 2019-11-25:
no this is not linked to my toher bootlooping error.

I'm using the crappy sd card, I will try with a good one.

The printer is currently unplugged because I'm building an enclosure, I will try all that in 1-2 days
Here is mt configuration, GRID_MAX_POINTS_X is set to 3

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3885176/Marlin.zip)

LoudMurmur, 2019-11-25:
So here is what I did, in that order :

based on commit b2a275891c5f6f9af0170ae24084ab9dd598651a

**GRID_MAX_POINTS_X 7**

with the stock sd card : bootloop
Samsung 64 evo formated in fat16 and fat32 with a partition of 128MB by ease us partion master : bootloop, because it is not reading the sd card

**GRID_MAX_POINTS_X 3**

with the stock sd card : working

fetched todays commits (55595d444d6969a7bff58f57959d518059acf31e), rebased my modification on them

**GRID_MAX_POINTS_X 7**

with the stock sd card : bootloop

**GRID_MAX_POINTS_X 3**

with the stock sd card : working

I ordered a 4GB card to see if that is the problem (doubt it)

[MyBuilds.zip](https://github.com/MarlinFirmware/Marlin/files/3888299/MyBuilds.zip)

boelle, 2019-11-25:
you dont need extra software to format the sd card :-D

and should it not be one big partition?

ellensp, 2019-11-26:
Neither of the two firmware.bin files provided in your zip boot loop on my SKR 1.3 ...  I do get TMC connection errors as I don't have TMC steppers.  Other than that everything seem fine.  I then compiled from your config files, adding in a _Bootscreen.h  and confirming GRID_MAX_POINTS_X is set to 7 in the AUTO_BED_LEVELING_BILINEAR section of the code. No issues.  I cannot replicate your boot loop issue.

LoudMurmur, 2019-11-26:
mmmmh. hardware issue then.

I'll ty with my new sd card tommorow, if it wants to read it.
Then I'll get a new SKR 1.3 if it still does not work.

@boelle 
windows 10 would not let me do a FAT partition so I used an third party software, could not make a bigger than 2GB partition anyway so I went for the same size than the original card.

ellensp, 2019-11-26:
FAT16 has a 2GB partition limit, should be using FAT32
See https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system#Size_limits

LoudMurmur, 2019-11-26:
@ellensp I made a FAT32 version too, it would not read it either.
I'm guessing my 64GB card is too big for the reader (some cheap reader cannot read car bigger than 16GB/32GB)

LoudMurmur, 2019-11-26:
I manage to make my 64GB sd card work, it did not solve the bootloop problem, the partition had to be of type "main" and not "logical".

I'll receive my new SKR 1.3 board in 2 days and try again

LoudMurmur, 2019-11-28:
Problem solved by activating eeprom reset on error (uncomment : #define EEPROM_AUTO_INIT)

closed 2019-11-28 by LoudMurmur