MarlinFirmware/Marlin #15985: [BUG] skr1.1Pro TMC2209 Z- Sensorles Homing

Lino77, 2019-11-23:
Hello Can it be That sensorless homing on the Z axis does not work. I want my Prusa clone on Z-max home, but no matter what value I enter (M914 Z0-255) he does not stop
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3882901/Config.zip)

hackebike, 2019-11-24:
As far as I know sensorless probing works fine. Which TMC 2209 do you have? I have BTT TMC 2209 v1.2 and both sensorless homing and probing works fine, you need to find the right sensitivity value. Here is a [video](https://www.youtube.com/watch?v=UJwzSTcQslo).

Grogyan, 2019-11-24:
Need to point out that using Sensorless homing on Z axis is not advisable as it can damage your printer.
Where the nozzle is being forced into the build surface

hackebike, 2019-11-24:
on the skr pro you do not need to use any jumpers

dch1921, 2019-11-25:
> Hello Can it be That sensorless homing on the Z axis does not work. I want my Prusa clone on Z-max home, but no matter what value I enter (M914 Z0-255) he does not stop
> [Config.zip](https://github.com/MarlinFirmware/Marlin/files/3882901/Config.zip)

You need to uncomment
//#define SENSORLESS_PROBING // StallGuard capable drivers only

for starters in configuration_adv.h

Lino77, 2019-11-25:
I put zmax on diag but he does not react. Z Probing I do not want to use for probing I have a Pindasensor I just want the home on z-max, and that does not work. look at my config. at the tmc 2130 under marlin 1.1.9 with a rumba the printer reacted to z-max. it seems to me that zMax or min only works with probing enabled, and sensorless homing works only on x and y
#if ENABLED(SENSORLESS_HOMING)

#define X_DIAG_PIN       PB10 
  #define Y_DIAG_PIN      PE12   // PC15
  #define Z_DIAG_PIN      PG8    // PC4
  #define E0_DIAG_PIN     PE15   // PB23
  #define E1_DIAG_PIN     PE10   // PD0
  #define E2_DIAG_PIN     PG5   // PD0

  #if X_HOME_DIR < 0
    #define X_MIN_PIN      X_DIAG_PIN
    #define X_MAX_PIN      PE15
  #else
    #define X_MIN_PIN      PB10
    #define X_MAX_PIN      X_DIAG_PIN
  #endif

  #if Y_HOME_DIR < 0
    #define Y_MIN_PIN      Y_DIAG_PIN
    #define Y_MAX_PIN      PE10
  #else
    #define Y_MIN_PIN      PE12
    #define Y_MAX_PIN      Y_DIAG_PIN
  #endif

  #if Z_HOME_DIR < 0
    #define Z_MIN_PIN      Z_DIAG_PIN
    #define Z_MAX_PIN      PG5
  #else
    #define Z_MIN_PIN      PB8
    #define Z_MAX_PIN      Z_DIAG_PIN
  #endif

boelle, 2019-11-26:
> 1.1.9

try with 2.0.x instead

Lino77, 2019-11-26:
You misunderstood me I already use the 2.0. just wanted to say that she ever ran under 1.19.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

Lino77, 2019-12-07:
I have not come to the test yet

boelle, 2019-12-24:
tried the test yet?

closed 2019-12-28 by Lino77