MarlinFirmware/Marlin #15991: [BUG] [Bugfix 2.0.x] Don't work M851 (for SKR V1.3)

colcolcheese, 2019-11-24:
### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

M420 V
Bilinear Leveling Grid:
 0      1      2
 0 +4.761 +4.618 +4.805
 1 +4.539 +4.403 +4.523
 2 +4.952 +4.757 +4.887
echo:Bed Leveling OFF
echo:Fade Height OFF

M851 Z[n]
M500
M501

hardware reset & power offon

G28
M420 S1
Bilinear Leveling Grid:
 0      1      2
 0 +4.761 +4.618 +4.805
 1 +4.539 +4.403 +4.523
 2 +4.952 +4.757 +4.887
echo:Bed Leveling ON
echo:Fade Height OFF

G90
G1 Z20
G1 X10 Y10
G1 Z0
nozzle height does not change even if [n] is changed
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3883618/config.zip)

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ReneRebsdorf, 2019-11-25:
I actually ran across this today as well
A G28 Z makes the offset change take effect.  I think it is by design, but couldn't live without it would not take effect immediately after issuing m851 and just adjust current z pos based on the difference

thinkyhead, 2019-12-02:
> couldn't quite decide why it would not take effect immediately after issuing m851

There are issues with that. So first you change the offset with `M851` and then you do `G28` and `G29` using the new offset.

closed 2019-12-02 by thinkyhead