MarlinFirmware/Marlin #15992: [BUG] Motor stutter return on Deltas after commit 4db68d704f with RRDFGDC

Kaween-prog, 2019-11-24:
Follow up to : "[BUG] Print resumes after pause, "wait for print to resume" stays on screen. #15414 

Marlin 2.0.1 Bugfix (all commits I've tested so far after 4db68d704f) on SKR1.3 and using RRDFGD. Delta printer.

Description
Heavy motor stutter during movements on all towers, resulting is unusable bad print quality on a Delta printer.

Steps to Reproduce
* use firmware prior to 4db68d704f : print quality is excellent, however the display refresh is erratic and will for example cause the #"wait for print to resume" delay (#15414)

after pause has been succesfully carried out, click to resume

Expected behavior:
Print Paused Wait for Print to resume ..." would be displayed until the print actually resumes, and after that the LCD should show the main info screen again.

Actual behavior:
Print Paused Wait for Print to resume ... STAYS on the LCD screen for 3 to 4 minutes, although the printer does correctly resume the print. After a (seemingly) artbitrary 3 to 4 minutes the LCD does return back to the info screen and everythings works correctly.

Hardware :
Anycubic Predator Delta
SKR 1.3 Board
Bigtreetech 2.4TFT with 12864 emulation mode
X/Y/Z/E0 = TMC2208 in Uart mode
latest bugfix 2.0 code used + several older versions (same issue)

Also tested with TMC2100 in Step/Dir to exclude UART functionality as issue. Does indeed behave the same way.

Anyone any idea what I'm missing here ? I'm pretty sure I botched up something, but no idea what or where. Note : using the "change filament" LCD option does not show this issue, and "resume" fully works as intended in that case without the LCD betting stuck on the same "Wait for Print to resume ...." LCD condition.

Update / Clarification :
the issue +only+ exsists when the users choses a manual "pause" function over the LCD. If the printer has to deal with a filament out condition or the manual "load/unload" functions are used, the issue does not occur and the LCD returns to the main menu flawlessly upon resuming the print.

Update 2
Another user (Shitcreek, thank you for chiming in and testing) confirmed the issue on a Cartesian, so it's not related to the type of printer like the block_buffer_size issue seemed to be.

boelle, 2019-11-24:
there is no 2.0.1 yet, typo?

boelle, 2019-11-24:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Kaween-prog, 2019-11-24:
Hello,

here's both config files, As said, when using the build prior to 4db68d704f,  the machine will react well in terms of producing good print quality and smooth motor movement but it will show the "LCD response is erratic/delayed.  Versions after this point do not have any issues with screen updates, but at the cost of bad motor movement.

[stutter issue.zip](https://github.com/MarlinFirmware/Marlin/files/3883888/stutter.issue.zip)

Several people have reported the same issue, specifically when using the combination of Delta machines and the RepReapDiscountFullgraph displays, using different controller boards than the SKR1.3, so I'm presuming it's not a specific SKR issue.

Kaween-prog, 2019-11-25:
Update, tested with my Fysetc Mini. Same result : the axis motor movements are very stuttery and print quality is unusable with 4db68d704f and beyond.  Prior to 4db68d704f, motor movements are smooth and print quality is excellent using basically the same configuration files (some changes where needed as my version prior to 4db68d704f needed a different declaration in terms of bed limits and it didn't have the "classic jerk" variables.

Kaween-prog, 2019-11-26:
Update : 

Whatever was broken in 4db68D7f seems to be fixed in  31e329f.  Movement with the Fysetc Mini is smooth, RRDFGDC not tested yet.  I'll check this later tomorrow, but so far whatever was broken seems to be fixed now.

Could someone please be so kind however to explain exactly how I have to define the value of the calibration radius ?  On the older builds, there was a simple DELTA_CALIBRATION_RADIUS, easy to use.

boelle, 2019-11-26:
remember to hit close if display also work

Kaween-prog, 2019-11-26:
I will ! If you or any other kind soul could give me a hint on the added question about the change in terms of Delta Calibration Radius I just added above, I'd be very greatful. :)

Kaween-prog, 2019-11-29:
Headsup : the newer versions (after 261119) solve the issue : no motor stutter, both with the Fysetc Mini's and variations of the RRDFGDC's, inclusive the BTT Hybrid LCD's.

closed 2019-11-29 by Kaween-prog