MarlinFirmware/Marlin #15994: [BUG] (short description)

jafidod, 2019-11-24:
### Description

G33 auto calibrate command doesn't produce report/output
 
### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 
[G33doesntPrintResults.zip](https://github.com/MarlinFirmware/Marlin/files/3883975/G33doesntPrintResults.zip)

1. connect from Serial Monitor to Delta Kossel Linear Plus printer
2. send G33 command 
3. follow on the screen instructions

**Expected behavior:** [What you expect to happen]
Displayed values similar to [http://marlinfw.org/docs/gcode/G033.html](url) example

> G33
> 
> > G33 Auto Calibrate
> > Checking... AC
> > .Height:295.00    Ex:+0.00  Ey:+0.00  Ez:+0.00    Radius:100.00
> > .Tower angle :    Tx:+0.00  Ty:+0.00  Tz:+0.00
> > Iteration : 01                                    std dev:2.665
> > .Height:297.85    Ex:-0.18  Ey:-0.13  Ez:+0.00    Radius:100.68
> > .Tower angle :    Tx:-0.05  Ty:+0.08  Tz:+0.00
> > Iteration : 02                                    std dev:0.128
> 

**Actual behavior:** [What actually happens]

> G33 Auto Calibrate
> Checking... AC
> .Height:269.58  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:134.40
> .               Tx:+0.00  Ty:+0.00  Tz:+0.00
> echo:busy: processing
> echo:busy: processing
> Deploy Z-Probe
> echo:busy: paused for user
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> Correct delta settings with M665 and M666
> echo:busy: processing
> Stow Z-Probe
> echo:busy: paused for user

#### Additional Information
SW: Marlin-bugfix-2.0.x downloaded yesterday
HW Anycubic Kossel Linear Plus with Trigorilla 1.1 board

vijexa, 2019-11-25:
Same thing. Kossel linear plus, SKR v1.3 board. Configuration.h from "config/delta/Anycubic/Kossel", I've modified only a motherboard. Marlin downloaded from bugfix-2.0.x branch on 23rd november. I believe I downloaded it after [that](https://github.com/MarlinFirmware/Marlin/commit/8f09d2e7904f8f743be149b6b213575e7f36ee29) commit. This is my first experience with deltas, so before seeing your issue I thought that I'm doing something wrong. 
Also, if delta auto calibration from LCD should set any delta settings, then it doesn't work too. Again, I don't know what is a normal behaviour of this feature, but it seems like it is doing nothing except meaninglessly probing 4 points. 1 in a center and 3 in a circle with 1cm radius (shouldn't it probe whole print surface to get better results?).

vijexa, 2019-11-25:
Ok guys! I think I can confirm that this is definitely a bug. I was scrolling through commits and saw a commit with suspicious comment ["Improve Delta probing / calibration (#15887)"](https://github.com/MarlinFirmware/Marlin/commit/b904ba0f291cdcb9759464f064f85ba066b778ce). I've uploaded a FM from [this commit] (https://github.com/MarlinFirmware/Marlin/commit/4cffee3ca391e6e066fa08ae4cb8aa9ddcd0c1c3) made a few hours earlier and finally everything works as intended, my delta calibrates itself in 7 full iterations. After that I've uploaded "Improve Delta probing / calibration" FM and nothing works again. Auto calibration probes 1 point in center, 3 points in 1cm radius and this time 1 point in front of one of towers. After that it goes up and prompts to stow Z-Probe. 
For everyone that encountered this problem - rollback to older commits while this issue isn't solved.

sjasonsmith, 2019-11-25:
Well, that was my commit, and I developed it on the same hardware a jafidod. I'll take a look sometime tonight to see what is going on.

sjasonsmith, 2019-11-25:
It looks like it works if `DELTA_CALIBRATION_DEFAULT_POINTS` is left at the default of 4 from the example config.

@vijexa, could you post your configs as well?

sjasonsmith, 2019-11-26:
I think I have the problem fixed. I'm doing some testing now and will post a change shortly.

sjasonsmith, 2019-11-26:
@vijexa and @jafidod, I have posted a small fix in pull request #16003. 

Could you both try this change, and comment on the pull request whether it solved the problem for you?

GurenWolf, 2019-11-26:
You people are fast on this, I was the slow one trying to find an answer everywhere else except here. Exactly same issue since yesterday, November 24th 2019. I will give that change a try as well to see what happens.

GurenWolf, 2019-11-26:
So I tested it, and it worked perfectly. The autocalibration runs the n amount of times and it correctly displays and changes the values. It is just a matter of hitting an M500 to save everything.

I don't know if it is related, but after further testing with UBL (now that all lengths and heights are correctly displayed and stored), when doing the step by step UBL, the command only probes 52 points on the base out of a 100, and after running step 3 on Validate Mesh, this message appears:

Unknown command: "G26 C B60 H200 P"

sjasonsmith, 2019-11-26:
@GurenWolf, The other issues you described are separate from this. I don't know how many of those UBL capabilities work on delta. With the number of probe points, I think the full 100 points is a square, and you only probe 52 of them because those are the only ones that land in your circular bed.

Thanks for testing the change, I'm glad to hear it worked.

vijexa, 2019-11-26:
Wow, that was fast! 
Thanks for your work, @GurenWolf. Problem is now fixed on my setup. 
If you still need my config files, here they are:
This one is from commit with bug 
[Configuration_bug.zip](https://github.com/MarlinFirmware/Marlin/files/3891075/Configuration_bug.zip)
This one is from your fix 
[Configuration_fix.zip](https://github.com/MarlinFirmware/Marlin/files/3891077/Configuration_fix.zip)
They should be identical, I believe. I haven't touched delta settings at all (well, I've uncommented `#define ANYCUBIC_KOSSEL_PLUS`, which directly affects delta settings), only some minor unrelated changes.

sjasonsmith, 2019-11-26:
Considering I introduced the problem I figured I should fix it :)

sjasonsmith, 2019-11-26:
@jafidod, this change was just merged into bugfix-2.0.x. Can you verify that it solves the problem for you, and close this Issue if it does?

vijexa, 2019-11-26:
I think I've encountered some other issue with calibration. Quick question: what behaviour is expected from "Calibrate X", "Calibrate Y" and "Calibrate Z" in Delta Settings LCD menu? These entries appear after successful auto calibration. When I choose one of them, printer just starts to shake printhead violently in corresponding tower direction and reboots after few seconds. All endstops and z-probe works. Calibrate Center works as intended (I believe). Here is a video: https://www.youtube.com/watch?v=qgJ1AZSIq4c
Do I need to create an issue for this or am I just doing something wrong?

sjasonsmith, 2019-11-26:
I have never tried those, I don't know what they are supposed to do. I just built a firmware before any of my original changes were merged, and it did the same thing, so it seems like an unrelated issue.

vijexa, 2019-11-26:
I've opened an [issue](https://github.com/MarlinFirmware/Marlin/issues/16007), could you confirm it in comments since you already checked this? Sorry if this is considered off-topic.

jafidod, 2019-11-27:
I've checked the latest version and it looks good. G33 command produces SD every iteration. After the last iteration is performed it reports calibration values.
I would expect little bit more talkative output of G33 e.g. report values of intermediate steps different from SD, because you can observe by this trend of calibration but this is more like feature request.

closed 2019-11-27 by jafidod