MarlinFirmware/Marlin #15998: Marlin-bugfix-2.0.x BOARD_MKS_ROBIN_LITE STM32F103RCT6

electrocoder, 2019-11-25:
### Description

STM32 MKS Robin lite Mother Board upload %100 not working. 

### Steps to Reproduce
My config files : [https://gofile.io/?c=jUKcwS](https://gofile.io/?c=jUKcwS)

1. Marlin bugfix-2.0.x download
2. Configuration.h -> #define MOTHERBOARD BOARD_MKS_ROBIN_LITE
3. Atom -> platformio.ini -> default_envs = mks_robin_lite
4. PIO Build 
5. Copy .pio/build/mks_robin_lite/mksLite.bin to SDCARD

upload %100 but froze

My Board Image: https://serving.photos.photobox.com/16920893a419135d6482a073ceeeaeac7b4d0740d99e7a3b1c58df559a69407d51512bbf.jpg
My TFT Image:
https://serving.photos.photobox.com/262235629f40b630be66146f1e379023d7191ef14ad62ea910049cb6fa6e55a6bb986dc7.jpg

RudyBenner, 2019-11-25:
I believe the file you need to copy to the uSD card is 

**firmware.bin**

I do see the mksLite.bin file.

RudyBenner, 2019-11-25:
mksLite.bin and firmware.bin are the same size, but are not identical.

RudyBenner, 2019-11-25:
> ### Description
> STM32 MKS Robin lite Mother Board upload %100 not working.
>
> ### Steps to Reproduce
> My config files : https://gofile.io/?c=jUKcwS
>
> 1. Marlin bugfix-2.0.x download
> 2. Configuration.h [...]

nice build

electrocoder, 2019-11-25:
firmware.bin does not work. I changed the file name to mksLite.bin. I get the same error. It's 100%.

RudyBenner, 2019-11-25:
Platformio generates 2 bin files. One is **firmware.bin**, the other is **mksLite.bin**.

RudyBenner, 2019-11-25:
I believe the uSD card needs to be formatted in FAT32 ? Is there a size limit?

noahdavis319, 2019-11-27:
As previously described, please upload the **firmware.bin** file to the SD card, and not **mksLite.bin**

Additionally, if you are building on a Windows-based machine in Visual Studio Code, please specify what version of PlatformIO Core you are using. There are reports of 4.1.0 not generating working binary files for some people.

boelle, 2019-12-02:
@electrocoder did you do what @noahdavis319 said?

> Platformio generates 2 bin files. One is **firmware.bin**, the other is **mksLite.bin**.
> As previously described, please upload the **firmware.bin** file to the SD card, and not **mksLite.bin**

boelle, 2019-12-07:
@electrocoder since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-11:
https://github.com/MarlinFirmware/Marlin/issues/15910 same issue so this one is a dublicate and we dont need those

i couldd be wrong and in that case will reopen

closed 2019-12-11 by boelle