MarlinFirmware/Marlin #16000: [BUG] G29 ABL skips last row of probe points

Kokorone, 2019-11-25:
### Description

When initiating a probe mesh creation (bilinear for me in a 3x3 grid), the final row of probe points will be entirely skipped.

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. Configure firmware to use bilinear abl
2. Start a print that includes G29 in the start gcode
3. Observe the printer probing the first 6 points (for 3x3 grid), before failing on the last row and skipping to the print

**Expected behavior:** [What you expect to happen]
Complete ABL of all 9 points without any issue and begin print
**Actual behavior:** [What actually happens]
Printer will probe the first 2 row of the bed and then skip straight to printing
#### Additional Information
To provide more visual aid of the issue, imagine this as the 3x3 grid
789
456
123

The printer will probe points 1-6, in a zig zag motion starting with point 1 and ending on point 4. then skip points 7-8. As an extra test, I used the flag #PROBE_Y_FIRST and it had an even stranger behavior. When probe Y first is enabled, the printer should zig zag through the grid starting from point 1 and moving to point 7 and so on. But with this flag, it will only probe points 1 and 4 before halting, meaning that the entire last row is causing some kind of bug with probing. 

The issue is present on any grid size, as testing with a 4x4 grid had it fail on the last row as well. 

I have pulled an older build of marlin from a few weeks ago and it does not exhibit this problem.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3888531/Marlin.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

pfillion42, 2019-11-26:
Try to start anew with the last bugfix if you have not already done so. I had the same strange problem last week. I found my solution by downloading the bugfix of November 23rd which was the last of the moment. I redid my configuration files and everything is back in order.

Kokorone, 2019-11-26:
Rebuilt with the latest commits as of 7 am PST on the 26th. Still exhibiting the same probing issues. Not even sure if it is using a mesh bed level as the mesh never completes

Kokorone, 2019-11-26:
Turned on #DEBUG_LEVELING_FEATURE and this is the readout when in #PROBE_Y_FIRST mode.

`Send: G29
Recv:   current_position= X149.00 Y205.25 Z10.85 : >>> G29
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X1.00 Y-55.25 Z-0.85 (Right-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv:   current_position= X149.00 Y205.25 Z10.85 : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to X149.00 Y205.25 Z10.85
Recv: <<< do_blocking_move_to
Recv: reset_bed_level
Recv: >>> probe_at_point(15.00, 15.00, raise, 0, probe_relative)
Recv:   current_position= X149.00 Y205.25 Z10.85 :
Recv: >>> do_blocking_move_to X14.00 Y70.25 Z10.85
Recv: <<< do_blocking_move_to
Recv:  T:71.48 /0.00 B:44.26 /0.00 @:0 B@:0
Recv:   current_position= X14.00 Y70.25 Z10.85 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X14.00 Y70.25 Z10.85 : >>> run_z_probe
Recv:   current_position= X14.00 Y70.25 Z10.85 : >>> do_probe_move
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X14.00 Y70.25 Z-1.15
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:71.23 /0.00 B:44.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv:   current_position= X14.00 Y70.25 Z0.95 : sync_plan_position
Recv:   current_position= X14.00 Y70.25 Z0.95 : <<< do_probe_move
Recv:   current_position= X14.00 Y70.25 Z0.95 : <<< run_z_probe
Recv: >>> do_blocking_move_to X14.00 Y70.25 Z10.95
Recv: <<< do_blocking_move_to
Recv:  T:70.88 /0.00 B:44.19 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< probe_at_point
Recv: >>> probe_at_point(15.00, 130.00, raise, 0, probe_relative)
Recv:   current_position= X14.00 Y70.25 Z10.95 :
Recv: >>> do_blocking_move_to X14.00 Y185.25 Z10.95
Recv: <<< do_blocking_move_to
Recv:  T:70.63 /0.00 B:44.12 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:   current_position= X14.00 Y185.25 Z10.95 : set_probe_deployed
Recv: deploy: 1
Recv:   current_position= X14.00 Y185.25 Z10.95 : >>> run_z_probe
Recv:   current_position= X14.00 Y185.25 Z10.95 : >>> do_probe_move
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X14.00 Y185.25 Z-1.15
Recv: <<< do_blocking_move_to
Recv:  T:70.39 /0.00 B:44.04 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:70.14 /0.00 B:44.04 /0.00 @:0 B@:0
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv:   current_position= X14.00 Y185.25 Z0.81 : sync_plan_position
Recv:   current_position= X14.00 Y185.25 Z0.81 : <<< do_probe_move
Recv:   current_position= X14.00 Y185.25 Z0.81 : <<< run_z_probe
Recv: >>> do_blocking_move_to X14.00 Y185.25 Z10.81
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:69.92 /0.00 B:43.97 /0.00 @:0 B@:0
Recv: <<< probe_at_point
Recv: >>> probe_at_point(15.00, 245.00, raise, 0, probe_relative)
Recv:   current_position= X14.00 Y185.25 Z10.81 :
Recv:   current_position= X14.00 Y185.25 Z10.81 : set_probe_deployed
Recv: deploy: 0
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to X14.00 Y185.25 Z10.85
Recv: <<< do_blocking_move_to
Recv: >>> do_blocking_move_to X14.00 Y185.25 Z10.85
Recv: <<< do_blocking_move_to
Recv:   current_position= X14.00 Y185.25 Z10.85 : > probing complete
Recv: <<< G29`

Kokorone, 2019-11-26:
Linked is a video of the issue while in #PROBE_Y_FIRST mode since its the quickest way to get the probing to halt on the last row. 

[https://cdn.discordapp.com/attachments/491105274464043026/648909595858829341/video0.mov](url)

Kokorone, 2019-11-26:
Did some further testing with some discord members and it appears to be an issue with setting a noninteger probe offset in the Y axis. Attached is a more in depth explanation of it.

[image]

Setting the offset to a full integer value has probing resume to a proper full mesh. This is a marlin bug

AnHardt, 2019-11-26:
Limitation - yes. Bug - no.
```
/**
 *   Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *   X and Y offsets must be integers. !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 *
 *   In the following example the X and Y offsets are both positive:
 *   #define X_PROBE_OFFSET_FROM_EXTRUDER 10
 *   #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
 *
 *      +-- BACK ---+
 *      |           |
 *    L |    (+) P  | R <-- probe (20,20)
 *    E |           | I
 *    F | (-) N (+) | G <-- nozzle (10,10)
 *    T |           | H
 *      |    (-)    | T
 *      |           |
 *      O-- FRONT --+
 *    (0,0)
 */
```
We really can't do anything more than document the limitations.
If the users can't read - it's their problem.

davidlm101, 2019-11-26:
Hi,
I have the very same problem with marlin 2 (I downloaded on November 20th) and I don't have a noninteger offset in the configuration that I'm aware of (sorry, I'm new).  It only probes the first 6 points (of 9) and it then stops with no errors.

I enabled debug and this is the output if it helps:
>>>M111 S32
SENDING:M111 S32
echo:DEBUG:LEVELING
>>>g29
SENDING:G29
  current_position= X80.00 Y115.00 Z13.40 : >>> G29
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X37.00 Y2.00 Z-3.40 (Right-Back & Below Nozzle)
Auto Bed Leveling: BILINEAR
 (disabled)
  current_position= X80.00 Y115.00 Z13.40 : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to X80.00 Y115.00 Z13.40
<<< do_blocking_move_to
reset_bed_level
>>> probe_at_point(37.00, 20.00, raise, 0, probe_relative)
  current_position= X80.00 Y115.00 Z13.40 :
>>> do_blocking_move_to X0.00 Y18.00 Z13.40
<<< do_blocking_move_to
  current_position= X0.00 Y18.00 Z13.40 : set_probe_deployed
deploy: 1
  current_position= X0.00 Y18.00 Z13.40 : >>> run_z_probe
  current_position= X0.00 Y18.00 Z13.40 : >>> do_probe_move
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
>>> do_blocking_move_to X0.00 Y18.00 Z1.40
<<< do_blocking_move_to
echo:busy: processing
echo:busy: processing
echo:busy: processing
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X0.00 Y18.00 Z3.47 : sync_plan_position
  current_position= X0.00 Y18.00 Z3.47 : <<< do_probe_move
  current_position= X0.00 Y18.00 Z3.47 : <<< run_z_probe
>>> do_blocking_move_to X0.00 Y18.00 Z8.47
<<< do_blocking_move_to
echo:busy: processing
<<< probe_at_point
>>> probe_at_point(126.00, 20.00, raise, 0, probe_relative)
  current_position= X0.00 Y18.00 Z8.47 :
>>> do_blocking_move_to X89.00 Y18.00 Z8.47
<<< do_blocking_move_to
echo:busy: processing
  current_position= X89.00 Y18.00 Z8.47 : set_probe_deployed
deploy: 1
  current_position= X89.00 Y18.00 Z8.47 : >>> run_z_probe
  current_position= X89.00 Y18.00 Z8.47 : >>> do_probe_move
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
>>> do_blocking_move_to X89.00 Y18.00 Z1.40
<<< do_blocking_move_to
echo:busy: processing
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X89.00 Y18.00 Z3.41 : sync_plan_position
  current_position= X89.00 Y18.00 Z3.41 : <<< do_probe_move
  current_position= X89.00 Y18.00 Z3.41 : <<< run_z_probe
>>> do_blocking_move_to X89.00 Y18.00 Z8.41
<<< do_blocking_move_to
[... 93 lines cut ...]
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X0.00 Y116.00 Z3.50 : sync_plan_position
  current_position= X0.00 Y116.00 Z3.50 : <<< do_probe_move
  current_position= X0.00 Y116.00 Z3.50 : <<< run_z_probe
>>> do_blocking_move_to X0.00 Y116.00 Z8.50
<<< do_blocking_move_to
echo:busy: processing
<<< probe_at_point
>>> probe_at_point(-52.00, 118.00, raise, 0, probe_relative)
  current_position= X0.00 Y116.00 Z8.50 :
  current_position= X0.00 Y116.00 Z8.50 : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to X0.00 Y116.00 Z13.40
<<< do_blocking_move_to
echo:busy: processing
>>> do_blocking_move_to X0.00 Y116.00 Z13.40
<<< do_blocking_move_to
  current_position= X0.00 Y116.00 Z13.40 : > probing complete
<<< G29

>>>M111 S0 
SENDING:M111 S0
echo:DEBUG:off

thinkyhead, 2019-11-26:
The offsets were made editable, and thus were transformed into `float` for the implementation. It was assumed this would free users to apply non-integer XY probe offsets. But the logic must still contain some assumptions that rely on these offsets being integers. So, I will either fix the issue in the code, or patch things so that only integer XY probe offsets are possible.

thinkyhead, 2019-11-26:
@davidlm101 — I sort of see what's going on with the loop, so I could have a patch soon. If you do a `G29 V2` there will be even more helpful output that should make it even more obvious that the loop is broken.

sjasonsmith, 2019-11-26:
@thinkyhead I've already been looking at this and have a fix I can clean up and post. It's up to you if you want me to proceed or handle it on your own.

thinkyhead, 2019-11-26:
I would like to see what you've got. So far I'm just staring at the code and speculating.

sjasonsmith, 2019-11-26:
OK, I need to clean up some of my temporary debug changes. I've been testing with hardware.

sjasonsmith, 2019-11-27:
@Kokorone, could you try this again using the changes from the pull request referenced above?

thinkyhead, 2019-11-27:
> `X and Y offsets must be integers.`

I note that this is out of date. The comment went away when the sanity check went away.

davidlm101, 2019-11-27:
> 
> 
> @davidlm101 — I sort of see what's going on with the loop, so I could have a patch soon. If you do a `G29 V2` there will be even more helpful output that should make it even more obvious that the loop is broken.

Hi, Thanks for your answers.  I ran a G29 V2 and in the point 7 I see a negative number, maybe that's the problem.  

>Probing mesh point 7/9.
>probe_at_point(**-45.00**, 110.00, raise, 2, probe_relative)

If it helps I attached a zip file with my Configuration.h and Configuration_adv.h and the complete debug output.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3899064/config.zip)

<details><summary>Complete debug output</summary>

```
>>>M111 S32
SENDING:M111 S32
echo:DEBUG:LEVELING
>>>G29 V2
SENDING:G29 V2
  current_position= X73.00 Y108.00 Z13.40 : >>> G29
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X37.00 Y2.00 Z-3.40 (Right-Back & Below Nozzle)
Auto Bed Leveling: BILINEAR
 (disabled)
G29 Auto Bed Leveling
  current_position= X73.00 Y108.00 Z13.40 : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to X73.00 Y108.00 Z13.40
<<< do_blocking_move_to
reset_bed_level
Probing mesh point 1/9.
>>> probe_at_point(37.00, 20.00, raise, 2, probe_relative)
  current_position= X73.00 Y108.00 Z13.40 :
>>> do_blocking_move_to X0.00 Y18.00 Z13.40
<<< do_blocking_move_to
  current_position= X0.00 Y18.00 Z13.40 : set_probe_deployed
deploy: 1
  current_position= X0.00 Y18.00 Z13.40 : >>> run_z_probe
  current_position= X0.00 Y18.00 Z13.40 : >>> do_probe_move
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
>>> do_blocking_move_to X0.00 Y18.00 Z1.40
<<< do_blocking_move_to
echo:busy: processing
echo:busy: processing
echo:busy: processing
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X0.00 Y18.00 Z3.45 : sync_plan_position
  current_position= X0.00 Y18.00 Z3.45 : <<< do_probe_move
  current_position= X0.00 Y18.00 Z3.45 : <<< run_z_probe
>>> do_blocking_move_to X0.00 Y18.00 Z8.45
<<< do_blocking_move_to
echo:busy: processing
<<< probe_at_point
Probing mesh point 2/9.
>>> probe_at_point(119.00, 20.00, raise, 2, probe_relative)
  current_position= X0.00 Y18.00 Z8.45 :
>>> do_blocking_move_to X82.00 Y18.00 Z8.45
<<< do_blocking_move_to
echo:busy: processing
  current_position= X82.00 Y18.00 Z8.45 : set_probe_deployed
deploy: 1
  current_position= X82.00 Y18.00 Z8.45 : >>> run_z_probe
[... 114 lines cut ...]
  current_position= X0.00 Y108.00 Z3.46 : sync_plan_position
  current_position= X0.00 Y108.00 Z3.46 : <<< do_probe_move
  current_position= X0.00 Y108.00 Z3.46 : <<< run_z_probe
>>> do_blocking_move_to X0.00 Y108.00 Z8.46
<<< do_blocking_move_to
echo:busy: processing
<<< probe_at_point
Probing mesh point 7/9.
>>> probe_at_point(-45.00, 110.00, raise, 2, probe_relative)
  current_position= X0.00 Y108.00 Z8.46 :
  current_position= X0.00 Y108.00 Z8.46 : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to X0.00 Y108.00 Z13.40
<<< do_blocking_move_to
echo:busy: processing
>>> do_blocking_move_to X0.00 Y108.00 Z13.40
<<< do_blocking_move_to
  current_position= X0.00 Y108.00 Z13.40 : > probing complete
<<< G29
>>>M111 S0
SENDING:M111 S0
echo:DEBUG:off
```
</details>

sjasonsmith, 2019-11-27:
@davidlm101 Did you test again using the latest commits in bugfix-2.0.x? A change to address the original issue was merged yesterday.

davidlm101, 2019-11-28:
> 
> 
> @davidlm101 Did you test again using the latest commits in bugfix-2.0.x? A change to address the original issue was merged yesterday.

Today I did it and now bltouch is working fine. Thank you very much to everyone.

closed 2019-11-29 by thinkyhead